Nuand/bladeRF #602: bladerf_set_sampling on bladerf2 returns BLADERF_ERR_UNSUPPORTED

rtucker, 2018-09-04:
This should likely just silently accept 'internal' or whatnot, and go about its business.

Ref: pothosware/SoapyBladeRF#19

pk-r7, 2018-09-05:
How about a reply code that developers can interpret as "not implemented"?

rtucker, 2018-09-05:
@pk-r7 hmmm... what situations would warrant a `BLADERF_ERR_NOT_IMPLEMENTED` return instead of the current `BLADERF_ERR_UNSUPPORTED`?

also not sure how much this would help at runtime... the message that libbladeRF needs to communicate at runtime is "right now, this function is not supported with this device handle" and `BLADERF_ERR_UNSUPPORTED` is how that's currently communicated

pk-r7, 2018-09-05:
Yes I see that now, BLADERF_ERR_UNSUPPORTED should work, it appears the error handling in CubicSDR needs some work.

rtucker, 2018-09-05:
Would something like this be helpful to implementers?

```
/**
 * Check if a libbladeRF API function is supported by a given device.
 * 
 * Given a device handle and a pointer to a libbladeRF function, this checks
 * to see if the function is supported and implemented for the device.
 * 
 * @param      dev   Device handle
 * @param      fn    Function pointer
 *
 * @return     true if the function exists and is supported, false otherwise
 */
API_EXPORT
bool CALL_CONV bladerf_is_function_supported(struct bladerf *dev, void *fn);
```

Usage would be something like:

```
if (bladerf_is_function_supported(dev, &bladerf_get_sampling)) {
  // this will likely succeed
  status = bladerf_get_sampling(dev, &sampling);
} else {
  // just skip it
  status = 0;
  sampling = BLADERF_SAMPLING_UNKNOWN;
}
```