Nuand/bladeRF #612: bladeRF-cli: board state insufficient after FPGA SPI flash load/erase

bglod, 2018-09-07:
**Steps to reproduce:**
1. Load FPGA into SPI flash

```
$ bladeRF-cli -v verbose -L /path/to/hostedx40.rbf
Deferring device init until after FPGA load
<snip>
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/usb.c:131] Reporting no FPGA present - BLADERF_FORCE_NO_FPGA_PRESENT is set.
<snip>
Writing FPGA to flash for autoloading...
<snip>
Done.
$
```

2. Power cycle board and wait for LEDs to turn on (FPGA loaded from flash)
3. Erase FPGA from SPI flash. After completion, an ERROR is posted:

```
$ bladeRF-cli -v verbose -L X
Deferring device init until after FPGA load
<snip>
Erasing stored FPGA to disable autoloading...
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 2
[INFO @ host/libraries/libbladeRF/src/backend/usb/usb.c:472] Erasing 55 blocks starting at block 4
[INFO @ host/libraries/libbladeRF/src/backend/usb/usb.c:477] Erased block 58
[INFO @ host/libraries/libbladeRF/src/backend/usb/usb.c:485] Done erasing 55 blocks
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 1
Done.
[ERROR @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:2153] Board state insufficient for operation (current "Firmware Loaded", requires "Initialized").
[ERROR @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:2153] Board state insufficient for operation (current "Firmware Loaded", requires "Initialized").
```

4. If the CLI is started in interactive mode and we're issuing an erase command, we get a similar behavior:

```
$ bladeRF-cli -v verbose -L X -i
Deferring device init until after FPGA load
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/usb.c:131] Reporting no FPGA present - BLADERF_FORCE_NO_FPGA_PRESENT is set.
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/usb.c:131] Reporting no FPGA present - BLADERF_FORCE_NO_FPGA_PRESENT is set.
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/usb.c:131] Reporting no FPGA present - BLADERF_FORCE_NO_FPGA_PRESENT is set.
[WARNING @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:1005] FPGA bitstream file not found.
[WARNING @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:1006] Skipping further initialization...
Erasing stored FPGA to disable autoloading...
[INFO @ host/libraries/libbladeRF/src/backend/usb/usb.c:472] Erasing 55 blocks starting at block 4
[INFO @ host/libraries/libbladeRF/src/backend/usb/usb.c:477] Erased block 58
[INFO @ host/libraries/libbladeRF/src/backend/usb/usb.c:485] Done erasing 55 blocks
Done.
bladeRF> version

  bladeRF-cli version:        1.6.1-git-deed07a3-dirty
  libbladeRF version:         2.0.1-git-deed07a3-dirty

[ERROR @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:1162] Board state insufficient for operation (current "Firmware Loaded", requires "FPGA Loaded").

  Error: Insufficient initialization for the requested operation

bladeRF> quit
[ERROR @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:2153] Board state insufficient for operation (current "Firmware Loaded", requires "Initialized").
[ERROR @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:2153] Board state insufficient for operation (current "Firmware Loaded", requires "Initialized").
```

The "Firmware Loaded, requires Initialized" case occurs in the CLI `main.c` during tear-down at `cli_state_destroy()`, which calls `rxtx_shutdown()` --> `disarm_triggers()` --> `bladerf_trigger_init()`, where the board state is checked and it errors out. The board was never initialized because of the flash operation.
https://github.com/Nuand/bladeRF/blob/aaca552637b03bacc82caf175b22a7cfdb873519/host/utilities/bladeRF-cli/src/main.c#L550
https://github.com/Nuand/bladeRF/blob/aaca552637b03bacc82caf175b22a7cfdb873519/host/utilities/bladeRF-cli/src/common.c#L144-L150
https://github.com/Nuand/bladeRF/blob/aaca552637b03bacc82caf175b22a7cfdb873519/host/utilities/bladeRF-cli/src/cmd/rxtx.c#L470-L474

For the interactive "version" command case, we also see the "Firmware loaded, requires FPGA loaded" error .. but the FPGA was loaded via SPI flash. The debug output says `BLADERF_FORCE_NO_FPGA_PRESENT` is set, so the `is_fpga_configured()` function will always return false, causing the error.

https://github.com/Nuand/bladeRF/blob/aaca552637b03bacc82caf175b22a7cfdb873519/host/libraries/libbladeRF/src/backend/usb/usb.c#L116-L133

rtucker, 2018-09-07:
I think I've reproduced this in the debugger, and yep, I concur...

```
(gdb) run
Starting program: /home/rtucker/dev/bladeRF/host/build/output/bladeRF-cli -L X
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff692e700 (LWP 18652)]
Deferring device init until after FPGA load
[Thread 0x7ffff692e700 (LWP 18652) exited]
[New Thread 0x7ffff692e700 (LWP 18653)]
Erasing stored FPGA to disable autoloading...
[INFO @ host/libraries/libbladeRF/src/backend/usb/usb.c:455] Erasing 55 blocks starting at block 4
[INFO @ host/libraries/libbladeRF/src/backend/usb/usb.c:460] Erased block 58
[INFO @ host/libraries/libbladeRF/src/backend/usb/usb.c:468] Done erasing 55 blocks
Done.

Thread 1 "bladeRF-cli" hit Breakpoint 1, bladerf2_trigger_init (
    dev=0x5555557a6220, ch=0, signal=BLADERF_TRIGGER_J71_4, 
    trigger=0x7fffffffdd30)
    at /home/rtucker/dev/bladeRF/host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:3343
3343	    CHECK_BOARD_STATE(STATE_INITIALIZED);
(gdb) bt
#0  bladerf2_trigger_init (dev=0x5555557a6220, ch=0, 
    signal=BLADERF_TRIGGER_J71_4, trigger=0x7fffffffdd30)
    at /home/rtucker/dev/bladeRF/host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:3343
#1  0x00007ffff7b720e5 in bladerf_trigger_init (dev=0x5555557a6220, ch=0, 
    signal=BLADERF_TRIGGER_J71_4, trigger=0x7fffffffdd30)
    at /home/rtucker/dev/bladeRF/host/libraries/libbladeRF/src/bladerf.c:830
#2  0x00005555555724d0 in disarm_triggers (s=0x555555792280, dir=BLADERF_RX)
    at /home/rtucker/dev/bladeRF/host/utilities/bladeRF-cli/src/cmd/rxtx.c:84
#3  0x00005555555731d5 in rxtx_shutdown (s=0x555555792280, rxtx=0x5555557922e0)
    at /home/rtucker/dev/bladeRF/host/utilities/bladeRF-cli/src/cmd/rxtx.c:473
#4  0x000055555555edd1 in cli_state_destroy (s=0x555555792280)
    at /home/rtucker/dev/bladeRF/host/utilities/bladeRF-cli/src/common.c:150
#5  0x000055555555eab7 in main (argc=3, argv=0x7fffffffdf08)
    at /home/rtucker/dev/bladeRF/host/utilities/bladeRF-cli/src/main.c:550
(gdb) 
```

rtucker, 2018-09-07:
I am able to reproduce by:

1. disconnect/reconnect USB
2. do `bladeRF-cli -e version` (w/ host-based FPGA autoloading enabled)
3. do `bladeRF-cli -L X`

closed 2018-09-07 by rtucker