Nuand/bladeRF #616: segmentation fault (core dumped)

scriptKidi, 2018-09-10:
Hello there!

I installed GNU Radio and bladeRF via PyBOMBS as described here:

https://github.com/Nuand/bladeRF/wiki/Getting-Started%3A-Linux#Building_bladeRF_libraries_and_tools_from_source

When I connect my bladeRF and try to probe via $ bladeRF-cli -p, I get the following error:

Segmentation fault (core dumped)

The output of dmesg is:

do_general_protection: 29 callbacks suppressed
traps: bladeRF-cli[1730] general protection ip:7f5b82bbb186 sp:7ffc23b2e990 error:0 in libusb-1.0.so.0.1.0[7f5b82bb6000+17000]

After doing some research, the closest issue I could find was a segmentation fault afer running gqrx. The solution here was to upgrade libusb.

I then upgraded libusb, just to see if it helps, but it didn't.

Has anyone an idea what may cause this problem and how to resolve it?

bglod, 2018-09-10:
That link you have points to the manual build instructions, not the PyBOMBS instructions. Can you confirm which set of instructions you used?

Can you run the following commands as well:

`bladeRF-cli -v verbose -p`
`bladeRF-cli --version`

scriptKidi, 2018-09-10:
If you click on the link, you have to scroll up a bit until you reach the section 'Easy installation: PyBOMBS'.

I followed down the instructions until the next section starts, 'Easy installation for Ubuntu: The bladeRF PPA' (of course, I did not follow these).

I am sorry, I forgot to mention that I work with an Ubuntu 18.04 Virtual Machine installed in VMware Workstation.

The output of `bladeRF-cli -v verbose -p` is:

[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:286] Found a bladeRF
Segmentation fault (core dumped)

The output of `bladeRF-cli --version` is:

1.6.1-git-5ed03a3b01

bglod, 2018-09-10:
Oh wow it seems to be bailing out super early... and indeed somewhere in libusb. A VM should work, but they can sometimes have issues, so I'm glad you mentioned that detail. What is your host OS? Are you able to test directly on the host OS or perhaps use a live image?

Some other thoughts:
- Is the host OS trying to access the bladeRF at all, or is it simply being redirected to the VM?
- Are there any dongles / adapters / usb hubs / etc. in the way, or is this a direct USB connection?
- Reboot the VM after updating libusb (might want to make sure you have the latest kernel and system updates as well, just in case it's all related)
  - After rebooting, rebuild bladeRF: `pybombs [-p <prefix>] rebuild bladeRF`

rtucker, 2018-09-10:
On a hunch... could you give this a try with the fix in PR #617 ?

Quick way to do this:

```
cd pybombs/src/bladeRF   # or whereever your pybombs directory is
git checkout -b rtucker-dev-issue615 master
git pull https://github.com/rtucker/bladeRF.git dev-issue615
pybombs rebuild bladeRF
```

Then send along the output from `bladeRF-cli -v verbose -p` and `bladeRF-cli --version` if it continues not working :)

scriptKidi, 2018-09-10:
@bglod 

My host OS is Windows 10, but I have no drivers installed for the bladeRF so far. I have a dual boot with Ubuntu 16.04, so I will give it a try. Or would you rather recommend 18.04?

I find this worth noting: I followed the installation of only bladeRF (and not GNU Radio) here

scriptKidi, 2018-09-10:
@rtucker 

I followed your steps, but it's still not working.

The output of `bladeRF-cli -v verbose -p` is:

[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:295] Found a bladeRF
Segmentation fault (core dumped)

and the output of `bladeRF-cli --version` is:

1.6.1-git-a0e82de1a8

rtucker, 2018-09-10:
@scriptKidi 

Dang. Okay, it's very likely code I added. Quick way to be sure is to build the rev before I touched libusb.c, a4b176c516ae0ddd6ec357d79c330257f2b45c49:

```
cd pybombs/src/bladeRF   # or whereever your pybombs directory is
git checkout a4b176c516ae0ddd6ec357d79c330257f2b45c49
pybombs rebuild bladeRF
```

This should absolutely work OK on 18.04; that is, in fact, what I'm running on my development PC. So something weird is happening and it's not being caught...

Could you send along the output of `lsusb -d 2cf0: -v` (about 160 lines of stuff), as well as the part of `dmesg` where the bladeRF is being recognized immediately after being plugged in?

Please feel free to omit the serial numbers from both of the outputs. :)

The relevant output from `dmesg` will look something like this:

```
[1830101.876274] usb 4-2: new SuperSpeed USB device number 69 using xhci_hcd
[1830101.900639] usb 4-2: LPM exit latency is zeroed, disabling LPM.
[1830101.901012] usb 4-2: New USB device found, idVendor=2cf0, idProduct=5250
[1830101.901015] usb 4-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[1830101.901018] usb 4-2: Product: bladeRF 2.0
[1830101.901021] usb 4-2: Manufacturer: Nuand
[1830101.901023] usb 4-2: SerialNumber: <redacted>
```

Thanks!

rtucker, 2018-09-10:
whoops... please try the `dev-issue615` branch again...

```
cd pybombs/src/bladeRF   # or whereever your pybombs directory is
git checkout -b rtucker-dev-issue615 master
git pull https://github.com/rtucker/bladeRF.git dev-issue615
pybombs rebuild bladeRF
```

You might have to do `git branch -d rtucker-dev-issue615` before doing the `git checkout` ... basically just pull the latest stuff from rtucker/dev-issue615 and rebuild that :)

scriptKidi, 2018-09-11:
@rtucker 

- regarding your second comment

Sorry for this question, but am I right?:

I installed pybombs into my home directory. Inside it lies the prefix directory 'bladeRF'. Following your instructions (and getting fatal errors) I ran the command `git checkout -b rtucker-dev-issue615 master` inside '~/pybombs/bladeRF/src/bladeRF'.

Note: The command `git branch -d rtucker-dev-issue615` resulted in:

`error: branch 'rtucker-dev-issue615' not found.
`
Anyway, I could checkout, pull and rebuild, but the segfault still remains :( (Therefore the question above).

- regarding your first comment:

I followed the instructions and then ran `bladeRF-cli -p` which gave me a warning:

`[WARNING @ host/libraries/libbladeRF/src/backend/usb/libusb.c:313] Found a bladeRF via VID/PID, but could not open it due to insufficient permissions.`

- [ ] The output of `lsusb -d 2cf0: -v` is:

```
Bus 004 Device 008: ID <omitted>  
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         9
  idVendor           <omitted> 
  idProduct          <omitted> 
  bcdDevice            0.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          123
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              200mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       1
      bNumEndpoints           4
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
[... 76 lines cut ...]
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               1
```

- [ ] The output of `dmesg `is:

```
[ 6598.837998] usb 4-1: new SuperSpeed USB device number 7 using xhci_hcd
[ 6599.003230] usb 4-1: LPM exit latency is zeroed, disabling LPM.
[ 6599.006150] usb 4-1: New USB device found, idVendor=<omitted>, idProduct=<omitted>
[ 6599.006156] usb 4-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 6599.006161] usb 4-1: Product: bladeRF
[ 6599.006164] usb 4-1: Manufacturer: Nuand
[ 6599.006168] usb 4-1: SerialNumber: <omitted>
[ 6606.272552] traps: bladeRF-cli[5605] general protection ip:7f9821f6c186 sp:7fff875dc490 error:0 in libusb-1.0.so.0.1.0[7f9821f67000+17000]
[ 7044.712209] usb 4-1: USB disconnect, device number 7
```

bglod, 2018-09-11:
```
[WARNING @ host/libraries/libbladeRF/src/backend/usb/libusb.c:313] Found a
bladeRF via VID/PID, but could not open it due to insufficient permissions.
```

Make sure your user is in the `plugdev` group.

`groups`

If it isn't, add it, log out, and log back in so the change takes effect.

`usermod -a -G plugdev <username>`

Then try the probe command again.

bglod, 2018-09-11:
```
[WARNING @ host/libraries/libbladeRF/src/backend/usb/libusb.c:313] Found a
bladeRF via VID/PID, but could not open it due to insufficient permissions.
```

Make sure your user is in the `plugdev` group, type:

`groups`

If it isn't, add it, log out, and log back in so the change takes effect.

`usermod -a -G plugdev <username>`

Then try the probe command again.

scriptKidi, 2018-09-11:
I already checked that. My user is in the `plugdev` group.

bglod, 2018-09-11:
We usually don't recommend doing this long-term, but does running the
command with 'sudo' change the behavior at all? Do you still get the
permissions error?

scriptKidi, 2018-09-11:
I also already thought about that :) but it does not work either.

In fact, I got this error when executing `sudo bladeRF-cli -p`:

`sudo: bladeRF-cli: command not found`

scriptKidi, 2018-09-11:
This worked! I got the following output:

```
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:259] Found a bladeRF
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:326] Added instance 0 to device list

  Backend:        libusb
  Serial:         <omitted>
  USB Bus:        4
  USB Address:    2
```

bglod, 2018-09-11:
Oh, you know what ... the `bladeRF.lwr` PyBOMBS recipe disables the generation and installation of udev rules.

1. Unplug the bladeRF device from your machine.

2. Create/edit this file:
```
sudo nano /etc/udev/rules.d/88-nuand.rules
```

3. Paste this into it:
```
# Nuand bladeRF
ATTR{idVendor}=="2cf0", ATTR{idProduct}=="5246", MODE="660", GROUP="plugdev"
ATTR{idVendor}=="2cf0", ATTR{idProduct}=="5250", MODE="660", GROUP="plugdev"

# Nuand bladeRF, legacy VID/PID
ATTR{idVendor}=="1d50", ATTR{idProduct}=="6066", MODE="660", GROUP="plugdev"

# Cypress Bootloader
ATTR{idVendor}=="04b4", ATTR{idProduct}=="00f3", MODE="660", GROUP="plugdev"
```

4. Reload udev rules:
```
sudo udevadm control --reload-rules && sudo udevadm trigger
```

5. Plug in the bladeRF and try `bladeRF-cli -p` again

scriptKidi, 2018-09-11:
Now it runs smoothly!! Thanks a lot to you two!! :)

Note: I added the file 88-nuand.rules with the same content before I opened this issue, but as it did not solve my problem, I deleted that file again. So for me, the checkout was important.

To sum up the steps that helped me solve the problem:

```
cd pybombs/src/bladeRF   # or whereever your pybombs directory is
git checkout a4b176c516ae0ddd6ec357d79c330257f2b45c49
pybombs rebuild bladeRF
sudo nano /etc/udev/rules.d/88-nuand.rules
sudo udevadm control --reload-rules && sudo udevadm trigger
```

bglod, 2018-09-11:
Awesome!! Thanks for reporting this issue so quickly and bearing with the debug process.

PR #624 (2018-09-12): host: Add manufacturer/product strings to device information
Adds 'manufacturer' and 'product' members to the bladerf_devinfo struct, and ensures they are properly initialized and populated during enumeration.

These contain the human-readable manufacturer and product names from the USB device descriptor, and are displayed by bladeRF-cli during 'probe' and 'info' commands.

Note: This is a reimplementation of PR #611, which tried to be a little too clever.

Fixes #611 
Fixes #615 
Fixes #616 
Fixes #617
closed 2018-09-12 by scriptKidi