Nuand/bladeRF #640: bladerf1: 'cal table agc rx' fails on x115 w/ FPGA 0.7.3

rtucker, 2018-09-17:
Forum thread: https://nuand.com/forums/viewtopic.php?f=4&t=4964

Performing `cal table agc rx` with 2018.08 and FPGA 0.7.3 (official release) on an x115 fails:

```
((2018.08)) rtucker@rho:~/dev/bladeRF/host/build$ time output/bladeRF-cli -l /nuand/builds/releases/2018.08/fpga_v0.7.3/hostedx115-2018-08-07_21.26.57/hostedx115.rbf -e 'cal table agc rx' -v verbose
Deferring device init until after FPGA load
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:604] Using libusb version: 1.0.21.11156
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:481] Found a bladeRF (idx=9)
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:543] USB port reset succeeded for bladeRF 3e0b29f136c8dca5eb87894b2645bb93
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 0
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:781] Read Firmware version: 2.0.0
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:790] Capability mask before FPGA load: 0x0000000700000000
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 2
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/usb.c:131] Reporting no FPGA present - BLADERF_FORCE_NO_FPGA_PRESENT is set.
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 3
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 2
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/usb.c:131] Reporting no FPGA present - BLADERF_FORCE_NO_FPGA_PRESENT is set.
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 3
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:920] Loading RX calibration image 3e0b29f136c8dca5eb87894b2645bb93_dc_rx.tbl
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/usb.c:131] Reporting no FPGA present - BLADERF_FORCE_NO_FPGA_PRESENT is set.
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:936] is_fpga_configured returned 0
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:960] Loading FPGA from: /usr/share/Nuand/bladeRF/hostedx115.rbf
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 3
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/usb.c:131] Reporting no FPGA present - BLADERF_FORCE_NO_FPGA_PRESENT is set.
...
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/usb.c:131] Reporting no FPGA present - BLADERF_FORCE_NO_FPGA_PRESENT is set.
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:986] fpga load complete
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 1
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:505] Read FPGA version: 0.7.3
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:510] Capability mask after FPGA load: 0x00000007000007ff
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:392] nios_config_read: Read 0x00180000
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:577] Default GPIO value found - initializing device
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:403] nios_config_write: Wrote 0x00000057
...
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:369] Default tuning mode: FPGA
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:1809] Setting TX1 frequency to 2447000000
[VERBOSE @ fpga_common/src/lms.c:359] VCOCAP estimate: 31
[VERBOSE @ fpga_common/src/lms.c:1679] ---- Frequency ----
[VERBOSE @ fpga_common/src/lms.c:1680]   x        : 2
[VERBOSE @ fpga_common/src/lms.c:1681]   nint     : 127
[VERBOSE @ fpga_common/src/lms.c:1682]   nfrac    : 3757397
[VERBOSE @ fpga_common/src/lms.c:1683]   freqsel  : 0x2c
[VERBOSE @ fpga_common/src/lms.c:1684]   reference: 38400000
[VERBOSE @ fpga_common/src/lms.c:1685]   freq     : 2446999999
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:2015] bladerf1_schedule_retune: bork
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:987] nios_retune: channel=TX1 timestamp=0 nint=127 nfrac=3757397
				freqsel=0x2c vcocap=0x1f low_band=0 quick_tune=0
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1000] nios_retune: duration=0 vcocap=21 resp_flags=3
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1004] TX1 retune operation: vcocap=33, duration=0
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:1809] Setting RX1 frequency to 2484000000
[VERBOSE @ fpga_common/src/lms.c:359] VCOCAP estimate: 35
[VERBOSE @ fpga_common/src/lms.c:1679] ---- Frequency ----
[VERBOSE @ fpga_common/src/lms.c:1680]   x        : 2
[VERBOSE @ fpga_common/src/lms.c:1681]   nint     : 129
[VERBOSE @ fpga_common/src/lms.c:1682]   nfrac    : 3145728
[VERBOSE @ fpga_common/src/lms.c:1683]   freqsel  : 0x2c
[VERBOSE @ fpga_common/src/lms.c:1684]   reference: 38400000
[VERBOSE @ fpga_common/src/lms.c:1685]   freq     : 2484000000
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:2015] bladerf1_schedule_retune: bork
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:987] nios_retune: channel=RX1 timestamp=0 nint=129 nfrac=3145728
				freqsel=0x2c vcocap=0x23 low_band=0 quick_tune=0
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1000] nios_retune: duration=0 vcocap=25 resp_flags=3
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1004] RX1 retune operation: vcocap=37, duration=0
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:504] nios_lms6_read: Read 0x80 from addr 0x71
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:516] nios_lms6_write: Wrote 0x83 to addr 0x71
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:504] nios_lms6_read: Read 0x80 from addr 0x72
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:516] nios_lms6_write: Wrote 0x93 to addr 0x72
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:1887] Set AGC DC offset cal (I, Q) to: Max (0, 4)  Mid (20, 5) Min (34, 9)
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:1891] Set RX DC offset cal (I, Q) to: (96, 608)
[VERBOSE @ host/libraries/libbladeRF/src/driver/dac161s055.c:46] dac161s055_write: Wrote 0x9004
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:392] nios_config_read: Read 0x0018002f
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:403] nios_config_write: Wrote 0x0018102f
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:411] Fetched register values from RX DC cal table.
(nios_lms6_read and nios_lms6_write flood omitted)
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:1809] Setting RX1 frequency to 2484000000
[VERBOSE @ fpga_common/src/lms.c:359] VCOCAP estimate: 35
[VERBOSE @ fpga_common/src/lms.c:1679] ---- Frequency ----
[VERBOSE @ fpga_common/src/lms.c:1680]   x        : 2
[VERBOSE @ fpga_common/src/lms.c:1681]   nint     : 129
[VERBOSE @ fpga_common/src/lms.c:1682]   nfrac    : 3145728
[VERBOSE @ fpga_common/src/lms.c:1683]   freqsel  : 0x2c
[VERBOSE @ fpga_common/src/lms.c:1684]   reference: 38400000
[VERBOSE @ fpga_common/src/lms.c:1685]   freq     : 2484000000
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:2015] bladerf1_schedule_retune: bork
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:987] nios_retune: channel=RX1 timestamp=0 nint=129 nfrac=3145728
				freqsel=0x2c vcocap=0x23 low_band=0 quick_tune=0
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1000] nios_retune: duration=0 vcocap=ff resp_flags=0
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1007] RX1 operation duration: 0
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1012] FPGA tuning reported failure.
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/usb.c:131] Reporting no FPGA present - BLADERF_FORCE_NO_FPGA_PRESENT is set.
Failed to open device (first available): An unexpected error occurred

real	0m2.597s
user	0m0.009s
sys	0m0.022s
((2018.08)) rtucker@rho:~/dev/bladeRF/host/build$ 

```
(note: omitted most si5338 and lms setup read/writes)

rtucker, 2018-09-17:
possibly #395 coming back to roost

rtucker, 2018-09-18:
As a workaround, the v0.8.0 FPGA image for the x115 seems to work okay: https://www.nuand.com/fpga_images/

My current theory is that the SPI controller/arbiter are in the wrong clock domain, resulting in unexpected (and incorrectly-defined) clock domain crossings.  Some builds will be fine, others will fail.

closed 2018-09-24 by rtucker

marwon34, 2019-01-26:
[ERROR @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:324] bladerf2_open: dev->backend->get_device_speed(dev, &usb_speed) failed: Operation not supported
Failed to open device (first available): Operation not supported

marwon34, 2019-01-27:
[ERROR @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:324] bladerf2_open: dev->backend->get_device_speed(dev, &usb_speed) failed: Operation not supported
Failed to open device (first available): Operation not supported
What is the problem here?

robertghilduta, 2019-01-27:
Please note this issue is closed, the issue only affected bladeRF 1 devices. bladeRF 2.0 micro devices do not use agc RX tables generated by bladeRF-cli, so 'cal table agc rx' is not necessary for the bladeRF 2.0 micro.

robertghilduta, 2019-01-27:
@marwon34 this issue is closed and resolved. Please email bladeRF@nuand.com if you have any questions or concerns.

marwon34, 2019-01-29:
مشكلتي هي [ERROR] أخفق التهيئة بعد التشغيل مرتين على الأقل.
[خطأ] حدثت مشكلة أثناء إنشاء حزمة pyqwt5: فشل التكوين [خطأ] خطأ في تثبيت
حزمة pyqwt5. إجهاض.

‫في الاثنين، 28 يناير 2019 في 12:10 ص تمت كتابة ما يلي بواسطة
‪robertghilduta‬‏ <‪notifications@github.com‬‏>:‬

PR #642 (2018-09-24): hdl: bladerf1: move AGC SPI driver into 80 MHz clock domain
This ensures that the SPI controllers have a consistent clock to work with, instead of rx_clock (which depends on the configured ADC sample rate). This ensures the SPI clock will always be 20 MHz, and that everything is passing between known clock domains.

Also registers the gain change request lines going into bladerf_agc_lms_drv to improve timing margin, and adds a set_output_delay timing constraint on lms_sclk.

Fixes #640