Nuand/bladeRF #641: bladeRF-cli: Windows: could not open device due to insufficient permissions

bglod, 2018-09-18:
Reported via e-mail. Reproducible in Windows 7 VM. The following occurs even when running as Administrator.

```
[INFO @ host/libraries/libbladeRF/src/helpers/version.c:82] Firmware version (v2.3.0) is newer than entries in libbladeRF's compatibility table. Please update libbladeRF if problems arise.
[INFO @ host/libraries/libbladeRF/src/helpers/version.c:106] FPGA version (v0.8.0) is newer than entries in libbladeRF's compatibility table. Please update libbladeRF if problems arise.
bladeRF> probe
[WARNING @ host/libraries/libbladeRF/src/backend/usb/libusb.c:309] Found a blade
RF via VID/PID, but could not open it due to insufficient permissions.

  probe: No devices are available. If one is attached, ensure it
         is not in use by another program and that the current
         user has permission to access it.

bladeRF> info

  Board:                    bladerf1
  Serial #:                 5ee7e0b4d33cfb087ea82121692c3565
  VCTCXO DAC calibration:   0x8e11
  FPGA size:                40 KLE
  FPGA loaded:              no
  USB bus:                  1
  USB address:              4
  USB speed:                SuperSpeed
  Backend:                  libusb
  Instance:                 0

bladeRF> rx config file=../tmp.csv format=csv n=1024
bladeRF> rx start

  Error: Insufficient permissions for the requested operation

bladeRF> quit
```

rtucker, 2018-09-20:
I am able to reproduce both of these on Windows 10.

For the `rx start` failure, check that `../tmp.csv` is actually writable... there are a number of directories which are protected from modification (even by Administrator), and that error message will bubble up from any permissions-related error.

For the `probe` failure... my best guess is that the device is already open, and that's why it can't get its device descriptor.  This can be verified with:

1. Unplug the bladeRF
2. Open `bladeRF-cli -i`, which will report no devices available.
3. `probe` should return no devices available (which is true) without the insufficient-permissions warning.
4. Plug in the bladeRF
5. `probe` again.  It should work fine.
6. `open` should work fine
7. `probe` again... this time it fails on permissions

bglod, 2018-09-20:
For the `rx start` -- agreed, silly me. :+1: 

Customer reports permission error begins in release 2016.01.

Running `bladeRF-cli.exe -v verbose -e probe` displays the insufficient permissions warning. You're probably correct here in that it is opening the first available device before running the `probe` command. I will note that this warning does not occur on Linux, but could be differences in libusb implementations.

rtucker, 2018-09-20:
That would make sense, since handling `LIBUSB_ERROR_ACCESS` was added between 2015.07 and 2016.01-rc1.  (prior to that, it would map to a `BLADERF_ERR_UNEXPECTED` and I think it would stop the probe immediately)

I am 99% sure this is a Windows-only error.  I am 90% sure it is harmless.  In theory, though, we ought to be able to determine that we already have the device open and _that's_ why we can't access it......

bglod, 2018-09-20:
Agreed.

rtucker, 2018-09-20:
PR #648 fixes this in a way that, somehow, isn't terrible.  There's a link to a Windows installer in the comments of the PR.

PR #648 (2018-09-24): host: improve handling of LIBUSB_ERROR_ACCESS during probe
In the Windows world, libusb_open() fails if the device is already open.  This results in a misleading error if a user does, say, `bladeRF-cli -e probe`.

As a workaround, lusb_probe now passes back the data it does have (backend, USB bus, USB address), and sets the serial number to "ANY".  bladeRF-cli's probe command now looks for this and handles it.

Fixes #641
closed 2018-09-24 by rtucker