Nuand/bladeRF #669: nanosleep.c build fails on windows clang-cl

rtucker, 2018-10-02:
This commit is really strange. I mean, this:

```c
#include "nanosleep.h"
#include <pthread.h>
```
causes forward implicit definition of `struct timespec` in `"nanosleep.h"`. And then when `<pthread.h>` gets included, the real (and conflicting) definition from *Pthreads-W32* gets pulled in. AFAIK, MSVC allows this., but the fantastic MSVC-replacement `clang-cl`, does not:
```
In file included from common/src/windows/nanosleep.c:29:
./common/include/windows\nanosleep.h(36,28):  warning: declaration of 'struct timespec' will not be visible outside of this function
      [-Wvisibility]
int nanosleep(const struct timespec *req, struct timespec *rem);
                           ^
common/src/windows/nanosleep.c(33,5):  error: conflicting types for 'nanosleep'
int nanosleep(const struct timespec *req, struct timespec *rem)
    ^
./common/include/windows\nanosleep.h(36,5):  note: previous declaration is here
int nanosleep(const struct timespec *req, struct timespec *rem);
    ^
1 warning and 1 error generated.
```

A simple fix that works with both MSVC/clang-cl is simply this:
```diff
--- a/host/common/src/windows/nanosleep.c 2018-09-30 06:48:02
+++ b/host/common/src/windows/nanosleep.c 2018-10-01 11:51:19
@@ -26,9 +26,9 @@
 #error "This file is intended for use with WIN32 systems only."
 #endif

-#include "nanosleep.h"
 #include <pthread.h>
 #include <windows.h>
+#include "nanosleep.h"

```

_Originally posted by @gvanem in https://github.com/Nuand/bladeRF/commit/8fdfa6abcd8cdd6b6751dcbf8de350fd815a525c#commitcomment-30716361_

rtucker, 2018-10-02:
@gvanem hmm, yep, that's a problem

how does 021c00d9 look as an alternative? that's similar to how we do it in clock_realtime.h.

(also added 378baf49 with additional cleanup)

gvanem, 2018-10-02:
> how does 021c00d look as an alternative?

Work fine.

Speaking of `struct timespec`. The instruction in `host/README.md` to add `#define HAVE_STRUCT_TIMESPEC`  to Pthreads-W32 seems fishy. Cmake should IMHO have a test for the presence `struct timespec` (in recent Windows-SDKs). I do not know Cmake and do not use it (otherwise I'd create a PR for that).

rtucker, 2018-10-02:
I've kinda been hoping a new pthreads-win32 would be released with that fix at some point, but that seems increasingly unlikely.  But, there are a few things CMake can do which will make that and other things easier, e.g. https://cmake.org/cmake/help/latest/module/ExternalProject.html.

I'm gonna open an issue on this and hopefully get a chance to revisit it at some point soon.  Thanks :)

PR #671 (2018-10-08): host: common: include pthread.h in nanosleep.h (#669)
default header sort on nanosleep.c puts nanosleep.h above pthread.h, which creates an implicit definition of struct timespec.

Fixes #669
commit c9a2230c (2018-10-08): host: common: include pthread.h in nanosleep.h (#669) (#671)

* host: common: include pthread.h in nanosleep.h (#669)

default header sort on nanosleep.c puts nanosleep.h above
pthread.h, which creates an implicit definition of struct
timespec.

Fixes #669

* host: common/nanosleep: cleanup includes
closed 2018-10-08 by rtucker