Nuand/bladeRF #674: MATLAB & Linux bindings error when running bladeRF_rx_gui

tichkadzinga, 2018-10-03:
I compiled the MATLAB Linux bindings and get an error when I run bladeRF_rx_gui 
My setup us as follows
```
bladeRF> version

  bladeRF-cli version:        1.6.1-git-f38dde8-ppabionic
  libbladeRF version:         2.0.2-git-f38dde8-ppabionic

  Firmware version:           2.2.0-git-3d38fac2
  FPGA version:               0.8.0
```
The error is:
```
Subscripted assignment between dissimilar structures.

Error in bladeRF.devices (line 181)
                    devs(x+1) = ptr.Value;

Error in bladeRF_rx_gui>devicelist_CreateFcn (line 800)
    devs = bladeRF.devices;

Error in gui_mainfcn (line 95)
        feval(varargin{:});

Error in bladeRF_rx_gui (line 50)
        gui_mainfcn(gui_State, varargin{:});

Error in
matlab.graphics.internal.figfile.FigFile/read>@(hObject,eventdata)bladeRF_rx_gui('devicelist_CreateFcn',hObject,eventdata,guidata(hObject)) 
Error using calllib
Parameter must be scalar.

Error in bladeRF_XCVR/set.samplerate (line 86)
            [status, ~, ~, actual] = calllib('libbladeRF', 'bladerf_set_rational_sample_rate', obj.bladerf.device,
            obj.module, rate, rate);

Error in bladeRF_XCVR (line 504)
            obj.samplerate = 3e6;

Error in bladeRF (line 537)
            obj.rx = bladeRF_XCVR(obj, 'RX', xb);

Error in bladeRF_rx_gui>bladeRF_fft_OpeningFcn (line 358)
    handles.bladerf = bladeRF('*:instance=0');

Error in gui_mainfcn (line 220)
    feval(gui_State.gui_OpeningFcn, gui_hFigure, [], guidata(gui_hFigure), varargin{:});

Error in bladeRF_rx_gui (line 50)
        gui_mainfcn(gui_State, varargin{:});
 ```

tecandrew, 2019-08-29:
was this ever resolved? running into the same issue. build_thunk() is also giving a bunch of random errors and warnings.

Output of `dev=bladeRF()'
```
Error using calllib
Parameter must be scalar.

Error in bladeRF_XCVR/set.samplerate (line 86)
            [status, ~, ~, actual] = calllib('libbladeRF', 'bladerf_set_rational_sample_rate', obj.bladerf.device, obj.module, rate,
            rate);

Error in bladeRF_XCVR (line 504)
            obj.samplerate = 3e6;

Error in bladeRF (line 537)
            obj.rx = bladeRF_XCVR(obj, 'RX', xb);
```

Output of `bladeRF.build_thunk()`:
```
Warning: The data type 'FcnPtr' used by function bladerf_init_stream does not exist. 
> In loadlibrary
  In bladeRF.build_thunk (line 269) 
Warning: The data type 'voidPtrPtrPtr' used by function bladerf_init_stream does not exist. 
> In loadlibrary
  In bladeRF.build_thunk (line 269) 
Warning: Encountered the following warnings while loading libbladeRF:

No match found for enum value expression sizeof in ( 1024 / ( 8 * sizeof ( unsigned long int ))) expression ignored.
Found on line 596 of input from line 30 of file /usr/include/x86_64-linux-gnu/bits/sigset.h

No match found for enum value expression int in  1024 / ( 8 * ( int ) sizeof ( __fd_mask )) expression ignored.
Found on line 645 of input from line 75 of file /usr/include/x86_64-linux-gnu/sys/select.h

Type 'intunionpthread_attr_t' was not found.  Defaulting to type error.

Found on line 724 of input from line 69 of file /usr/include/x86_64-linux-gnu/bits/pthreadtypes.h

Failed to parse type 'union { struct __pthread_mutex_s { int __lock ; unsigned int __count ; int __owner ; unsigned int __nusers ; int
__kind ; short __spins ; short __elision ; __pthread_list_t __list ; } __data ; char *__size; long int __align ; } pthread_mutex_t'
original input 'union { struct __pthread_mutex_s { int __lock ; unsigned int __count ; int __owner ; unsigned int __nusers ; int __kind ;
short __spins ; short __elision ; __pthread_list_t __list ; } __data ; char __size [ 40 ]; long int __align ; } pthread_mutex_t'
Found on line 757 of input from line 128 of file /usr/include/x86_64-linux-gnu/bits/pthreadtypes.h

Failed to parse type 'union { char *__size; int __align ; } pthread_mutexattr_t' original input 'union { char __size [ 4 ]; int __align ;
} pthread_mutexattr_t'
Found on line 763 of input from line 134 of file /usr/include/x86_64-linux-gnu/bits/pthreadtypes.h

Failed to parse type 'union { struct { int __lock ; unsigned int __futex ; __extension__ unsigned long long int __total_seq ;
__extension__ unsigned long long int __wakeup_seq ; __extension__ unsigned long long int __woken_seq ; void * __mutex ; unsigned int
__nwaiters ; unsigned int __broadcast_seq ; } __data ; char *__size; __extension__ long long int __align ; } pthread_cond_t' original
input 'union { struct { int __lock ; unsigned int __futex ; __extension__ unsigned long long int __total_seq ; __extension__ unsigned long
long int __wakeup_seq ; __extension__ unsigned long long int __woken_seq ; void * __mutex ; unsigned int __nwaiters ; unsigned int
__broadcast_seq ; } __data ; char __size [ 48 ]; __extension__ long long int __align ; } pthread_cond_t'
Found on line 783 of input from line 154 of file /usr/include/x86_64-linux-gnu/bits/pthreadtypes.h

Failed to parse type 'union { char *__size; int __align ; } pthread_condattr_t' original input 'union { char __size [ 4 ]; int __align ; }
pthread_condattr_t'
Found on line 789 of input from line 160 of file /usr/include/x86_64-linux-gnu/bits/pthreadtypes.h

Failed to parse type 'union { struct { int __lock ; unsigned int __nr_readers ; unsigned int __readers_wakeup ; unsigned int
__writer_wakeup ; unsigned int __nr_readers_queued ; unsigned int __nr_writers_queued ; int __writer ; int __shared ; signed char
__rwelision ; unsigned char *__pad1; unsigned long int __pad2 ; unsigned int __flags ; } __data ; char *__size; long int __align ; }
pthread_rwlock_t' original input 'union { struct { int __lock ; unsigned int __nr_readers ; unsigned int __readers_wakeup ; unsigned int
__writer_wakeup ; unsigned int __nr_readers_queued ; unsigned int __nr_writers_queued ; int __writer ; int __shared ; signed char
__rwelision ; unsigned char __pad1 [ 7 ]; unsigned long int __pad2 ; unsigned int __flags ; } __data ; char __size [ 56 ]; long int
__align ; } pthread_rwlock_t'
Found on line 833 of input from line 222 of file /usr/include/x86_64-linux-gnu/bits/pthreadtypes.h

Failed to parse type 'union { char *__size; long int __align ; } pthread_rwlockattr_t' original input 'union { char __size [ 8 ]; long int
__align ; } pthread_rwlockattr_t'
Found on line 839 of input from line 228 of file /usr/include/x86_64-linux-gnu/bits/pthreadtypes.h

Failed to parse type 'union { char *__size; long int __align ; } pthread_barrier_t' original input 'union { char __size [ 32 ]; long int
__align ; } pthread_barrier_t'
Found on line 854 of input from line 243 of file /usr/include/x86_64-linux-gnu/bits/pthreadtypes.h

Failed to parse type 'union { char *__size; int __align ; } pthread_barrierattr_t' original input 'union { char __size [ 4 ]; int __align
; } pthread_barrierattr_t'
Found on line 860 of input from line 249 of file /usr/include/x86_64-linux-gnu/bits/pthreadtypes.h

Type 'unsignedlonglong__extension__' was not found.  Defaulting to type error.

Found on line 971 of input from line 420 of file /usr/include/stdlib.h

Type 'bladerf_quick_tunePtr' was not found.  Defaulting to type voidPtr.

Found on line 1923 of input from line 1865 of file /home/atec/_workspaces/brf-matlab/libbladerf/libbladeRF.h

Type 'bladerf_quick_tunePtr' was not found.  Defaulting to type voidPtr.

Found on line 1932 of input from line 1906 of file /home/atec/_workspaces/brf-matlab/libbladerf/libbladeRF.h
```

Ubuntu 16.04
4.15.0-58-lowlatency kernel
MATLAB 2019a

tecandrew, 2019-09-03:
also doesn't work for Windows (out-of-the-box installer). `libbladeRF_thunk_pcwin64.dll` was already generated and can make some libbladeRF calls.

Running `bladeRF_tx_tone.m` from **/bladeRF/matlab/test** directory
```matlab
>> bladeRF_tx_tone
Creating 5.000000 seconds of a 250000.000000 Hz tone.
Warning: libbladeRF error (-8) in bladerf_set_vga2(): Operation not supported 
> In bladeRF.check_status (line 114)
  In bladeRF_XCVR/set.vga2 (line 324)
  In bladeRF_tx_tone (line 61) 
Warning: libbladeRF error (-8) in bladerf_set_vga1(): Operation not supported 
> In bladeRF.check_status (line 114)
  In bladeRF_XCVR/set.vga1 (line 292)
  In bladeRF_tx_tone (line 62) 
Running with the following settings:

val =

    'BLADERF_CHANNEL_TX1'

  bladeRF_XCVR with properties:

          config: [1×1 bladeRF_StreamConfig]
     corrections: [1×1 bladeRF_IQCorr]
             agc: 'auto'
            gain: 56
         biastee: 0
      samplerate: 5000000
       frequency: 914999998
       bandwidth: 1500000
            vga1: 0
            vga2: 0
             lna: 0
    xb200_filter: 'N/A'
         channel: 'BLADERF_CHANNEL_TX1'
         running: 0

  bladeRF_StreamConfig with properties:

      num_buffers: 64
      buffer_size: 16384
    num_transfers: 16
       timeout_ms: 5000

Waiting for transmission to complete.
Error using calllib
There is no enumerated value named 'BLADERF_CHANNEL_TX1'.

Error in bladeRF_XCVR/get.timestamp (line 448)
            [status, ~, val] = calllib('libbladeRF', 'bladerf_get_timestamp', obj.bladerf.device,
            obj.module, val);

Error in bladeRF_tx_tone (line 126)
t = dev.tx.timestamp + max_buffered;
```

---
Windows 10 Pro
MATLAB 2019a

> was this ever resolved? running into the same issue. build_thunk() is also giving a bunch of random errors and warnings.
>
> Output of `dev=bladeRF()`
>
> ```
> Error using calllib
> Parameter must be [...]

rghilduta, 2019-09-04:
Please take a look at the latest master, the bladeRF_tx_tone issue should not appear.

What compiler it it using on Linux? It appears that cstdint types may not be declared.

tecandrew, 2019-09-04:
`gcc (Ubuntu 5.4.0-6ubuntu1~16.04.11) 5.4.0 20160609`

just gave it another shot with the latest commit e9ef4cad79fda4db9358e0eacf13030c2aaf6d13. still running into the same problem after running

```matlab
>>bladeRF.build_thunk()
```

:disappointed: 

Output:
```matlab
Warning: The data type 'FcnPtr' used by function bladerf_init_stream does not exist. 
> In loadlibrary
  In bladeRF.build_thunk (line 269) 
Warning: The data type 'voidPtrPtrPtr' used by function bladerf_init_stream does not exist. 
> In loadlibrary
  In bladeRF.build_thunk (line 269) 
Warning: Encountered the following warnings while loading libbladeRF:

No match found for enum value expression sizeof in ( 1024 / ( 8 * sizeof ( unsigned long int ))) expression ignored.
Found on line 596 of input from line 30 of file /usr/include/x86_64-linux-gnu/bits/sigset.h

No match found for enum value expression int in  1024 / ( 8 * ( int ) sizeof ( __fd_mask )) expression ignored.
Found on line 645 of input from line 75 of file /usr/include/x86_64-linux-gnu/sys/select.h

Type 'intunionpthread_attr_t' was not found.  Defaulting to type error.

Found on line 724 of input from line 69 of file /usr/include/x86_64-linux-gnu/bits/pthreadtypes.h

Failed to parse type 'union { struct __pthread_mutex_s { int __lock ; unsigned int __count ; int __owner ; unsigned int __nusers ; int __kind ; short
__spins ; short __elision ; __pthread_list_t __list ; } __data ; char *__size; long int __align ; } pthread_mutex_t' original input 'union { struct
__pthread_mutex_s { int __lock ; unsigned int __count ; int __owner ; unsigned int __nusers ; int __kind ; short __spins ; short __elision ;
__pthread_list_t __list ; } __data ; char __size [ 40 ]; long int __align ; } pthread_mutex_t'
Found on line 757 of input from line 128 of file /usr/include/x86_64-linux-gnu/bits/pthreadtypes.h

Failed to parse type 'union { char *__size; int __align ; } pthread_mutexattr_t' original input 'union { char __size [ 4 ]; int __align ; }
pthread_mutexattr_t'
Found on line 763 of input from line 134 of file /usr/include/x86_64-linux-gnu/bits/pthreadtypes.h

Failed to parse type 'union { struct { int __lock ; unsigned int __futex ; __extension__ unsigned long long int __total_seq ; __extension__ unsigned
long long int __wakeup_seq ; __extension__ unsigned long long int __woken_seq ; void * __mutex ; unsigned int __nwaiters ; unsigned int __broadcast_seq
; } __data ; char *__size; __extension__ long long int __align ; } pthread_cond_t' original input 'union { struct { int __lock ; unsigned int __futex ;
__extension__ unsigned long long int __total_seq ; __extension__ unsigned long long int __wakeup_seq ; __extension__ unsigned long long int __woken_seq
; void * __mutex ; unsigned int __nwaiters ; unsigned int __broadcast_seq ; } __data ; char __size [ 48 ]; __extension__ long long int __align ; }
pthread_cond_t'
Found on line 783 of input from line 154 of file /usr/include/x86_64-linux-gnu/bits/pthreadtypes.h

Failed to parse type 'union { char *__size; int __align ; } pthread_condattr_t' original input 'union { char __size [ 4 ]; int __align ; }
pthread_condattr_t'
Found on line 789 of input from line 160 of file /usr/include/x86_64-linux-gnu/bits/pthreadtypes.h

Failed to parse type 'union { struct { int __lock ; unsigned int __nr_readers ; unsigned int __readers_wakeup ; unsigned int __writer_wakeup ; unsigned
int __nr_readers_queued ; unsigned int __nr_writers_queued ; int __writer ; int __shared ; signed char __rwelision ; unsigned char *__pad1; unsigned
long int __pad2 ; unsigned int __flags ; } __data ; char *__size; long int __align ; } pthread_rwlock_t' original input 'union { struct { int __lock ;
unsigned int __nr_readers ; unsigned int __readers_wakeup ; unsigned int __writer_wakeup ; unsigned int __nr_readers_queued ; unsigned int
__nr_writers_queued ; int __writer ; int __shared ; signed char __rwelision ; unsigned char __pad1 [ 7 ]; unsigned long int __pad2 ; unsigned int
__flags ; } __data ; char __size [ 56 ]; long int __align ; } pthread_rwlock_t'
Found on line 833 of input from line 222 of file /usr/include/x86_64-linux-gnu/bits/pthreadtypes.h

Failed to parse type 'union { char *__size; long int __align ; } pthread_rwlockattr_t' original input 'union { char __size [ 8 ]; long int __align ; }
pthread_rwlockattr_t'
Found on line 839 of input from line 228 of file /usr/include/x86_64-linux-gnu/bits/pthreadtypes.h

Failed to parse type 'union { char *__size; long int __align ; } pthread_barrier_t' original input 'union { char __size [ 32 ]; long int __align ; }
pthread_barrier_t'
Found on line 854 of input from line 243 of file /usr/include/x86_64-linux-gnu/bits/pthreadtypes.h

Failed to parse type 'union { char *__size; int __align ; } pthread_barrierattr_t' original input 'union { char __size [ 4 ]; int __align ; }
pthread_barrierattr_t'
[... 8 lines cut ...]
Found on line 1923 of input from line 1865 of file /home/atec/_workspaces/bladeRF/host/libraries/libbladeRF_bindings/matlab/libbladeRF.h

Type 'bladerf_quick_tunePtr' was not found.  Defaulting to type voidPtr.

Found on line 1932 of input from line 1906 of file /home/atec/_workspaces/bladeRF/host/libraries/libbladeRF_bindings/matlab/libbladeRF.h
```

`bladeRF_tx_tone.m` Output:
```c
Creating 5.000000 seconds of a 250000.000000 Hz tone.
Error using calllib
Parameter must be scalar.

Error in bladeRF_XCVR/set.samplerate (line 91)
            [status, ~, ~, actual] = calllib('libbladeRF', 'bladerf_set_rational_sample_rate', obj.bladerf.device, obj.module, rate, rate);

Error in bladeRF_XCVR (line 545)
            obj.samplerate = 3e6;

Error in bladeRF (line 537)
            obj.rx = bladeRF_XCVR(obj, 'RX', xb);

Error in bladeRF_tx_tone (line 56)
dev = bladeRF(devstr);
```

rghilduta, 2019-09-05:
Please ensure you look closely at the instructions for bladeRF.build_thunk(), and follow the required step of adding the #define it suggests to libbladeRF.h . "Must be scalar" can occur if libbladeRF_proto.m isn't loaded, try restarting the program.

Alternatively, please take a look at the recent master, it should include more insightful messages.

jordongriffiths, 2020-02-19:
Did you ever get any further than this? I'm also using the thunk which was already generated and provided with the installer; this is giving me the same issue as you when trying to run bladeRF_tx_tone.m (Error in bladeRF_XCVR/get.timestamp (line 448)). 

The example is trying to check the value of dev.tx.timestamp to see if the entirety of the buffer has been transmitted but dev.tx.timestamp does not exist - where can I see the value of timestamp? Has this moved out of dev.tx or should I be doing something different to what the example is doing to check if all data has been transmitted before disposing of the bladeRF object?

> also doesn't work for Windows (out-of-the-box installer). `libbladeRF_thunk_pcwin64.dll` was already generated and can make some libbladeRF calls.
>
> Running `bladeRF_tx_tone.m` from [...]

BushraAnsarii, 2020-08-24:
Getting the below error, while running bladeRF.m in MATLAB in linux:
While it is running well in windows. 
Any solution for this error????
-----------------------------------------------------------------------
Error using calllib
Parameter must be scalar.

Error in bladeRF_XCVR/set.samplerate (line 86)
            [status, ~, ~, actual] = calllib('libbladeRF', 'bladerf_set_rational_sample_rate', obj.bladerf.device,
            obj.module, rate, rate);

Error in bladeRF_XCVR (line 504)
            obj.samplerate = 3e6;

Error in bladeRF (line 537)
            obj.rx = bladeRF_XCVR(obj, 'RX', xb);

rghilduta, 2020-08-25:
Please ensure GCC version 9 or later are being used when compiling from scratch. The libbladeRF header files should be in the same directory as where bladeRF.m is and runs from ( host/libraries/libbladeRF_bindings/matlab )

If installing on Windows, ensure previous installations of bladeRF have been properly cleaned out. Also, ensure the path to the MEX files is in the MATLAB path (addpath).

If you issues persist, please create another thread, or email technical support at bladeRF@nuand.com .