Nuand/bladeRF #676: can't track satellite

matteocutugno, 2018-10-11:
Hi
i tried to run gnss-sdr for real time acquisition.
I have a nuand bladerf x40 with u-blox active antenna.
Gnss-sdr can't track satellite and reports loss of lock in every channel continuously.
I tried a lot of different conf files and now i begin to think that is an antenna issue.
On gnss sdr site they say that:
"you will need an active antenna (that is, an antenna with an integrated low noise amplifier delivering a gain in the band of interest greater than 20 dB, with a noise figure below 2 dB), and you will need to feed it."
U-blox antenna is an active antenna with Built in low noise amplifier with 29 dB gain and 0.9dB noise figure so i think is suitable. 
The problem is that i don't know if i have to feed it externally (maybe with a bias-tee) or it is auto feed by front-end.
Thank you in advance for your answer.
MC

odrisci, 2018-10-14:
You need an external bias tee with the x40. The newer micro models have built in bias tees

matteocutugno, 2018-10-14:
Thank u for the answer.
An external bias tee in which I can apply +5V it's suitable?
Like this?  https://www.tindie.com/products/gpio/bias-tee-operates-from-10mhz-to-7000mhz/

odrisci, 2018-10-14:
I've used that one with a u-blox antenna and it worked fine for me

rtucker, 2018-10-30:
Closing issue (correct answer from @matteocutugno )

closed 2018-10-30 by rtucker