Nuand/bladeRF #679: host loading of FPGA fails w/ USB2

rtucker, 2018-10-22:
Doing `bladeRF-cli -l ~/Downloads/hostedxA4-latest.rbf -v verbose -e version` on an xA4 without an FPGA image loaded results in some sort of situation where the FPGA version reply is corrupted.

actual (usb 2.0):
```
(master) rtucker@rho:~/dev/bladeRF/build/host$ output/bladeRF-cli -l ~/Downloads/hostedxA4-latest.rbf -v verbose -e version
Deferring device initialization due to requested FPGA or firmware actions.
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:647] Using libusb version: 1.0.21.11156
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:524] Found a bladeRF (idx=12)
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/libusb.c:200] Bus 003 Device 018: Nuand bladeRF 2.0, serial X
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:586] USB port reset succeeded for bladeRF X
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 0
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1381] Read Firmware version: 2.3.1
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1387] Capability mask before FPGA load: 0x0000003f00000000
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:336] Found SPI flash manufacturer: WINBOND.
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:339] Found SPI flash device: W25Q32JV (32 Mbit).
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:382] SPI flash total size = 32 Mbit
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:383] SPI flash page size = 256 bytes
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:384] SPI flash erase block size = 65536 bytes
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:385] SPI flash number of pages = 16384
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:386] SPI flash number of erase blocks = 64 pages
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 2
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/usb.c:131] Reporting no FPGA present - BLADERF_FORCE_NO_FPGA_PRESENT is set.
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 3
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1498] Skipping FPGA configuration and initialization - BLADERF_FORCE_NO_FPGA_PRESENT is set.
Loading fpga...
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 3
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 1
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1113] Read FPGA version: 0.1.770
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1120] Capability mask after FPGA load: 0x0000003f00000000
[DEBUG @ host/libraries/libbladeRF/src/helpers/version.c:152] version_check is missing FPGA version compat table entry?
bladeRF-cli: /home/rtucker/dev/bladeRF/host/libraries/libbladeRF/src/helpers/version.c:153: version_check: Assertion `!"BUG!"' failed.
Aborted (core dumped)
```

expected (usb 3.0):
```
(master) rtucker@rho:~/dev/bladeRF/build/host$ output/bladeRF-cli -l ~/Downloads/hostedxA4-latest.rbf -v verbose -e version
Deferring device initialization due to requested FPGA or firmware actions.
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:647] Using libusb version: 1.0.21.11156
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:524] Found a bladeRF (idx=9)
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/libusb.c:200] Bus 004 Device 080: Nuand bladeRF 2.0, serial X
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:586] USB port reset succeeded for bladeRF X
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 0
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1381] Read Firmware version: 2.3.1
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1387] Capability mask before FPGA load: 0x0000003f00000000
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:336] Found SPI flash manufacturer: WINBOND.
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:339] Found SPI flash device: W25Q32JV (32 Mbit).
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:382] SPI flash total size = 32 Mbit
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:383] SPI flash page size = 256 bytes
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:384] SPI flash erase block size = 65536 bytes
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:385] SPI flash number of pages = 16384
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:386] SPI flash number of erase blocks = 64 pages
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 2
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/usb.c:131] Reporting no FPGA present - BLADERF_FORCE_NO_FPGA_PRESENT is set.
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 3
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1498] Skipping FPGA configuration and initialization - BLADERF_FORCE_NO_FPGA_PRESENT is set.
Loading fpga...
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 3
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 1
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1113] Read FPGA version: 0.8.0
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1120] Capability mask after FPGA load: 0x0000003f00000000
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:550] nios_ina219_write: Wrote 0x8000 to addr 0x00
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:537] nios_ina219_read: Read 0x399f from addr 0x00
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:550] nios_ina219_write: Wrote 0x019f to addr 0x00
[DEBUG @ host/libraries/libbladeRF/src/driver/ina219.c:70] Configuration register: 0x019f
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:550] nios_ina219_write: Wrote 0xa000 to addr 0x05
[DEBUG @ host/libraries/libbladeRF/src/driver/ina219.c:82] Calibration register: 0xa000
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1181] bladerf2_initialize: ad9361_init starting
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1183] bladerf2_initialize: ad9361_init complete, status = 0
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1006] _set_tx_mute: muting TX1 (cached: 10000)
[... 16 lines cut ...]
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:5095] reg 2 gets 0x001f8092
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:723] nios_adf400x_write: Wrote 0x001f8090 to addr 0x02
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:392] nios_config_read: Read 0x00000801
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:403] nios_config_write: Wrote 0x00000001
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:685] nios_ad56x1_vctcxo_trim_dac_read: Read 0xc000
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:4201] trim DAC is disabled
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:685] nios_ad56x1_vctcxo_trim_dac_read: Read 0xc000
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:4247] restoring old trim DAC value: 0x1ec1
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:697] nios_ad56x1_vctcxo_trim_dac_write: Wrote 0x1ec1
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1293] bladerf2_initialize: complete
Successfully loaded FPGA bitstream!

  bladeRF-cli version:        1.6.1-git-ff1650bb
  libbladeRF version:         2.0.2-git-ff1650bb

  Firmware version:           2.3.1
  FPGA version:               0.8.0 (configured by USB host)

[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1076] RX1 trigger read value 0x08
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1076] RX1 trigger read value 0x08
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1116] RX1 trigger write value 0x00
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1076] TX1 trigger read value 0x08
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1076] TX1 trigger read value 0x08
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1116] TX1 trigger write value 0x00
```

rtucker, 2018-10-22:
USB 2.0:

```
NIOS II access request:
  0x4e, 0x81, 0x0c, 0xff, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
 
NIOS II access response:
  0x00, 0x4e, 0x81, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
 
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_legacy_access.c:221] nios_legacy_get_fpga_version: Read FPGA version word: [0x0000000c]=0x00000000
NIOS II access request:
  0x4e, 0x81, 0x0d, 0xff, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
 
NIOS II access response:
  0x00, 0x4e, 0x81, 0x01, 0x08, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
 
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_legacy_access.c:221] nios_legacy_get_fpga_version: Read FPGA version word: [0x0000000d]=0x00000001
NIOS II access request:
  0x4e, 0x81, 0x0e, 0xff, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
 
NIOS II access response:
  0x00, 0x4e, 0x81, 0x02, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
 
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_legacy_access.c:221] nios_legacy_get_fpga_version: Read FPGA version word: [0x0000000e]=0x00000002
NIOS II access request:
  0x4e, 0x81, 0x0f, 0xff, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
 
NIOS II access response:
  0x00, 0x4e, 0x81, 0x03, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
 
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_legacy_access.c:221] nios_legacy_get_fpga_version: Read FPGA version word: [0x0000000f]=0x00000003
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1113] Read FPGA version: 0.1.770
```

USB 3.0:
```
NIOS II access request:
  0x4e, 0x81, 0x0c, 0xff, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
 
NIOS II access response:
  0x4e, 0x81, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
 
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_legacy_access.c:221] nios_legacy_get_fpga_version: Read FPGA version word: [0x0000000c]=0x00000000
NIOS II access request:
  0x4e, 0x81, 0x0d, 0xff, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
 
NIOS II access response:
  0x4e, 0x81, 0x01, 0x08, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
 
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_legacy_access.c:221] nios_legacy_get_fpga_version: Read FPGA version word: [0x0000000d]=0x00000008
NIOS II access request:
  0x4e, 0x81, 0x0e, 0xff, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
 
NIOS II access response:
  0x4e, 0x81, 0x02, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
 
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_legacy_access.c:221] nios_legacy_get_fpga_version: Read FPGA version word: [0x0000000e]=0x00000000
NIOS II access request:
  0x4e, 0x81, 0x0f, 0xff, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
 
NIOS II access response:
  0x4e, 0x81, 0x03, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
 
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_legacy_access.c:221] nios_legacy_get_fpga_version: Read FPGA version word: [0x0000000f]=0x00000000
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1113] Read FPGA version: 0.8.0
```

rtucker, 2018-10-22:
oh neat...

```
NIOS II access response:
  0x00, 0x4e, 0x81, 0x01, 0x08, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
```

vs

```
NIOS II access response:
  0x4e, 0x81, 0x01, 0x08, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
```

yep that's a slip, ugh why

rtucker, 2018-10-22:
tried FW 2.1.0, no difference
also tried FPGA 0.7.3, no difference

buffalo960, 2018-10-22:
So there is an extra 0x00 byte slipping in the front when it is running under USB2?

rtucker, 2018-10-22:
> So there is an extra 0x00 byte slipping in the front when it is running under USB2?

Apparently!  Where it's coming from, I don't yet know...

rtucker, 2018-10-22:
it's showing up that way in the USB stream, so it's definitely somewhere between the Nios and the FX3, inclusive

i suspect there's exactly one byte of flotsam ending up at the start of the Nios->FX3 command stream under some conditions, somehow.  the probability of hitting this goes down if i use my bench dev A4 and/or i put the USB debugger in line, so it feels like a race condition :/

situations where i haven't been able to reproduce this:

- USB 3.0
- bladeRF x40/x115

I've been able to reproduce it with both Nios packet formats (legacy and new).

Note: usb_get_fpga_version is where the transition from USB_IF_CONFIG to USB_IF_RF_LINK happens, so the FPGA version check is literally the first thing going across that interface.

rtucker, 2018-10-24:
This is a particularly annoying one to track down, as it appears to be occurring within the first few moments of the FPGA being programmed and the Nios coming up.  Nowhere near enough time to get Signal Tap up and capturing :<

The approach that seems to work so far is to tie the Nios reset line to the pll_reset block's output, which brings it out of reset only when the sys_clock PLL is properly locked; and to do something similar w/ the select lines into the fx3_uart_* muxes, but with a short countdown.

This is okay, but if the countdown is too long, things end up in a different bad state because the bladerf_get_fpga_version() function is called very very soon after the FPGA is programmed, and... yes, race condition.

What I might end up doing is having the Nios assert a signal to indicate that it has booted and is ready to handle commands, and have that control the fx3_uart_* muxes (and also indicated back to the host?).  I think that's the surefire way to manage this race condition, and is probably also useful for headless applications.

(This will probably not be in the next tagged FPGA build, but hopefully one soon thereafter...)

bglod, 2018-10-25:
@rtucker would the power-up trigger feature of SignalTap be helpful for this?

rtucker, 2018-10-25:
@bglod oh heck, quite possibly!  didn't know that was in there, but it sounds like what i'm after.  thanks!

rtucker, 2018-10-26:
@buffalo960 Could you give this .rbf a try?

[hostedxA4-2b4efa8.rbf.zip](https://github.com/Nuand/bladeRF/files/2520356/hostedxA4-2b4efa8.rbf.zip)

This should avoid a troublesome race condition :)

buffalo960, 2018-10-26:
That seems to be working now! 
I do not have osmocom compiled properly yet but loading the FPGA works just fine and does not throw any errors anymore. I will try recompiling osmocom this weekend and see if it works.  Thank you!!! :smile: 

```
$ bladeRF-cli -l sandbox/hostedxA4-2b4efa8.rbf -v verbose
Deferring device initialization due to requested FPGA or firmware actions.
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:647] Using libusb version: 1.0.21.11156
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:524] Found a bladeRF (idx=2)
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/libusb.c:200] Bus 001 Device 015: Nuand bladeRF 2.0, serial 5f3bca769426485baefb0c097148dd76
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:586] USB port reset succeeded for bladeRF 5f3bca769426485baefb0c097148dd76
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 0
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1381] Read Firmware version: 2.2.0-git-3d38fac2
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1387] Capability mask before FPGA load: 0x0000000f00000000
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1461] FX3 firmware v2.2.0 does not support SPI flash ID. A firmware update is recommended in order to probe the SPI flash ID information.
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:355] Unknown flash manufacturer ID.
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:364] Could not decode flash manufacturer/device ID and have an unknown FPGA size. Assume default flash architecture.
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:382] SPI flash total size = 32 Mbit
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:383] SPI flash page size = 256 bytes
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:384] SPI flash erase block size = 65536 bytes
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:385] SPI flash number of pages = 16384
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:386] SPI flash number of erase blocks = 64 pages
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 2
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/usb.c:131] Reporting no FPGA present - BLADERF_FORCE_NO_FPGA_PRESENT is set.
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 3
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:355] Unknown flash manufacturer ID.
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:375] Could not decode flash manufacturer/device ID, but found a 49 kLE FPGA. Setting the most probable flash architecture.
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:382] SPI flash total size = 32 Mbit
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:383] SPI flash page size = 256 bytes
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:384] SPI flash erase block size = 65536 bytes
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:385] SPI flash number of pages = 16384
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:386] SPI flash number of erase blocks = 64 pages
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1491] Assumptions were made about the SPI flash architecture! Flash commands may not function as expected.
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1498] Skipping FPGA configuration and initialization - BLADERF_FORCE_NO_FPGA_PRESENT is set.
Loading fpga...
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 3
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 1
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1113] Read FPGA version: 0.8.0
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1120] Capability mask after FPGA load: 0x0000000f00000000
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:550] nios_ina219_write: Wrote 0x8000 to addr 0x00
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:537] nios_ina219_read: Read 0x399f from addr 0x00
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:550] nios_ina219_write: Wrote 0x019f to addr 0x00
[DEBUG @ host/libraries/libbladeRF/src/driver/ina219.c:70] Configuration register: 0x019f
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:550] nios_ina219_write: Wrote 0xa000 to addr 0x05
[DEBUG @ host/libraries/libbladeRF/src/driver/ina219.c:82] Calibration register: 0xa000
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1181] bladerf2_initialize: ad9361_init starting
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1183] bladerf2_initialize: ad9361_init complete, status = 0
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1006] _set_tx_mute: muting TX1 (cached: 10000)
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1006] _set_tx_mute: muting TX2 (cached: 10000)
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 2
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 1
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:697] nios_ad56x1_vctcxo_trim_dac_write: Wrote 0x1e37
[... cut ...]
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1076] RX1 trigger read value 0x08
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1116] RX1 trigger write value 0x00
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1076] TX1 trigger read value 0x08
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1076] TX1 trigger read value 0x08
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1116] TX1 trigger write value 0x00
```

rtucker, 2018-10-26:
Awesome!  :)

It looks like there was a race condition between the FX3 reconfiguring GPIO[56] from SPI_MOSI (TX) to UART_RX, versus the FPGA coming out of sys_reset and driving the pin high (the idle/deasserted state for a serial line).  Nothing was driving the bus during the intermediate time, and it was floating at a low enough voltage that it could be interpreted as a logic low (asserted).

The fix is to enable a weak pull-up on the FPGA pin.  This ensures that the bus snaps up to a logic high as soon as the FX3 stops driving it low.

Thanks for your patience, and thanks for trying it out!

PR #683 (2018-10-26): hdl: bladerf-micro: weak pull-up on fx3_uart_rxd (#679)
Add a weak pull-up on fx3_uart_rxd

This is a bus used for both FPGA-to-FX3 UART data and FX3-to-Flash
SPI data, and without something driving it, it tends to float around
0.55 volts (which is very much a low/asserted signal for the UART).

When the FX3 SPI interface has the bus, it puts the FPGA into reset
via fx3_ctl(7), and fx3_uart_rxd goes high-Z accordingly.

I suspect what we're running into is a race condition between the FX3
reconfiguring the pin to UART mode and bringing the FPGA out of reset,
and the FPGA shifting from high-Z t [... cut]
closed 2018-10-26 by rtucker