Nuand/bladeRF #684: No succesfull makefile for BladeRF(latest) in gr-osmosdr(latest)

hkopp64, 2018-10-27:
Compilation BladeRF succesfull:
[100%] Building C object host/utilities/bladeRF-cli/CMakeFiles/bladeRF-cli.dir/__/__/common/src/parse.c.o
[100%] Building C object host/utilities/bladeRF-cli/CMakeFiles/bladeRF-cli.dir/src/input/fgets.c.o
[100%] Linking C executable ../../output/bladeRF-cli
[100%] Built target bladeRF-cli

Compilation gr-osmosdr by Cmake:

-- ######################################################
-- # Gnuradio enabled components                         
-- ######################################################
--   * Python support
--   * Osmocom IQ Imbalance Correction
--   * sysmocom OsmoSDR
--   * FUNcube Dongle
--   * FUNcube Dongle Pro+
--   * IQ File Source & Sink
--   * Osmocom RTLSDR
--   * RTLSDR TCP Client
--   * Ettus USRP Devices
--   * HackRF & rad1o Badge
--   * nuand bladeRF
--   * RFSPACE Receivers
--   * AIRSPY Receiver
--   * Red Pitaya SDR
-- 
-- ######################################################
-- # Gnuradio disabled components                        
-- ######################################################
--   * Osmocom MiriSDR
--   * SoapySDR support
--   * FreeSRP support

 Compilation gr-osmosdr by make:
In file included from /usr/local/src/gr-osmosdr/lib/bladerf/bladerf_source_c.h:26:0,
                 from /usr/local/src/gr-osmosdr/lib/source_impl.cc:72:
/usr/local/src/gr-osmosdr/lib/bladerf/bladerf_common.h:291:50: error: ‘>>’ should be ‘> >’ within a nested template argument list
   static std::list<boost::weak_ptr<struct bladerf>> _devs;  /**< dev cache */

lib/CMakeFiles/gnuradio-osmosdr.dir/build.make:134: recipe for target 'lib/CMakeFiles/gnuradio-osmosdr.dir/device.cc.o' failed
make[2]: *** [lib/CMakeFiles/gnuradio-osmosdr.dir/device.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
lib/CMakeFiles/gnuradio-osmosdr.dir/build.make:62: recipe for target 'lib/CMakeFiles/gnuradio-osmosdr.dir/source_impl.cc.o' failed
make[2]: *** [lib/CMakeFiles/gnuradio-osmosdr.dir/source_impl.cc.o] Error 1
lib/CMakeFiles/gnuradio-osmosdr.dir/build.make:86: recipe for target 'lib/CMakeFiles/gnuradio-osmosdr.dir/sink_impl.cc.o' failed
make[2]: *** [lib/CMakeFiles/gnuradio-osmosdr.dir/sink_impl.cc.o] Error 1
[ 51%] Built target osmosdr_swig_swig_2d0df
Scanning dependencies of target pygen_swig_3045c
[ 55%] Generating osmosdr_swig.pyc
[ 55%] Generating osmosdr_swig.pyo
[ 57%] Built target pygen_swig_3045c
CMakeFiles/Makefile2:135: recipe for target 'lib/CMakeFiles/gnuradio-osmosdr.dir/all' failed
make[1]: *** [lib/CMakeFiles/gnuradio-osmosdr.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

How can this be solved, please let me know

Regards,

hkopp64

hkopp64, 2018-10-27:
Tried another version of gr-osmosdr with cmake same as above execpt with make;

[ 74%] Building CXX object lib/CMakeFiles/gnuradio-osmosdr.dir/bladerf/bladerf_source_c.cc.o
[ 76%] Building CXX object lib/CMakeFiles/gnuradio-osmosdr.dir/bladerf/bladerf_sink_c.cc.o
/usr/local/src/gr-osmosdr/lib/bladerf/bladerf_source_c.cc: In member function ‘virtual double bladerf_source_c::get_center_freq(size_t)’:
/usr/local/src/gr-osmosdr/lib/bladerf/bladerf_source_c.cc:250:69: error: cannot convert ‘uint32_t* {aka unsigned int*}’ to ‘bladerf_frequency* {aka long unsigned int*}’ for argument ‘3’ to ‘int bladerf_get_frequency(bladerf*, bladerf_channel, bladerf_frequency*)’
   ret = bladerf_get_frequency( _dev.get(), BLADERF_MODULE_RX, &freq );
                                                                     ^
lib/CMakeFiles/gnuradio-osmosdr.dir/build.make:422: recipe for target 'lib/CMakeFiles/gnuradio-osmosdr.dir/bladerf/bladerf_source_c.cc.o' failed
make[2]: *** [lib/CMakeFiles/gnuradio-osmosdr.dir/bladerf/bladerf_source_c.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/usr/local/src/gr-osmosdr/lib/bladerf/bladerf_sink_c.cc: In member function ‘virtual double bladerf_sink_c::get_center_freq(size_t)’:
/usr/local/src/gr-osmosdr/lib/bladerf/bladerf_sink_c.cc:350:69: error: cannot convert ‘uint32_t* {aka unsigned int*}’ to ‘bladerf_frequency* {aka long unsigned int*}’ for argument ‘3’ to ‘int bladerf_get_frequency(bladerf*, bladerf_channel, bladerf_frequency*)’
   ret = bladerf_get_frequency( _dev.get(), BLADERF_MODULE_TX, &freq );
                                                                     ^
lib/CMakeFiles/gnuradio-osmosdr.dir/build.make:446: recipe for target 'lib/CMakeFiles/gnuradio-osmosdr.dir/bladerf/bladerf_sink_c.cc.o' failed
make[2]: *** [lib/CMakeFiles/gnuradio-osmosdr.dir/bladerf/bladerf_sink_c.cc.o] Error 1
CMakeFiles/Makefile2:135: recipe for target 'lib/CMakeFiles/gnuradio-osmosdr.dir/all' failed
make[1]: *** [lib/CMakeFiles/gnuradio-osmosdr.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

hkopp64, 2018-10-27:
I use opensuse leap 42.3 64bit and gnuradio-3.7.13.4 and latest git source for bladerf

mnhauke, 2018-10-27:
Is there a specific reason that you're compiling this yourself?

For openSUSE users there are ready made packages of the mentioned programs/libraries (and lots of other SDR related stuff) availalable. See [hardware:sdr](
https://build.opensuse.org/project/show/hardware:sdr) on [OBS](https://build.opensuse.org/).

Just add the repo and install the needed packages...

    $ sudo zypper addrepo -f obs://hardware:sdr hardware:sdr
    $ sudo zypper install bladeRF gr-osmosdr

rtucker, 2018-10-30:
Greetings @hkopp64 !

To build from scratch against libbladeRF 2.x, you will need the latest master of gr-osmosdr, as there have been incompatible changes to the API.

It looks like the root cause of the error is:

`/usr/local/src/gr-osmosdr/lib/bladerf/bladerf_common.h:291:50: error: ‘>>’ should be ‘> >’ within a nested template argument list`

Quick fix would be to edit bladerf_common.h:291 (and any other lines it errors out on) to change:

`static std::list<boost::weak_ptr<struct bladerf>> _devs;  /**< dev cache */`

to:

`static std::list<boost::weak_ptr<struct bladerf> > _devs;  /**< dev cache */`

These three lines in bladerf_common.cc will likely also need changing:

```
bladerf_common.cc:54:std::list<boost::weak_ptr<struct bladerf>> bladerf_common::_devs;
bladerf_common.cc:1112:  _devs.push_back(static_cast<boost::weak_ptr<struct bladerf>>(dev));
bladerf_common.cc:1120:  std::list<boost::weak_ptr<struct bladerf>>::iterator it(_devs.begin());
```

hkopp64, 2018-10-30:
Hello, thanks for respond and will look at the by your making suggestions but also tried it with gr-osmosdr on master branch with next results;

-- ######################################################
-- # Gnuradio enabled components                         
-- ######################################################
--   * Python support
--   * Osmocom IQ Imbalance Correction
--   * sysmocom OsmoSDR
--   * FUNcube Dongle
--   * FUNcube Dongle Pro+
--   * IQ File Source & Sink
--   * Osmocom RTLSDR
--   * RTLSDR TCP Client
--   * Ettus USRP Devices
--   * HackRF & rad1o Badge
--   * nuand bladeRF
--   * RFSPACE Receivers
--   * AIRSPY Receiver
--   * Red Pitaya SDR
-- 
-- ######################################################
-- # Gnuradio disabled components                        
-- ######################################################
--   * Osmocom MiriSDR
--   * SoapySDR support
--   * FreeSRP support
-- 
-- Building for version: v0.1.4-127-g4d83c606 / 0.1.5git
-- Using install prefix: /usr/local
-- Configuring done
-- Generating done
-- Build files have been written to: /home/hkopp64/Downloads/gr-osmosdr/build

So far everything is oke, but then when to start make I get still the next errors;

hkopp64@linux:~/Downloads/gr-osmosdr/build> make -j4
[  2%] Built target osmosdr_swig_swig_doc
[  4%] Built target pygen_python_8a1fc
[ 10%] Built target osmosdr_grc_xml_blocks
[ 14%] Built target pygen_apps_502c7
[ 19%] Built target _osmosdr_swig_swig_tag
[ 27%] Built target pygen_apps_005a5
Scanning dependencies of target osmosdr_swig_swig_2d0df
[ 29%] Building CXX object swig/CMakeFiles/osmosdr_swig_swig_2d0df.dir/osmosdr_swig_swig_2d0df.cpp.o
[ 31%] Building CXX object lib/CMakeFiles/gnuradio-osmosdr.dir/sink_impl.cc.o
[ 34%] Building CXX object lib/CMakeFiles/gnuradio-osmosdr.dir/source_impl.cc.o
[ 36%] Building CXX object lib/CMakeFiles/gnuradio-osmosdr.dir/device.cc.o
[ 38%] Linking CXX executable osmosdr_swig_swig_2d0df
Swig source
In file included from /home/hkopp64/Downloads/gr-osmosdr/lib/bladerf/bladerf_sink_c.h:26:0,
                 from /home/hkopp64/Downloads/gr-osmosdr/lib/sink_impl.cc:40:
/home/hkopp64/Downloads/gr-osmosdr/lib/bladerf/bladerf_common.h:291:50: error: ‘>>’ should be ‘> >’ within a nested template argument list
   static std::list<boost::weak_ptr<struct bladerf>> _devs;  /**< dev cache */
                                                  ^
In file included from /home/hkopp64/Downloads/gr-osmosdr/lib/sink_impl.cc:40:0:
/home/hkopp64/Downloads/gr-osmosdr/lib/bladerf/bladerf_sink_c.h:136:32: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11 [enabled by default]
   const float SCALING_FACTOR = 2048.0f;
                                ^
In file included from /home/hkopp64/Downloads/gr-osmosdr/lib/bladerf/bladerf_source_c.h:26:0,
                 from /home/hkopp64/Downloads/gr-osmosdr/lib/device.cc:70:
/home/hkopp64/Downloads/gr-osmosdr/lib/bladerf/bladerf_common.h:291:50: error: ‘>>’ should be ‘> >’ within a nested template argument list
   static std::list<boost::weak_ptr<struct bladerf>> _devs;  /**< dev cache */
                                                  ^
In file included from /home/hkopp64/Downloads/gr-osmosdr/lib/device.cc:70:0:
/home/hkopp64/Downloads/gr-osmosdr/lib/bladerf/bladerf_source_c.h:141:32: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11 [enabled by default]
   const float SCALING_FACTOR = 2048.0f;
                                ^
In file included from /home/hkopp64/Downloads/gr-osmosdr/lib/bladerf/bladerf_source_c.h:26:0,
                 from /home/hkopp64/Downloads/gr-osmosdr/lib/source_impl.cc:72:
/home/hkopp64/Downloads/gr-osmosdr/lib/bladerf/bladerf_common.h:291:50: error: ‘>>’ should be ‘> >’ within a nested template argument list
   static std::list<boost::weak_ptr<struct bladerf>> _devs;  /**< dev cache */
                                                  ^
In file included from /home/hkopp64/Downloads/gr-osmosdr/lib/source_impl.cc:72:0:
/home/hkopp64/Downloads/gr-osmosdr/lib/bladerf/bladerf_source_c.h:141:32: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11 [enabled by default]
   const float SCALING_FACTOR = 2048.0f;
                                ^
lib/CMakeFiles/gnuradio-osmosdr.dir/build.make:134: recipe for target 'lib/CMakeFiles/gnuradio-osmosdr.dir/device.cc.o' failed
make[2]: *** [lib/CMakeFiles/gnuradio-osmosdr.dir/device.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
lib/CMakeFiles/gnuradio-osmosdr.dir/build.make:86: recipe for target 'lib/CMakeFiles/gnuradio-osmosdr.dir/sink_impl.cc.o' failed
make[2]: *** [lib/CMakeFiles/gnuradio-osmosdr.dir/sink_impl.cc.o] Error 1
lib/CMakeFiles/gnuradio-osmosdr.dir/build.make:62: recipe for target 'lib/CMakeFiles/gnuradio-osmosdr.dir/source_impl.cc.o' failed
make[2]: *** [lib/CMakeFiles/gnuradio-osmosdr.dir/source_impl.cc.o] Error 1
CMakeFiles/Makefile2:135: recipe for target 'lib/CMakeFiles/gnuradio-osmosdr.dir/all' failed
make[1]: *** [lib/CMakeFiles/gnuradio-osmosdr.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 40%] Built target osmosdr_swig_swig_2d0df
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
hkopp64@linux:~/Downloads/gr-osmosdr/build> 

I shall look at the by your making sugestions in the code and hope I can solve it but I am not a routinated C programmer but I want to give it a try
Thanks for the support so far !

Regards,
Huib

rtucker, 2018-10-30:
I also did check, and the zypper commands suggested by @mnhauke do install libbladeRF 2.0.2 and the correct gr-osmosdr version under OpenSUSE Leap 42.3 and 15.0 (both tested in Docker containers).  I recommend using this approach, if possible.

The "default" GCC installed for OpenSUSE 42.3 is 4.8.5 and for OpenSUSE 15.0 is 7.3.1, so it is entirely possible this is related to a very old version of GCC.  I've been unable to get PyBOMBS working properly under either of the OpenSUSE versions; however, CentOS 7 also uses GCC 4.8.5 and "pybombs install gr-osmosdr" builds without issue, so I'm honestly not sure.

I would recommend opening an issue w/ the gr-osmosdr issue tracker at https://osmocom.org/projects/gr-osmosdr/issues ... it's not a problem with libbladeRF itself, so any fix would have to go through there regardless.

Thanks!

hkopp64, 2018-10-30:
I have tried your suggestions and yes that soleved greater part of the problem but unfortunately not everything yet. But we are on the right direction :-)
This is the result so far with " make" ;

In file included from /usr/local/src/gr-osmosdr/lib/bladerf/bladerf_source_c.cc:42:0:
/usr/local/src/gr-osmosdr/lib/bladerf/bladerf_source_c.h:141:32: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11 [enabled by default]
   const float SCALING_FACTOR = 2048.0f;
                                ^
In file included from /usr/local/src/gr-osmosdr/lib/bladerf/bladerf_sink_c.cc:42:0:
/usr/local/src/gr-osmosdr/lib/bladerf/bladerf_sink_c.h:136:32: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11 [enabled by default]
   const float SCALING_FACTOR = 2048.0f;
                                ^
[ 76%] Building CXX object lib/CMakeFiles/gnuradio-osmosdr.dir/bladerf/bladerf_common.cc.o
[ 78%] Building CXX object lib/CMakeFiles/gnuradio-osmosdr.dir/rfspace/rfspace_source_c.cc.o
/usr/local/src/gr-osmosdr/lib/bladerf/bladerf_common.cc: In constructor ‘bladerf_common::bladerf_common()’:
/usr/local/src/gr-osmosdr/lib/bladerf/bladerf_common.cc:145:34: error: no matching function for call to ‘boost::shared_ptr<bladerf>::shared_ptr(NULL)’
   _format(BLADERF_FORMAT_SC16_Q11)
                                  ^
/usr/local/src/gr-osmosdr/lib/bladerf/bladerf_common.cc:145:34: note: candidates are:
In file included from /usr/include/boost/shared_ptr.hpp:17:0,
                 from /usr/include/boost/format/alt_sstream.hpp:21,
                 from /usr/include/boost/format/internals.hpp:23,
                 from /usr/include/boost/format.hpp:38,
                 from /usr/local/src/gr-osmosdr/lib/bladerf/bladerf_common.cc:39:
/usr/include/boost/smart_ptr/shared_ptr.hpp:472:14: note: template<class Ap> boost::shared_ptr<T>::shared_ptr(Ap, typename boost::detail::sp_enable_if_auto_ptr<Ap, int>::type)
     explicit shared_ptr( Ap r, typename boost::detail::sp_enable_if_auto_ptr<Ap, int>::type = 0 ): px( r.get() ), pn()
              ^
/usr/include/boost/smart_ptr/shared_ptr.hpp:472:14: note:   template argument deduction/substitution failed:
/usr/include/boost/smart_ptr/shared_ptr.hpp: In substitution of ‘template<class Ap> boost::shared_ptr<T>::shared_ptr(Ap, typename boost::detail::sp_enable_if_auto_ptr<Ap, int>::type) [with Ap = long int]’:
/usr/local/src/gr-osmosdr/lib/bladerf/bladerf_common.cc:145:34:   required from here
/usr/include/boost/smart_ptr/shared_ptr.hpp:472:14: error: no type named ‘type’ in ‘struct boost::detail::sp_enable_if_auto_ptr<long int, int>’
/usr/include/boost/smart_ptr/shared_ptr.hpp:446:14: note: template<class Y> boost::shared_ptr<T>::shared_ptr(std::auto_ptr<_Tp1>&)
     explicit shared_ptr( std::auto_ptr<Y> & r ): px(r.get()), pn()
              ^
/usr/include/boost/smart_ptr/shared_ptr.hpp:446:14: note:   template argument deduction/substitution failed:
/usr/local/src/gr-osmosdr/lib/bladerf/bladerf_common.cc:145:34: note:   mismatched types ‘std::auto_ptr<Y>’ and ‘long int’
   _format(BLADERF_FORMAT_SC16_Q11)
                                  ^
In file included from /usr/include/boost/shared_ptr.hpp:17:0,
                 from /usr/include/boost/format/alt_sstream.hpp:21,
                 from /usr/include/boost/format/internals.hpp:23,
                 from /usr/include/boost/format.hpp:38,
                 from /usr/local/src/gr-osmosdr/lib/bladerf/bladerf_common.cc:39:
/usr/include/boost/smart_ptr/shared_ptr.hpp:439:5: note: template<class Y> boost::shared_ptr<T>::shared_ptr(const boost::shared_ptr<Y>&, boost::shared_ptr<T>::element_type*)
     shared_ptr( shared_ptr<Y> const & r, element_type * p ) BOOST_NOEXCEPT : px( p ), pn( r.pn )
     ^
/usr/include/boost/smart_ptr/shared_ptr.hpp:439:5: note:   template argument deduction/substitution failed:
/usr/local/src/gr-osmosdr/lib/bladerf/bladerf_common.cc:145:34: note:   mismatched types ‘const boost::shared_ptr<X>’ and ‘long int’
   _format(BLADERF_FORMAT_SC16_Q11)
                                  ^
In file included from /usr/include/boost/shared_ptr.hpp:17:0,
                 from /usr/include/boost/format/alt_sstream.hpp:21,
                 from /usr/include/boost/format/internals.hpp:23,
                 from /usr/include/boost/format.hpp:38,
                 from /usr/local/src/gr-osmosdr/lib/bladerf/bladerf_common.cc:39:
/usr/include/boost/smart_ptr/shared_ptr.hpp:425:5: note: template<class Y> boost::shared_ptr<T>::shared_ptr(const boost::shared_ptr<Y>&, typename boost::detail::sp_enable_if_convertible<Y, T>::type)
     shared_ptr( shared_ptr<Y> const & r, typename boost::detail::sp_enable_if_convertible<Y,T>::type = boost::detail::sp_empty() )
     ^
/usr/include/boost/smart_ptr/shared_ptr.hpp:425:5: note:   template argument deduction/substitution failed:
/usr/local/src/gr-osmosdr/lib/bladerf/bladerf_common.cc:145:34: note:   mismatched types ‘const boost::shared_ptr<X>’ and ‘long int’
   _format(BLADERF_FORMAT_SC16_Q11)
                                  ^
In file included from /usr/include/boost/shared_ptr.hpp:17:0,
                 from /usr/include/boost/format/alt_sstream.hpp:21,
                 from /usr/include/boost/format/internals.hpp:23,
                 from /usr/include/boost/format.hpp:38,
                 from /usr/local/src/gr-osmosdr/lib/bladerf/bladerf_common.cc:39:
[... cut ...]
make[2]: *** [lib/CMakeFiles/gnuradio-osmosdr.dir/bladerf/bladerf_common.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:135: recipe for target 'lib/CMakeFiles/gnuradio-osmosdr.dir/all' failed
make[1]: *** [lib/CMakeFiles/gnuradio-osmosdr.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
hkopp64@linux:/usr/local/src/gr-osmosdr/build> 

I you can help me further with the code then the problem is solved and can I test my BladeRF micro A9 with GQRX and gnuradio.
So far many thanks in advance !

Regards,
Huib

hkopp64, 2018-10-31:
I have tried also to install bladeRF and gr-osmosdr and gqrx from opensuse leap 42.3 installer files but what I have seen is that the Library for BladeRF-2 is not available (I am not shure If my BladeRF micro A9 make use of it).
Tried to open device in drop downbox from gqrx but my micro A9 is not detected.

That is my biggest reason why I install gnuradio / osmo / airspy / uhd / bladeRF always build from scratch and til now I had most of the time when I use the correct dependencies always succes by compilation from scratch in opensuse and had never succes for all my sdr radio's i want to use in GQRX.

Regards,
huib

hkopp64, 2018-10-31:
Building BladeRF from scratch was succesful and could communicate and update the firmware by USB3 to my micro A9 without any problems only a slide modification in udev was nessesary with the next lines 

# Nuand bladeRF
SUBSYSTEMS=="usb", ATTRS{idVendor}=="2cf0", ATTRS{idProduct}=="5246", MODE="0666", GROUP=100

# Nuand bladeRF, legacy VID/PID
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1d50", ATTRS{idProduct}=="6066", MODE="0666", GROUP=100

# Cypress Bootloader
SUBSYSTEMS=="usb", ATTRS{idVendor}=="04b4", ATTRS{idProduct}=="00f3", MODE="0666", GROUP=100

And then it works fine but unfortunately does it not for gr-osmosdr so far.

regards

hkopp64, 2018-10-31:
Oke that is nice to know but I am not familiar with suse OBS and gives my the feeling that is very complex of the lots of possibilities it has and makes for me very difficult how to start and what I am doing and how to test. This is a very big difference in front of suse studio.
It should be possible in suse studio express which I have tried but getting lost

Befor OBS I was familiar with SUSe Studio and gives me a good understanding about dependencies and compiling a personal image so that it support al my sdr radio kits (Airspy, RTL-Sdr, USRP-b210 and now tried BladeRF micro A9) in gnuradio and gqrx
As I earlier said I am not a software design engineer or routined programmer but have a little understanding about compilation and little understandin about C but my roots laying more on hardware RF or digitalm/ analog components (Hamradio)

hkopp64, 2018-10-31:
But come back to the last problem in compiling gr-osmosdr after the corrected c-code in the bladeRF library have now an another issue and looks like somthing wrong with boost dependency and maybe also incombination with the gcc compiler.
Which version of boost and/or gcc are nessesary to solve my last issue here ?
I am happy if someone can help me out with this 
regards
huib

rtucker, 2018-11-01:
G'day,

Unfortunately, I'm not really familiar with building gr-osmosdr, beyond using PyBOMBS to set up the build environment.

I might suggest posting to the osmocom-sdr mailing list at https://lists.osmocom.org/mailman/listinfo/osmocom-sdr ... this would be a good way to get in touch with folks who know gr-osmosdr much better than we do.

Thanks!

hkopp64, 2018-11-08:
Hello all,

I have made a change from opensuse leap 42.3 to opensuse leap 15.0 which have the following compiler and boost library;

- gcc version 7.3.1 20180323 [gcc-7-branch revision 258812] (SUSE Linux)
- boost-1.66
- libboost-1.66

Compiling uhd, gnuradio, bladeRF, gr-osmosdr and gqrx where now fully succefull

closed 2018-11-08 by hkopp64