Nuand/bladeRF #686: Problem with loading FPGA for bladeRF x40

hakou123, 2018-11-06:
My problem:
_**_hadom@hadom:~/gps/gps-sdr-sim$ bladeRF-cli -l /home/hadom/hostedx40-latest.rbf
Deferring device initialization due to requested FPGA or firmware actions.
Loading fpga...
[ERROR @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:1644] RX DC calibration table not found for this board.
Successfully loaded FPGA bitstream!_**_
BladeRF parameter:
  **_Board:                    Nuand bladeRF (bladerf1)
  Serial #:                 9f5f9aea909416d947c44b6a07ed42db
  VCTCXO DAC calibration:   0x91e4
  FPGA size:                40 KLE
  FPGA loaded:              yes
  Flash size:               32 Mbit
  USB bus:                  4
  USB address:              2
  USB speed:                SuperSpeed
  Backend:                  libusb
  Instance:                 0_**
How can fix "loading ERROR" for my broad? Please let me know
Regards,
hakou123

rtucker, 2018-11-06:
Greetings,

That error indicates that there is no DC calibration table present, and so automatic gain control is not available.  If you do not need AGC, you can ignore the error; everything will work fine.  Otherwise, you can create a DC calibration table using the instructions at:

https://github.com/Nuand/bladeRF/wiki/DC-offset-and-IQ-Imbalance-Correction#Generating_a_DC_offset_table

The message should probably be a warning, rather than an error.

hakou123, 2018-11-08:
thanks a lot.

commit cd6d78c7 (2018-11-12): libbladeRF: bladerf1: improve AGC-unavailable messaging (fixes #686)

New messages have a lower severity, explain what will happen
instead (manual gain control), and include an info message with
guidance for solving the issue.
closed 2018-11-12 by rtucker