Nuand/bladeRF #687: BladeRF micro A9 invalid operation or parameter (-3)

hkopp64, 2018-11-08:
Hello,

When starting GQRX, bladeRF micro A9 is detected and initialized and using 10MHz samplerate.
But when GQRX tried to configure  the bladeRF gain settings the next thing happens;

gr-osmosdr v0.1.4-127-g4d83c606 (0.1.5git) gnuradio 3.7.13.4
built-in source types: file fcd rtl_tcp uhd bladerf rfspace airspy redpitaya 
Resampling audio 96000 -> 48000
BookmarksFile is /home/hkopp64/.config/gqrx/bookmarks.csv
[INFO] [UHD] linux; GNU C++ version 7.3.1 20180323 [gcc-7-branch revision 258812]; Boost_106600; UHD_3.14.0.0-165-g1d5c186c
gr-osmosdr v0.1.4-127-g4d83c606 (0.1.5git) gnuradio 3.7.13.4
built-in source types: file fcd rtl_tcp uhd bladerf rfspace airspy redpitaya 
[bladeRF common] init: DEBUG: entering initialization
[bladeRF source] init: Opening Nuand bladeRF with device identifier string '*:instance=0'
[bladeRF source] Device: Nuand bladeRF 2.0 Serial # c344...0e61 FW v2.3.1 FPGA v0.9.0
[bladeRF source] init: Buffers: 512, samples per buffer: 4096, active transfers: 32
[bladeRF source] bladerf_source_c: DEBUG: initialization complete
FOUND full
[ERROR @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1997] bladerf2_get_gain_stage_range: _get_gain_range failed: Invalid operation or parameter
[ERROR @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:2036] bladerf2_set_gain_stage: bladerf2_get_gain_stage_range failed: Invalid operation or parameter
terminate called after throwing an instance of 'std::runtime_error'
  what():  set_gain: Failed to set gain for stage 'IF': Invalid operation or parameter (-3)

Warning: Program '/usr/local/bin/gqrx' crashed.

How can I resolve this issue ?

robertghilduta, 2018-11-08:
Can you please run ver from bladeRF-cli -i?

Also, it might be useful to temporarily clear out your ~/.gqrx directory,
perhaps there's some old setting causing this.

hkopp64, 2018-11-08:
hkopp64@linux-s1l3:~> bladeRF-cli -i
bladeRF> v

  bladeRF-cli version:        1.7.0-git-db24d410-dirty
  libbladeRF version:         2.1.0-git-db24d410-dirty

  Firmware version:           2.3.1
  FPGA version:               0.9.0 (configured by USB host)

bladeRF> i

  Board:                    Nuand bladeRF 2.0 (bladerf2)
  Serial #:                 c344cd346616498ba7a0fd21e2e80e61
  VCTCXO DAC calibration:   0x1d9f
  FPGA size:                301 KLE
  FPGA loaded:              yes
  Flash size:               128 Mbit
  USB bus:                  2
  USB address:              2
  USB speed:                SuperSpeed
  Backend:                  libusb
  Instance:                 0

bladeRF>

hkopp64, 2018-11-08:
Hello,

I cleared the settings of GQRX by commandline with $ gqrx -r and that solved the problem.

Thanks for the help

closed 2018-11-11 by hkopp64