Nuand/bladeRF #718: Headles driver for ad9361 in Nios2

hyanki, 2018-12-19:
Hi
I wanted to know that how to import and run headless software in Nios2 eclipse.

bglod, 2018-12-19:
Read through the [Eclipse GUI Debugging section of the Nios readme.md](https://github.com/Nuand/bladeRF/tree/master/hdl/fpga/platforms/common/bladerf/software#eclipse-gui-debugging) and reply with any questions/comments.

Personally, I find it easier and faster to just use gdb [(instructions)](https://github.com/Nuand/bladeRF/tree/master/hdl/fpga/platforms/common/bladerf/software#command-line-debugging-with-gdb) because Eclipse can be a little janky.

rtucker, 2018-12-19:
As for how to actually control the RFIC...

I'm working on an example Nios application to demonstrate controlling the RFIC from within the Nios.  It's still very much a work-in-progress so I'm not going to link to it yet, but the basic gist of it is:

```c
#define RFIC_ADDRESS(cmd, ch) ((cmd & 0xFF) + ((ch & 0xF) << 8))

bool queue_is_empty(void)
{
    uint64_t read_data;
    bool rv;
    rv = rfic_command_read(RFIC_ADDRESS(BLADERF_RFIC_COMMAND_STATUS, -1),
                           &read_data);

    return (rv && 0 == ((read_data >> BLADERF_RFIC_STATUS_WQLEN_SHIFT) &
                        BLADERF_RFIC_STATUS_WQLEN_MASK));
}

int main(void)
{
    bool setup_done = false;

    /* Initialize the structures for RFIC control */
    rfic_command_init();

    /* Queue the RFIC initialization routine */
    rfic_command_write(RFIC_ADDRESS(BLADERF_RFIC_COMMAND_INIT, -1), 1);

    while (true) {
        /* Main program loop. */
        /* Run any queued work. */
        rfic_command_work();

        /* Enqueue any additional commands here */
        if (!setup_done && queue_is_empty()) {
            /* These can only be enqueued after BLADERF_RFIC_COMMAND_INIT is
             * done, hence the queue_is_empty() check */
            rfic_command_write(RFIC_ADDRESS(BLADERF_RFIC_COMMAND_FREQUENCY,
                                            BLADERF_CHANNEL_TX(0)),
                               433920000);
            rfic_command_write(RFIC_ADDRESS(BLADERF_RFIC_COMMAND_SAMPLERATE,
                                            BLADERF_CHANNEL_TX(0)),
                               30720000);
            // etc...
            setup_done = true;
        }
    }
}
```

The important parts:

* Call `rfic_command_init()` once during initialization
* Inside of the main loop, call `rfic_command_work()` on every iteration
* Do `rfic_command_write(RFIC_ADDRESS(BLADERF_RFIC_COMMAND_INIT, -1), 1);` to initialize the RFIC
* Don't queue up any additional commands until INIT is complete

To make this a little easier, I'm planning to implement a non-queuing version of `rfic_command_write()`, to call the functions directly without going through the command queue.  This would eliminate the `rfic_command_init()` and `rfic_command_work()` calls, and free up some memory etc.

hyanki, 2018-12-21:
Hi rtucker
Thanks for update.
Regards
hyanki

hyanki, 2018-12-21:
Hi rtucker
For testing the nios hardware when I run hello world in it I am not getting any message in Nios2 console.
Regards
J S Hyanki

rtucker, 2019-01-30:
Make sure that, in the Makefile (hdl/fpga/platforms/bladerf-micro/software/bladeRF_nios/Makefile), you have -DBLADERF_NIOS_DEBUG on the CFLAGS line:

`CFLAGS              += -DBOARD_BLADERF_MICRO -DBLADERF_NIOS_DEBUG`

Do a make clean_all, then on your next build, it should have the console debugging.  (It's not included by default.)

hyanki, 2019-03-02:
Hi rtucker
Any update regarding headless software on nios2?
Regards
J S Hyanki