Nuand/bladeRF #728: Osmosdr source nchan arg missing ?

oscarmezyk, 2019-01-27:
After updating gnuradio with pybombs I can't get osmocom source&sink running again in a previously working flow:

> Traceback (most recent call last):
>   File "/home/oskar/gr-flows/bladeRF_loopback.py", line 601, in <module>
>     main()
>   File "/home/oskar/gr-flows/bladeRF_loopback.py", line 589, in main
>     tb = top_block_cls(dc_offset_i=options.dc_offset_i, dc_offset_q=options.dc_offset_q, delay=options.delay, instance=options.instance, iq_mag=options.iq_mag, iq_phase=options.iq_phase, 
> loopback=options.loopback, rx_bandwidth=options.rx_bandwidth, rx_frequency=options.rx_frequency, rx_lna_gain=options.rx_lna_gain, rx_num_buffers=options.rx_num_buffers, rx_num_xfers=options.rx_num_xfers, rx_vga_gain=options.rx_vga_gain, serial=options.serial, tx_gain=options.tx_gain, tx_num_buffers=options.tx_num_buffers, tx_num_xfers=options.tx_num_xfers, verbosity=options.verbosity)
> 
>   File "/home/oskar/gr-flows/bladeRF_loopback.py", line 153, in __init__
>     self.osmosdr_source_0 = osmosdr.source( args="numchan=" + str(1) + " " + bladerf_rx_args )
>   File "/home/oskar/gnurf/lib/python2.7/dist-packages/osmosdr/osmosdr_swig.py", line 1173, in make
>     return _osmosdr_swig.source_make(*args, **kwargs)
> RuntimeError: Wrong device arguments specified. Missing nchan?

Genrated code for source:

> 
> # Imports
> import osmosdr
> import time
> 
> # Blocks
> osmosdr.source( args="numchan=" + str(1) + " " + bladerf_rx_args )
> self.osmosdr_source_0.set_sample_rate(sampling_rate)
> self.osmosdr_source_0.set_center_freq(gui_rx_frequency, 0)
> self.osmosdr_source_0.set_freq_corr(0, 0)
> self.osmosdr_source_0.set_dc_offset_mode(0, 0)
> self.osmosdr_source_0.set_iq_balance_mode(0, 0)
> self.osmosdr_source_0.set_gain_mode(False, 0)
> self.osmosdr_source_0.set_gain(gui_rx_vga_gain, 0)
> self.osmosdr_source_0.set_if_gain(0, 0)
> self.osmosdr_source_0.set_bb_gain(0, 0)
> self.osmosdr_source_0.set_antenna('1', 0)
> self.osmosdr_source_0.set_bandwidth(gui_rx_bandwidth, 0)

> 
> bladeRF-cli --lib-version
> 2.2.0-git-e4bf84a
> bladeRF-cli --version
> 1.7.1-git-e4bf84a
> gnuradio-config-info -v
> 3.7.13.4
 
gqrx and osmocom_siggen work 

any hints?
thanks

robertghilduta, 2019-01-28:
Try adding a bladerf=0 to the source block. Also, please consider doing a
clean install and seeing if the bladerf osmo source block works.

oscarmezyk, 2019-01-28:
I almost dug out bare wires from my board to find out what's wrong... 

verbosity => "(space)debug" makes such crash - those strings need to be treated with proper care...

I have missed that and it was really hard to catch. I am not really sure now whether I have changed it before updating pybombs or after, but it wasted my weekend ;)

Issue can be closed, but I'd add such "improvement" to strip R/L spaces from those strings

b.r.

closed 2019-01-28 by oscarmezyk