Nuand/bladeRF #729: bladeRF 2.0 micro dual RX empty samples

karel, 2019-01-31:
When using the `osmocom-source` in a GNU Radio flowgraph to simultaneously receive from both of the bladeRF's inputs, some samples are consistently empty on both of the streams at the same time. This happens with the bladeRF 2.0 micro in GNU Radio (3.7.13.4) with libbladerf and gr-osmosdr built from their master branches. When using a single RX channel, everything works fine. `osmocom-source` is configured with the device argument `bladerf=0,nchan=2`. The problem happens regardless of the sample rate and on all tested systems (albeit with similar software, i.e. an up-to-date Arch Linux).

[image]

The following GIF illustrates how the receiver noise floor either drops or freezes as the result of the empty samples. Having saved the samples to a file, it's evident that the issue arises periodically and the rate seems somehow related to the `buflen` parameter. With `buflen=1024` the number of blank samples is lesser than with the default setting.

[image]

Nothing seems to be wrong with the output (verbose). After all, the issue might arise in the gr-osmosdr module.

```
Generating: '/home/user/projects/sweep-mitigation/grc/top_block.py'

Executing: /usr/bin/python2 -u /home/user/projects/sweep-mitigation/grc/top_block.py

gr-osmosdr v0.1.4-127-g4d83c606 (0.1.5git) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy soapy redpitaya 
[bladeRF common] init: DEBUG: entering initialization
[bladeRF source] init: Opening Nuand bladeRF with device identifier string '*:instance=0'
[DEBUG @ host/libraries/libbladeRF/src/devinfo.c:278] Instance: 0
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:642] Using libusb version: 1.0.22.11312
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:519] Found a bladeRF (idx=8)
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/libusb.c:191] Bus 004 Device 020: Nuand bladeRF 2.0, serial eea66a512f7c41d29007e7bc81220be4
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:581] USB port reset succeeded for bladeRF eea66a512f7c41d29007e7bc81220be4
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 0
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:292] Read Firmware version: 2.3.2
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:299] Capability mask before FPGA load: 0x0000003f00000000
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:399] Found SPI flash manufacturer: WINBOND.
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:402] Found SPI flash device: W25Q32JV (32 Mbit).
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:445] SPI flash total size = 32 Mbit
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:446] SPI flash page size = 256 bytes
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:447] SPI flash erase block size = 65536 bytes
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:448] SPI flash number of pages = 16384
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:449] SPI flash number of erase blocks = 64 pages
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 2
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 1
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 1
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:125] Read FPGA version: 0.10.2
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:132] Capability mask after FPGA load: 0x0000003f00000abc
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:587] nios_ina219_write: Wrote 0x8000 to addr 0x00
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:574] nios_ina219_read: Read 0x399f from addr 0x00
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:587] nios_ina219_write: Wrote 0x019f to addr 0x00
[DEBUG @ host/libraries/libbladeRF/src/driver/ina219.c:70] Configuration register: 0x019f
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:587] nios_ina219_write: Wrote 0xa000 to addr 0x05
[DEBUG @ host/libraries/libbladeRF/src/driver/ina219.c:82] Calibration register: 0xa000
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/common.c:270] Default tuning mode: FPGA
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:2276] bladerf2_set_tuning_mode: New tuning mode: FPGA
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:2337] bladerf2_set_tuning_mode: Maintaining FPGA RFIC control
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 2
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 1
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:785] nios_ad56x1_vctcxo_trim_dac_write: Wrote 0x1eb3
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:429] nios_config_read: Read 0x00000001
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:773] nios_ad56x1_vctcxo_trim_dac_read: Read 0x1eb3
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:2558] trim DAC is enabled
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:773] nios_ad56x1_vctcxo_trim_dac_read: Read 0x1eb3
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:2591] saving current trim DAC value: 0x1eb3
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:785] nios_ad56x1_vctcxo_trim_dac_write: Wrote 0xc000
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:429] nios_config_read: Read 0x00000001
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:440] nios_config_write: Wrote 0x00000801
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:811] nios_adf400x_write: Wrote 0x00000064 to addr 0x00
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:811] nios_adf400x_write: Wrote 0x00006000 to addr 0x01
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:811] nios_adf400x_write: Wrote 0x001f8090 to addr 0x02
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:429] nios_config_read: Read 0x00000801
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:440] nios_config_write: Wrote 0x00000001
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:773] nios_ad56x1_vctcxo_trim_dac_read: Read 0xc000
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:2558] trim DAC is disabled
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:773] nios_ad56x1_vctcxo_trim_dac_read: Read 0xc000
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:2595] restoring old trim DAC value: 0x1eb3
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:785] nios_ad56x1_vctcxo_trim_dac_write: Wrote 0x1eb3
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:207] _bladerf2_initialize: complete
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1142] Clearing RX1 retune queue.
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1165] RX1 operation duration: 0
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1142] Clearing TX1 retune queue.
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1165] TX1 operation duration: 0
[bladeRF source] Device: Nuand bladeRF 2.0 Serial # eea6...0be4 FW v2.3.2 FPGA v0.10.2
[bladeRF source] init: Buffers: 512, samples per buffer: 4096, active transfers: 32
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 0
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 1
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:429] nios_config_read: Read 0x00000001
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:440] nios_config_write: Wrote 0x00000001
[bladeRF source] bladerf_source_c: DEBUG: initialization complete
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/common.c:339] check_total_sample_rate: active_channels=2, rate_accum=4000000, maximum=80000000
[bladeRF source] start: DEBUG: starting source
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:429] nios_config_read: Read 0x00000001
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:440] nios_config_write: Wrote 0x00000001
[DEBUG @ host/libraries/libbladeRF/src/streaming/sync.c:401] sync_rx: Worker is idle. Going to reset buf mgmt.
[DEBUG @ host/libraries/libbladeRF/src/streaming/sync.c:421] sync_rx: Reset buf_mgmt consumer index
[DEBUG @ host/libraries/libbladeRF/src/streaming/sync.c:436] sync_rx: Worker is now running.
[bladeRF source] stop: DEBUG: stopping source

>>> Done
```

karel, 2019-02-01:
Tested also the bladeRF-cli with the following configuration while transmitting a linear sweep using different hardware.

```
rx set frequency rx1 2.4G
rx set frequency rx2 2.4G
rx set samplerate rx1 2M
rx set samplerate rx2 2M
rx config channels=1,2 format=bin n=20M file=samples.raw
```

That should give 10 s of samples from either channel. But gives +5 s and the linear sweep is not so linear (see the picture below). I suppose that the SoapyBladeRF and gr-osmosdr are not to blame here.

[image]

gasparka, 2019-02-01:
I am experiencing this as well.

rtucker, 2019-02-01:
What version of the libbladeRF library are you using?  `bladeRF-cli -e version` will print out the relevant info.

Thanks!  -rt

karel, 2019-02-01:
Built from the master branch. Tried various previous versions as well but no help.

gasparka, 2019-02-01:
```
  bladeRF-cli version:        1.7.1-git-9dd43725-dirty
  libbladeRF version:         2.2.0-git-9dd43725-dirty

  Firmware version:           2.3.2
  FPGA version:               0.10.2 (configured by USB host)
```

A way to reproduce this is to record a sweeping signal, it will be evident that some, periodic, parts of the sweep are missing from both channels.
Thanks for looking into it!

karel, 2019-02-06:
It seems that the `sync_rx` function is incompatible with the incoming buffer structure when using metadata and thus results in discontinuities. This makes it impossible to, e.g., synchronize TX and RX. This is in addition to the issues in gr-osmosdr, which initially led to believe that samples go missing.

https://github.com/Nuand/bladeRF/blob/9dd43725c981ede12088dadfe8193365bec07bb2/host/libraries/libbladeRF/src/streaming/sync.c#L343

gasparka, 2019-02-15:
@rtucker this issue is blocking our work, can you give us an estimate when this could be fixed?

KarlL2, 2019-02-22:
I realized the issue I opened [Strange timestamps with bladeRF 2.0](https://github.com/Nuand/bladeRF/issues/730) is probably the same as this one.

I don't see `Sample discontinuity detected` as I'm probably not in debug, and it happens even in single RX with no TX. In dual RX it seems to happen 50% of the time.

KarlL2, 2019-03-18:
Any update on this issue? Maybe [Strange timestamps with bladeRF 2.0](https://github.com/Nuand/bladeRF/issues/730) should be closed, if it's indeed the same issue.

robertghilduta, 2019-03-18:
@KarlL2 This bug is now being investigated along with a solution for it. I hope to provide an update soon.

robertghilduta, 2019-03-18:
@KarlL2 This bug/issue is now being looked into along with a fix for it. I hope to provide an update soon.

robertghilduta, 2019-03-27:
This is likely a bug in the HDL and may be related to a timing constraint. I am able to reproduce the issue, however there does not appear to be a pattern to the manifestation.
t=1311747200790053665 in hexadecimal is 0x1234432112344321 . The noteworthy part is that 0x12344321 is the value that should only appear in the reserved field, however it is appearing in the 64bit timestamp space twice.

robertghilduta, 2019-03-30:
@karel @KarlL2 @gasparka here is the [xA4 RBF](https://nuand.com/tmp/d9f8a39/hostedxA4.rbf) and [xA9 RBF](https://nuand.com/tmp/d9f8a39/hostedxA9.rbf) for this [branch](https://github.com/Nuand/bladeRF/tree/dev-hdl-rx-fifo-fixes )
This branch attempts to keep the RX FIFOs from getting into a situation where the meta FIFO is empty but the samples FIFO contains enough data for a GPIF transaction.

KarlL2, 2019-04-02:
I tried the `xA4 RBF` you provided and I don't get those timing issues even at 55MS/s (mono channel). I would previously have those jumps every other block of samples.

karel, 2019-04-03:
@rghilduta, the [xA4 RBF](https://nuand.com/tmp/d9f8a39/hostedxA4.rbf) results in immediate transfer timeouts.

```
[bladeRF source] start: DEBUG: starting source
[DEBUG @ host/libraries/libbladeRF/src/streaming/sync.c:401] sync_rx: Worker is idle. Going to reset buf mgmt.
[DEBUG @ host/libraries/libbladeRF/src/streaming/sync.c:421] sync_rx: Reset buf_mgmt consumer index
[DEBUG @ host/libraries/libbladeRF/src/streaming/sync.c:436] sync_rx: Worker is now running.
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:306] wait_for_buffer: Timed out waiting for buf_ready after 3000 ms
[bladeRF source] work: bladerf_sync_rx error: Operation timed out
[ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1083] Transfer timed out for buffer 0x7f52d0006af0
[ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1083] Transfer timed out for buffer 0x7f52d000ab00
```

Here's the simple GNU Radio graph to test dual RX with metadata enabled [bladerf_dual_RX.grc.zip](https://github.com/Nuand/bladeRF/files/3037781/bladerf_dual_RX.grc.zip).

robertghilduta, 2019-04-20:
@karel and @KarlL2 please use the following branch: https://github.com/Nuand/bladeRF/tree/dev-fixes with the RBFs located here for the [xA4](https://nuand.com/tmp/d99a4fa/hostedxA4.rbf) and [xA9](https://nuand.com/tmp/d99a4fa/hostedxA9.rbf)

robertghilduta, 2019-04-20:
@karel and @KarlL2 there is another branch to try with a few fixes and feature improvements. Please try the following branch: https://github.com/Nuand/bladeRF/tree/dev-fixes . The RBFs are located here for the [xA4](https://nuand.com/tmp/d99a4fa/hostedxA4.rbf) and [xA9](https://nuand.com/tmp/d99a4fa/hostedxA9.rbf)

robertghilduta, 2019-04-20:
@karel and @KarlL2 there is another branch to try with a few fixes and feature improvements. Please try the following [branch](https://github.com/Nuand/bladeRF/tree/dev-fixes) . The RBFs are located here for the [xA4](https://nuand.com/tmp/d99a4fa/hostedxA4.rbf) and [xA9](https://nuand.com/tmp/d99a4fa/hostedxA9.rbf)

kjopek, 2019-06-20:
Found similar issues with bladeRF 1 x115:
```
[DEBUG @ host/libraries/libbladeRF/src/streaming/sync.c:565] Sample discontinuity detected @ buffer 8, message 26: Expected t=1311747200790054173, got t=181443577
Ts: 1311747200790053665 Count: 508
[DEBUG @ host/libraries/libbladeRF/src/streaming/sync.c:565] Sample discontinuity detected @ buffer 8, message 31: Expected t=181446117, got t=779305138806604577
Ts: 181443577 Count: 2540
[DEBUG @ host/libraries/libbladeRF/src/streaming/sync.c:565] Sample discontinuity detected @ buffer 8, message 32: Expected t=779305138806605085, got t=1311747200790053665
Ts: 779305138806604577 Count: 508
[DEBUG @ host/libraries/libbladeRF/src/streaming/sync.c:565] Sample discontinuity detected @ buffer 8, message 33: Expected t=1311747200790054173, got t=181446625
Ts: 1311747200790053665 Count: 508
Ts: 181446625 Count: 1048576
```

mando238, 2019-07-05:
For what its worth I am also experiencing this, even with the new branch (FPGA v 0.11) from above, Though now only when attempting multiple channel receive (receive on both Rx channels simultaneously) regardless of sample rate. It happens with Soapy and Osmocom,  problem is nearly identical to 2 issues that were referenced/merged with this topic (https://github.com/Nuand/bladeRF/issues/729 https://github.com/pothosware/SoapyBladeRF/issues/26) . Receiving on a single Rx channel seems to work fine. 

  bladeRF-cli version:        1.7.1
  libbladeRF version:         2.2.0

  Firmware version:           2.3.2
  FPGA version:               0.11.0 (configured by USB host)

Device is A4 Micro Rev 1.3.
_A note of improvement_. During 2 channel receive, With  0.10 I could only receive 508 samples, regardless of other stream settings. When using  0.11 instead I can get to 10240 samples before issues start appearing. I haven't messed around to much with the newer branch beyond basic tests, I'll update if I find something interesting.

gorgiaxx, 2019-07-10:
The similar issues with x115:

```
  bladeRF-cli version:        1.7.1-2018.12-rc3-2-ppaxenial
  libbladeRF version:         2.2.0-2018.12-rc3-2-ppaxenial

  Firmware version:           2.3.2
  FPGA version:               0.1.2 (configured by USB host)
```

The yate shows `invalid timestamps`
```
2019-07-10_16:20:34.301557 <ybts:NOTE> State changed WaitHandshake -> Running
Yate engine is initialized and starting up on ubuntu
2019-07-10_16:20:35.193308 <gsmtrx:NOTE> State changed Invalid -> Idle [0x7f4884025590]
2019-07-10_16:20:35.193776 <gsmtrx:NOTE> State changed Idle -> PowerOff [0x7f4884025590]
2019-07-10_16:20:35.258749 <bladerf/1:NOTE> Powered ON the radio [0x7f4884000eb0]
2019-07-10_16:20:35.265439 <gsmtrx:NOTE> State changed PowerOff -> PowerOn [0x7f4884025590]
2019-07-10_16:20:35.307966 <mbts:NOTE> GSMConfig.cpp:532:createCombinationI: Configuring combination I on C0T1
2019-07-10_16:20:35.318566 <mbts:NOTE> GSMConfig.cpp:532:createCombinationI: Configuring combination I on C0T2
2019-07-10_16:20:35.325506 <bladerf/1:MILD> RX buf_samples=508: 4 buffers: invalid timestamps (buf=ts/delta) 2=137273/1615 [0x7f4884000eb0]
2019-07-10_16:20:35.327199 <bladerf/1:MILD> RX buf_samples=508: 4 buffers: invalid timestamps (buf=ts/delta) 2=144390/1529 [0x7f4884000eb0]
2019-07-10_16:20:35.328086 <bladerf/1:MILD> RX buf_samples=508: 4 buffers: invalid timestamps (buf=ts/delta) 2=146588/674 [0x7f4884000eb0]
2019-07-10_16:20:35.348467 <bladerf/1:MILD> RX buf_samples=508: 4 buffers: invalid timestamps (buf=ts/delta) 2=178460/7996 [0x7f4884000eb0]
2019-07-10_16:20:35.350023 <bladerf/1:MILD> RX buf_samples=508: 4 buffers: invalid timestamps (buf=ts/delta) 2=192683/10667 [0x7f4884000eb0]
```

mando238, 2019-07-13:
Tried Bladerf-Cli directly instead of Soapy or Osomocon, same results.. With the new RBF, multi channel RX causes buffer overruns regardless of stream settings, but single channel RX works fine. 
Output below:

bladeRF> version

  bladeRF-cli version:        1.7.1
  libbladeRF version:         2.2.0

  Firmware version:           2.3.2
  FPGA version:               0.11.0 (configured by USB host)

bladeRF> set samplerate rx2 2M

[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/common.c:339] check_total_sample_rate: active_channels=0, rate_accum=0, maximum=80000000
  Setting RX2 sample rate - req:   2000000 0/1Hz, actual:   2000000 0/1Hz

bladeRF> set samplerate rx 2M

[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/common.c:339] check_total_sample_rate: active_channels=0, rate_accum=0, maximum=80000000
  Setting RX1 sample rate - req:   2000000 0/1Hz, actual:   2000000 0/1Hz

bladeRF> rx config file=my_samples.csv format=csv n=20M channel=1,2
bladeRF> rx start
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:429] nios_config_read: Read 0x00000000
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:440] nios_config_write: Wrote 0x00000000
bladeRF> [VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/common.c:339] check_total_sample_rate: active_channels=1, rate_accum=2000000, maximum=80000000
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/common.c:339] check_total_sample_rate: active_channels=2, rate_accum=4000000, maximum=80000000
[DEBUG @ host/libraries/libbladeRF/src/streaming/sync.c:401] sync_rx: Worker is idle. Going to reset buf mgmt.
[DEBUG @ host/libraries/libbladeRF/src/streaming/sync.c:421] sync_rx: Reset buf_mgmt consumer index
[DEBUG @ host/libraries/libbladeRF/src/streaming/sync.c:436] sync_rx: Worker is now running.
[DEBUG @ host/libraries/libbladeRF/src/streaming/sync_worker.c:100] RX overrun @ buffer 4
[DEBUG @ host/libraries/libbladeRF/src/streaming/sync_worker.c:100] RX overrun @ buffer 23
[DEBUG @ host/libraries/libbladeRF/src/streaming/sync_worker.c:100] RX overrun @ buffer 9
[DEBUG @ host/libraries/libbladeRF/src/streaming/sync_worker.c:100] RX overrun @ buffer 27
[DEBUG @ host/libraries/libbladeRF/src/streaming/sync_worker.c:100] RX overrun @ buffer 12
[DEBUG @ host/libraries/libbladeRF/src/streaming/sync_worker.c:100] RX overrun @ buffer 28
[DEBUG @ host/libraries/libbladeRF/src/streaming/sync_worker.c:100] RX overrun @ buffer 14
[DEBUG @ host/libraries/libbladeRF/src/streaming/sync_worker.c:100] RX overrun @ buffer 0
[DEBUG @ host/libraries/libbladeRF/src/streaming/sync_worker.c:100] RX overrun @ buffer 20
[DEBUG @ host/libraries/libbladeRF/src/streaming/sync_worker.c:100] RX overrun @ buffer 6
[DEBUG @ host/libraries/libbladeRF/src/streaming/sync_worker.c:100] RX overrun @ buffer 25
[DEBUG @ host/libraries/libbladeRF/src/streaming/sync_worker.c:100] RX overrun @ buffer 10

--------------------Single Channel-------------------------
bladeRF> rx config file=my_samples.csv format=csv n=20M channel=1
bladeRF> rx start
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:429] nios_config_read: Read 0x00000000
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:440] nios_config_write: Wrote 0x00000000
bladeRF> [VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/common.c:339] check_total_sample_rate: active_channels=1, rate_accum=2000000, maximum=80000000
[DEBUG @ host/libraries/libbladeRF/src/streaming/sync.c:401] sync_rx: Worker is idle. Going to reset buf mgmt.
[DEBUG @ host/libraries/libbladeRF/src/streaming/sync.c:421] sync_rx: Reset buf_mgmt consumer index
[DEBUG @ host/libraries/libbladeRF/src/streaming/sync.c:436] sync_rx: Worker is now running.

robertghilduta, 2019-07-31:
Please try release tag 2019.07, a good amount of the sample discontinuity issue should be resolved. This issue should be reserved for sampling discontinuities where "t=1311747200790054173" appears in the logs. Please note to use the fastest available storage medium such as a solid state drive when using bladeRF-cli with CSV files. If the issues persist or otherwise, please start a new issue.

closed 2019-07-31 by robertghilduta

canopus19652000, 2021-11-30:
Any updates yet? Or nobody cares about the crap?