Nuand/bladeRF #732: Blade v2 issue on GRC and Pothos

KarlL2, 2019-03-07:
Hi,

I'm just monitoring 20MHz on Windows 10 with both GRC v3.7.13.4 and Pothos v2019.01.26  (I'm already seeing Os in the console of Pothos at such a low sampling rate in mono channel).

On GRC the spectrum is pretty flat, except for the far sides which is expected. The gain doesn't seem to work though, as I get a similar spectrum whether I put 0 or 100 for each gain, so I always get a noise floor around -75dB.

[image]

On Pothos the gain does work, but the spectrum is quite weird. Here is what I get with a gain of 0:

[image]

When I raise the gain, the RSSI of the center of the spectrum goes up, the one of the side (which is already high) mostly stays the same.

Gain 15:
[image]

Gain 30:
[image]

Gain 45:
[image]

Gain 60:
[image]

With a gain of 60, the spectrum is flat (except on the far sides), similar to GRC.

KarlL2, 2019-03-18:
Am I the only one seeing this? Is my board defective?

Also with the current software I'm getting overflows already at 20MS/s in single channel. I'm not sure if it's linked with my problem. On the same computer I can go up to around 56MS/s in dual channel, using SC12, with a Lime SDR.

Are there tests I can run to make sure it's just a software issue?

robertghilduta, 2019-03-18:
@KarlL2 just to confirm, you are unable to set the bladeRF's gains using the osmosdr source block but are able to successfully gain in Pothos?

KarlL2, 2019-03-18:
@robertghilduta Indeed, with the osmocom source, whether I put 0 for RF, IF and BB gains, or I put 100 for all of them, I end up with the same spectrum.

With the Soapy SDR Source, I get different spectrums depending on the value I pass to the `Gain Value` field.

The strange thing though with the low gains in Soapy SDR Source, is that the spectrum is really low in the middle and really high on the sides. I only get a flat spectrum with a gain of 60.

robertghilduta, 2019-03-31:
@KarlL2 please try writing "agc off" to bladeRF.conf in the directory from which you launch gnuradio-companion. Alternatively you can give this [branch](https://github.com/Nuand/bladeRF/tree/dev-gain-defaults) a try. Ultimately, the last few patches in https://github.com/Nuand/gr-osmosdr should be pushed up to gr-osmosdr .

KarlL2, 2019-04-02:
@robertghilduta I created a `bladeRF.conf` file containing only `agc off`.
I'm on Windows and I put that config file

- in the same folder as GRC 3.7.13.4 ( `C:\Program Files\GNURadio-3.7\bin` )
- in the same folder as Pothos v0.7.0 2019.01.26 and then 2019.03.24 ( `C:\Program Files\PothosSDR\bin` )
- and in the same folder as a SoapySDR program

I didn't see any difference when running GRC, Pothos Flow or my program.

In GRC I tried using `bladerf=0,agc=0` but got
`RuntimeError: bladerf_common::set_gain_mode: Setting gain mode to 'manual' failed: Operation not supported (-8)`

robertghilduta, 2019-04-02:
@KarlL2 the config file should go in the directory from which you launch gnuradio-companion. 
This is easier to determine if launching from the command line.

Have you given the two branches a try?

closed 2019-04-20 by robertghilduta

KarlL2, 2019-06-28:
I've tried with the latest version on master 896d2431b3a35a4b31b6e729386202ebf5fdc5c0. It contains the cherry picked commits of `dev-gain-defaults` and `dev-fixes`.

We've also found a bug in SoapyBladeRF (see https://github.com/pothosware/SoapyBladeRF/issues/30#issuecomment-506585253 - I don't know if this should be done in SoapyBladeRF on in BladeRF). Now with Soapy we can change the gain.

We still have the issue with the weird spectrum, which is very high on both sides when the gain is low. Eg with a gain of 0, the noise floor is around -95dB in the middle, but on both sides the power goes up to -65dB, even if there's no real signal. This happens it seems on all frequencies. If we push the gain up, the noise floor goes up and eventually is at the same level as the sides.

Is there a test we can do with bladerf-cli? The GRC flow graph mentioned in https://github.com/Nuand/bladeRF/wiki/bladeRF-CLI-Tips-and-Tricks#GNU_Radio_Flowgraph doesn't exist anymore. I've tried using Octave with https://github.com/Nuand/bladeRF/wiki/bladeRF-CLI-Tips-and-Tricks#Binary_Data_SC16Q11_format . I display the periodogram with `plot(periodogram(signal, hamming(8096)));` (that should display the power spectrum, right?) but it seems I still see the high power sides.

marianoboragno, 2020-11-21:
@KarlL2 I've got my hands on two BladeRF xA4 micro the last two weeks and found the same problem as you. Did you find any solution? Is it just the way the BladeRF works?

This doesn't seem to be related to any particular software, the same happens on Windows, Linux, BladeRF-cli captures, GNURadio, Gqrx, etc.
 Any feedback will be appreciated.

Thanks!

KarlL2, 2020-12-15:
@marianoboragno If you're talking about the shape of the spectrum, with very high powers on the sides when the gain is too low, then I would say it's just the way the BladeRF xA4 micro works. There's a filter that tries to flatten the spectrum, but it expects a high gain, and it messes things in low gain. It's possible to remove that filter or modify it, but it requires modifying the code and recompiling it.

gisforgirard, 2021-10-30:
why was this closed if it's clearly not fixed?

rghilduta, 2021-10-30:
This is related to a change in gain settings, changing relative power of between RF noise and noise shaping of the data converters. The conversation is continued in #745  . This is an inherent and expected characteristic of the AD9361 at low gain settings, and should not impact the system SNR at low gains.

The side lobes occur at low gains in the AD936x, please see https://ez.analog.com/wide-band-rf-transceivers/design-support/f/q-a/78834/ad9361-passband-at-low-rx-gain-isn-t-flat and https://ez.analog.com/wide-band-rf-transceivers/design-support/f/q-a/79015/ad9361-wide-bandwidth-noise-floor-shape-issue . Increasing gain should flatten the pass band, and by the highest gain it should be almost completely flat.