Nuand/bladeRF #733: No libbladeRF backends are enabled.  Please enable one or more backends.

handhikadj, 2019-03-26:
I was following this tutorial: https://www.youtube.com/watch?v=6Dtp06-uI38
but got this after `cmake ../`

[image]

i've been following this guide: https://github.com/Nuand/bladeRF/wiki/Getting-Started%3A-Linux#Installing_dependencies

but it doesn't fix my problem.

Please help

robertghilduta, 2019-03-27:
This might be due to libusb not being installed correctly or fully. Can you please run `dpkg -s libusb-1.0-0 libusb-1.0-0-dev ` ?

handhikadj, 2019-03-27:
it shows:

[image]

robertghilduta, 2019-03-27:
Can you please show me the output of "dpkg -L libusb-1.0-0-dev" ?

handhikadj, 2019-03-27:
[image]

robertghilduta, 2019-03-27:
There may be some caching issue going on with cmake in your bladeRF repository checkout. Try cleaning out the bladeRF git repository, or cloning it again. You may also want to try running ` cmake -DENABLE_BACKEND_LIBUSB=ON ../ `

handhikadj, 2019-03-28:
I removed and recloned this repo and it works

closed 2019-03-28 by handhikadj

dakota56, 2020-03-31:
Hello 
I have exactly the same problem when I try to build libbladeRF with the latest git repo. Don't know what to do to make my new blade rf 2.0 card working with gnu radio / gr-osmosdr.

barthogenes, 2021-10-09:
I followed [this](https://github.com/Nuand/bladeRF/wiki/Getting-Started%3A-Linux#Building_bladeRF_libraries_and_tools_from_source) tutorial and added the `-DENABLE_BACKEND_LIBUSB=TRUE` flag to the cmake command so that it becomes like this:
```
cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr/local -DINSTALL_UDEV_RULES=ON -DENABLE_BACKEND_LIBUSB=TRUE ../
```
Doing this fixed it for me.

auliyarusyda, 2023-03-06:
I have this case too, how can I solve this problem?

auliyarusyda, 2023-03-06:
[image]

rghilduta, 2023-03-06:
It doesn't appear like CMake was able to find libusb. Do you know if the libusb package and dev files are installed?

Shawn-cao741, 2024-08-28:
> I removed and recloned this repo and it works

It's right, I solve my problem by recloning the repo. Maybe because when the dependecies updated then the repo also need be updated

mani-pandian-skydio, 2025-01-18:
This seemed to help on Ubuntu:
In `host/CMakeLists.txt`
I replaced `include(FindLibUSB)` with:
`find_package(PkgConfig REQUIRED)`
`pkg_check_modules(LIBUSB REQUIRED libusb-1.0)`