Nuand/bladeRF #735: when using schedule retune with xb200, bladerf frequency tuning gets limited based on first tuned frequency

henlemalick, 2019-03-28:
Hi,

I have been working on using schedule retune functionality in bladerf.... i was interested in transmitting multiple frequencies upto six starting from 100 MHz and above for that i used xb200 expansion board... but there is a strange thing i am experiencing if i input the first frequency to be tuned within 100 to 237 MHz, and rest all others greater than 237.5 MHz,
 the schedule retune program only performs retuning for frequencies less than 237.5 MHz. 

`105000000`
`115000000`
`125000000`
`240000000`    -   does not tune
`255000000`    -   does not tune
`505000000`    -   does not tune

Similarly, if it is the other way around i.e. if first frequency to be tuned is greater than 237.5 MHz then only the frequencies above 237.5 MHz are tuned and lower frequencies get clamped to 237.5 MHz...

`305000000`
`315000000`
`125000000`    -   clamps to 237.5 MHz
`240000000`   
`155000000`    -   clamps to 237.5 MHz
`505000000`

The same thing i have experienced after compiling the quick retune program provided in example c-code library test codes available in bladerf respository.

Am i the only one experiencing this?? May be i did not configure something correctly....

robertghilduta, 2019-03-30:
The quick tune functionality relies on bladerf_set_frequency() to control the XB-200's switches (which determine whether the XB-200 transverters a signal or not). So before quick tuning to frequencies below BLADERF_FREQUENCY_MIN ( 237.5MHz ), bladerf_set_frequency() must be called with a frequency below 237.5MHz to configure the XB-200's switches for transverting (mixing) operation. When the XB-200 is configured to be transverting (mixing), the upper quick tune frequency limit is 237.5MHz, this can be disabled by calling bladerf_set_frequency() with a frequency higher than 237.5MHz .

henlemalick, 2019-04-01:
Ok, i get it that the bladerf_set_frequency() controls the XB200 switches. As schedule retune program only uses this once for initializing and in this way the XB-200 switches are configured only once.
` /* Set initial frequency */
    status = bladerf_set_frequency(dev, module, hop_set_next(hops, NULL));`

But i am curious that when i schedule a retune using the code below, if BLADERF retunes the LO to the scheduled provided hopping frequency through bladerf_set_frequency() or some other way???

`freq = hop_set_next(hops, NULL);
        status = bladerf_schedule_retune(dev, m, *hop_ts, freq, NULL);`

If yes, then XB-200 switches should be reconfigured on every such timestamp where some retune is to occur.
 
But, If this is not the case then calling bladerf_set_frequency() with additional conditional statements would add more latencies and even more than this, make it difficult to use the schedule retune functionalilty with XB-200 transverter. As it will be difficult to sync bladerf_set_frequency() command with the scheduling queue....

Is there any way around this.... I mean if transverter switches could be configured with every scheduled retune using the same schedule retune library functionality. Need help as i was trying to use this in making a hopping link with XB-200 transverter board.

robertghilduta, 2019-04-02:
If you can use quicktune parameter with `bladerf_schedule_retune()` by saving a number of `struct bladerf_quick_tune`, please give this [branch](https://github.com/Nuand/bladeRF/tree/dev-quicktune-xb-gpio-reg) a try. It should let quicktune control the xb gpio register, allowing the XB-200 to be controlled. The libbladeRF compiled from that branch will require these bladeRF RBFs ( [x40 RBF](https://nuand.com/tmp/0aca485/hostedx40.rbf) or [x115 RBF](https://nuand.com/tmp/0aca485/hostedx115.rbf) ).

henlemalick, 2019-04-05:
I have been trying to compile the branch where you made changes with respect to XB200 switches configuration. But the problem I am facing is that I am unable to build the solution after doing configuration and generating  using cmake. I am using Windows 10 Pro with Visual Studio 12. 

Previously i was using the prebuilt latest windows installer so I had not gone through this whole procedure. I have followed the wiki tutorial "Building libbladeRF for Windows" but got stuck after the cmake procedure. One of the errors it shows in the output is 

`Error	1211	error LNK1104: cannot open file '..\..\..\output\Debug\bladeRF.lib'	C:\brf\host\build\libraries\libbladeRF_test\test_cpp\LINK	libbladeRF_test_cpp`

I tried to find the bladeRF.lib file in the build folder but its not there. Is this library file created with cmake or after building with VS. I tried to find the related problems in other forums but all were unsuccessful. As i am a totally beginner in this, there must be some procedural mistake i might be doing. On the other hand, if there is a way to get the prebuilt version of that modified branch. 
Any help regarding this would be appreciated.

Regards

robertghilduta, 2019-04-06:
@henlemalick please use the forums or email bladeRF@nuand.com for guidance with compiling libbladeRF. Do not create pull requests for my commits. This thread is now closed.

robertghilduta, 2019-04-06:
@henlemalick that appears to be a build issue unrelated to quicktune. Please take a look at our wiki instructions for compiling with Visual Studio.

closed 2019-04-06 by robertghilduta

robertghilduta, 2019-04-06:
Feel free to reopen this issue if an XB gpio register related issue appears. However, if bladeRF.lib cannot be found that usually is indicative of an error occurring while compiling libbladeRF. Building libbladeRF has a number of specific requirements for libraries (and their respective C headers) to be available, so please ensure those are installed.