Nuand/bladeRF #738: bladeRF-cli calibrate with x115

wtfuzz, 2019-04-03:
The current git master gives me an error with "cal lms" using a Lime bladeRF x115.

```
bladeRF> version

  bladeRF-cli version:        1.7.1-git-708ccad9
  libbladeRF version:         2.2.0-git-708ccad9

  Firmware version:           2.3.2
  FPGA version:               0.10.2 (configured from SPI flash)

bladeRF> info

  Board:                    Nuand bladeRF (bladerf1)
  Serial #:                 83885ebdf8a6894877a6fbb98ee651f1
  VCTCXO DAC calibration:   0x8f68
  FPGA size:                115 KLE
  FPGA loaded:              yes
  Flash size:               32 Mbit
  USB bus:                  0
  USB address:              12
  USB speed:                SuperSpeed
  Backend:                  libusb
  Instance:                 0

bladeRF> cal lms

  cal: Only use this calibration on bladeRF x40 and x115 devices.

bladeRF>
```

wtfuzz, 2019-04-03:
The board functions to get the FPGA size return 0 on success. I was able to get it to work with this patch.

```patch
diff --git a/host/utilities/bladeRF-cli/src/main.c b/host/utilities/bladeRF-cli/src/main.c
index daa7033e..8e978182 100644
--- a/host/utilities/bladeRF-cli/src/main.c
+++ b/host/utilities/bladeRF-cli/src/main.c
@@ -357,7 +357,7 @@ static int open_device(struct rc_config *rc,
         } else {
             status = bladerf_get_fpga_size(state->dev, &state->dev_info.fpga_size);

-            if (status != 0) {
+            if (status == 0) {
                 if (state->dev_info.fpga_size == BLADERF_FPGA_40KLE ||
                         state->dev_info.fpga_size == BLADERF_FPGA_115KLE) {
                     state->dev_info.is_bladerf_x40_x115 = true;
```

robertghilduta, 2019-04-03:
Thanks for reporting the bug!

closed 2019-04-03 by robertghilduta