Nuand/bladeRF #741: bladeRF-cli: tx start: File operation failed (File Size)

Meown12, 2019-04-11:
Hi everyone,
I was working on some binary GPS I/Q data generated by gps-sdr-sim. These files varied in size between 1-3.2 GBytes. With files larger than roughly 2 GB the following series of commands fails:
`set frequency tx 1575.42M`
`set samplerate 2.6M`
`set bandwidth 2.5M`
`set gain tx 30`
`cal lms`
`cal dc tx`
`tx config file=/my/file/on/Data/Partition/name.bin`
`tx config format=bin`
`tx start`
With a `Error: File operation failed` after the last command.
As smaller files work fine I assumed it would be a buffering issue but I may be completely wrong. I am still trying to find the right spot to look, if this issue is related to bladeRF-cli - so far I have reached the input processing files of bladeRF-cli (I know - not very far, but I haven't done much C so far).
Any suggestions where to look or what other reasons this could be caused by are greatly appreciated.

Thanks
 
gps-sdr-sim version:  ? git1ada56c
bladeRF version: 2.2.0-git 1da130c
Ubuntu 16.04 32Bit
RAM 16 GB
Intel i5-6300U
System partition: ext4
Data Partition: NTFS

robertghilduta, 2019-04-20:
@Meown12 this might be related to the (usermode?) NTFS filesystem driver. Using the native EXT4 filesystem might work best.

Meown12, 2019-05-17:
Hi @rghilduta,
thanks for your idea, I moved the files onto the ext4 partition and created some new ones, that were stored on the ext4 straighaway, but the error seems to persist. I think I located its source however, which seems to be in host/utilities/bladeRF-cli/src/common.c in line 336 in a fopen() command. I found some initial threads in different forums about the limitations of fopen() but I am still looking for a solution.
Thanks :)

Meown12, 2019-05-17:
Quick update:
dug out a 64 bit system and it works a treat. So I guess this is an inherent problem with c and 32-bit machines. If anyone wants to verify that on their system I would be grateful.
Thanks!

closed 2019-05-17 by Meown12