Nuand/bladeRF #745: Big side lobes on BladeRFv2

KarlL2, 2019-07-08:
With the BladeRFv2, I have big side lobes easily visible on the spectrum when the gain is low. With higher gains, the side lobes are less visible but only because the noise floor goes up.

Is that normal behavior? If this comes from internal filters, how do we change them to have a more flat spectrum no matter the gain?

Below are a few examples at different sampling rates, bandwidth filter and gain:

- 55MS/s sampling rate, 55MHz bandwidth, 0 gain
[image]

- 55MS/s sampling rate, 55MHz bandwidth, 30 gain
[image]

- 55MS/s sampling rate, 10MHz bandwidth, 15 gain
[image]

- 15MS/s sampling rate, 3MHz bandwidth, 15 gain
[image]

KarlL2, 2019-07-24:
I modified the default RX filter https://github.com/Nuand/bladeRF/blob/896d2431b3a35a4b31b6e729386202ebf5fdc5c0/fpga_common/src/ad936x_params.c#L414 to be the same as the default TX filter that doesn't do anything https://github.com/Nuand/bladeRF/blob/896d2431b3a35a4b31b6e729386202ebf5fdc5c0/fpga_common/src/ad936x_params.c#L449

The side lobes are less visible but they are still there in low gains like -15 or 0. At gain 15 there's still +5dB on the sides. At gain 30 and above there's up to -10dB on the sides. Is that normal?

All following spectrums have the same 55MS/s sampling rate and 55MHz bandwidth. Only the gain changes.

- Gain -15, with default RX filter then with TX filter:
[image]
[image]

- Gain 0, with default RX filter then with TX filter:
[image]
[image]

- Gain 15, with default RX filter then with TX filter:
[image]
[image]

- Gain 30, with default RX filter then with TX filter:
[image]
[image]

closed 2019-07-25 by Nuand

sbourdeauducq, 2020-08-08:
Why was this closed? I have a bladeRF 2.0 micro and I'm also seeing this problem.

robertghilduta, 2020-08-08:
The side lobes are expected behavior at certain (usually low gain) settings
on the bladeRF 2.0. Please see the following discussion thread about the
AD9361,
https://ez.analog.com/wide-band-rf-transceivers/design-support/f/q-a/78834/ad9361-passband-at-low-rx-gain-isn-t-flat

This is an artefact of the pulse shaping filter. Essentially things are
optimized for high gain situations when SNR is likely to be low.

zehm, 2020-09-15:
I'm curious about this too. The lobes on the sides are massive while my Ettus and other SDR radios don't exhibit this behaviour. Is there a simple fix or is this just the way it is with the bladeRF 2.0?

u-erdemir, 2021-05-05:
Same problem. What is the solution?

rghilduta, 2021-05-05:
The side lobes occur at low gains in the AD936x, please see https://ez.analog.com/wide-band-rf-transceivers/design-support/f/q-a/78834/ad9361-passband-at-low-rx-gain-isn-t-flat and https://ez.analog.com/wide-band-rf-transceivers/design-support/f/q-a/79015/ad9361-wide-bandwidth-noise-floor-shape-issue . Increasing gain should flatten the pass band, and by the highest gain it should be almost completely flat.

KimTLee, 2022-05-20:
Hello

It is possible to pass an argument to choose the type of filter:

https://github.com/Nuand/bladeRF/blob/1e02533da52a0b06f476330d8e267c6bbb427d8f/host/libraries/libbladeRF/src/board/bladerf2/rfic_host.c#L120

Can we have an argument when connecting to the Blade to choose the type of filter wanted? It would allow you to choose BLADERF_RFIC_RXFIR_BYPASS if you don't want the default one.

There is even BLADERF_RFIC_RXFIR_CUSTOM, unfortunately with the comment "Custom FIR filter (currently unused)".

Cordially