Nuand/bladeRF #746: Channel must be enabled on BladeRFv2 for RX gain changes to register

KarlL2, 2019-07-11:
I opened https://github.com/pothosware/SoapyBladeRF/issues/30 because the gain couldn't be set. It was because the channel wasn't enabled.

A solution was found https://github.com/pothosware/SoapyBladeRF/pull/33 to forcefully enable all channels in the constructor, but then this broke the transmission.

What's your take on this? Can this be fixed in libbladerf, maybe by ensuring the channel is enabled in `setGain` if this doesn't have side effects?

closed 2019-07-25 by Nuand

gisforgirard, 2021-10-30:
i like how they just close the issue without fixing anything, seems typical of bladerf support...

rghilduta, 2021-10-30:
This is not a bug in libbladeRF. It is intended behavior for the API due to specifics of the RFIC. The gain values can only be applied when the RFIC is powered on. An application calling libbladeRF should be mindful of the APIs requirements and capabilities.

Nuand, 2021-10-30:
@gisforgirard can you please provide information on what you mean by "typical"?

Nuand, 2021-10-30:
> i like how they just close the issue without fixing anything, seems typical of bladerf support...

Can you please provide us with the information on which this assertion is based?