Nuand/bladeRF #748: I feel wrong that picture, XB200 connection

takurx, 2019-07-24:
Hi, I feel wrong that picture of XB200 Transverter board's example connection in following wiki.
https://github.com/Nuand/bladeRF/wiki/Getting-Started:-XB200-Transverter-Board#Example_Connection
Is uploaded picture fine?
[image]
Thanks.

robertghilduta, 2019-07-24:
Please follow the written instructions on the page. It appears as if your wiring looks good. Please contact us at bladeRF@nuand.com if you have any questions or concerns.

robertghilduta, 2019-07-24:
Please follow the written instructions on the page. Please contact us at bladeRF@nuand.com if you have any questions or concerns.

closed 2019-07-24 by robertghilduta