Nuand/bladeRF #749: bladeRF.sync_rx(buf, num_samples) Timeout issue

rsemple621, 2019-08-13:
I am attempting to transfer a digital signal across the TX and RX antennas of the Nuand BladeRF.

As far as I can tell right now, it looks like I am appropriately sending the signals across the TX antenna, however, when I go to receive the numpy array through the RX antenna using asynchronous threads, I am receiving a timeout error when attempting to receive the signal.

TRANSFER:
device.sync_config(layout=_bladerf.ChannelLayout.TX_X1,
                           fmt=_bladerf.Format.SC16_Q11,
                           num_buffers=16,
                           buffer_size=8192,
                           num_transfers=8,
                           stream_timeout=3500)
ch.enable = True
buf = numpy.array(data, dtype=complex)
device.sync_tx(buf, len(data))

RECEIVE:
device.sync_config(layout         = _bladerf.ChannelLayout.RX_X1,
                           fmt            = _bladerf.Format.SC16_Q11,
                           num_buffers    = 16,
                           buffer_size    = 8192,
                           num_transfers  = 8,
                           stream_timeout = 3500)
ch.enable = True
buf = numpy.array([],dtype=complex)
device.sync_rx(buf, num_samples)

ERROR:
[ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1083] Transfer timed out for buffer 0x7fe5880018c0
[ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1083] Transfer timed out for buffer 0x7fe5880098d0
[ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1083] Transfer timed out for buffer 0x7fe5880118e0
[ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1083] Transfer timed out for buffer 0x7fe5880198f0
[ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1083] Transfer timed out for buffer 0x7fe588021900
[ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1083] Transfer timed out for buffer 0x7fe588029910
[ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1083] Transfer timed out for buffer 0x7fe588031920
[ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1083] Transfer timed out for buffer 0x7fe588039930

>info
  Board:                    Nuand bladeRF (bladerf1)
  Serial #:                 hidden
  VCTCXO DAC calibration:   0x8864
  FPGA size:                115 KLE
  FPGA loaded:              yes
  Flash size:               32 Mbit
  USB bus:                  3
  USB address:              6
  USB speed:                Hi-Speed
  Backend:                  libusb
  Instance:                 0

>version
  bladeRF-cli version:        1.7.1-git-896d243
  libbladeRF version:         2.2.0-git-896d243
  Firmware version:           2.3.2
  FPGA version:               0.10.2 (configured by USB host)

I'm sure this is something very simple that I am misunderstanding about the structure, however, I have been overlooking it now for over a week. Any suggestions?

rsemple621, 2019-08-22:
Has this portion of the bladeRF library been fixed yet?

Failure counts
--------------------------------------------
        sampling 0
        lpf_mode 0
   enable_module 0
        loopback 0
          rx_mux 0
           xb200 0
      correction 0
      samplerate 0
       bandwidth 0
            gain 0
       frequency 0
         threads 192133

rghilduta, 2019-08-23:
Can you try to run bladeRF-cli -i and see if it works? Are you able to run this on a superspeed USB interface?

rsemple621, 2019-08-23:
yes, I can access the bladeRF command line interface without any issues, can receive garbage from there without errors, and can supposedly transmit without errors through that interface. However, I am going through the backend code, and the exact line of code that is failing for me is line 302 in the sync.c file found at /bladeRF/host/libraries/libbladeRF/src/streaming/sync.c. 

The line is copied and pasted below for your viewing. By my understanding of the library, it looks like the pthread is timing out because it is trying to do something to a buffer that isn't ready yet? Still working on my end to figure it out.

status = pthread_cond_timedwait(&b->buf_ready, &b->lock, &timeout);

rsemple621, 2019-08-23:
I am attempting (in python code) to transfer a file over the TX channel of my bladeRF x115 device, and then receive the sent file over the RX channel afterwards, however, it is failing in the receive portion of the code at that line, though it is picking up data there. I did see a comment in the backend libusb.c code, that follows as such. Do you think this could have an affect on the functionality of the receive portion?

    /* FIXME We have an inherent issue here with lock ordering between
     *       stream->lock and libusb's underlying event lock, so we
     *       have to drop the stream->lock as a workaround.
     *
     *       This implies that a callback can potentially execute,
     *       including a callback for this transfer. Therefore, the transfer
     *       has been setup and its metadata logged.
     *
     *       Ultimately, we need to review our async scheme and associated
     *       lock schemes.
     */

rghilduta, 2019-08-23:
So bladeRF-cli -i works well for receiving? If bladeRF-cli -i can be used to successfully receive samples, it is a good indicator that libbladeRF is functioning as expected. If bladeRF-cli works, it might be indicative of configuration issues when calling the libbladeRF bindings. 

Try to use the RX settings used by bladeRF-cli -i in configurations, you can query these by running "rx" is bladeRF-cli -i.

rsemple621, 2019-08-23:
I had done that before, but just tried it again with no success (same errors).

It is unclear as to whether the receive command in the bladeRF-cli -i actually works. Yes, it creates samples in the samples.csv file when I go through the "verify functionality portion" of the bladeRF documentation, but the actual device generated data when there wasn't an antenna on the device...

rghilduta, 2019-08-23:
What does the spectrum look like when a 50ohm terminator is attached to the RX port? This sounds like a device usage issue and not a libbladeRF problem, please contact email support at bladeRF@nuand.com or use the forums.

closed 2019-08-23 by rghilduta

rsemple621, 2019-08-23:
I'm quite positive that there is an issue with your libbladeRF library. This is the exact line of code that fails, which goes into your library. device.sync_rx is passed an empty bytearray rx_samples, and is passed the number that it is attempting to receive. Like I said, the transfer function worked fine, but the receive is timing out, and then passing successful statuses back through the backend code. Wanna explain that to me?

`device.sync_rx(rx_samples, num)`

rghilduta, 2019-08-23:
As a first step, can you please confirm you are able to receive and transmit samples using bladeRF-cli -i? What do the received samples look like? If bladeRF-cli -i is able to function without problems, it is very indicative that the issue is not libbladeRF.

Are the sampling rate, bandwidth, and frequency set to valid settings prior to the bladeRF's channel being enabled?

rsemple621, 2019-08-23:
I am able to receive and transmit through the CLI interface without errors. I have already mentioned that a few times, and yes, I have set all of those values to valid settings prior to enabling the channel, as shown below:
        rx_channel = b.Channel(rx_ch)
        rx_channel.frequency = 3000000000
        rx_channel.sample_rate = 1000000
        rx_channel.bandwidth = 2000000