Nuand/bladeRF #751: rf_blade_imp.c.456: RX failed: Operation timed out

allanjinhui, 2019-08-18:
I run the srsLTE UE on the BladeRF, and always meet this "time out" error.

----------------------------------------------------------------------
Opening bladeRF...
Set RX sampling rate 1.92 Mhz, filter BW: 1.92 Mhz
Waiting PHY to initialize ... done!
Attaching UE...
Searching cell in DL EARFCN=3400, f_dl=2685.0 MHz, f_ul=2565.0 MHz
set RX frequency to 2684999998
set TX frequency to 2564999998
Set RX sampling rate 1.92 Mhz, filter BW: 1.92 Mhz
.
Found Cell:  Mode=FDD, PCI=1, PRB=50, Ports=1, CFO=-4.0 KHz
Set RX sampling rate 15.36 Mhz, filter BW: 12.29 Mhz
/srsLTE/lib/src/phy/rf/rf_blade_imp.c.456: RX failed: Operation timed out

SYNC:  Receiving from radio while in IDLE_RX
/srsLTE/lib/src/phy/rf/rf_blade_imp.c.456: RX failed: Operation timed out

/srsLTE/lib/src/phy/ue/ue_sync.c.727: Error receiving samples

/srsLTE/lib/src/phy/rf/rf_blade_imp.c.456: RX failed: Operation timed out

SYNC:  Receiving from radio while in IDLE_RX
Set RX sampling rate 1.92 Mhz, filter BW: 1.92 Mhz

sodaBitters, 2019-11-13:
This issue still occurs with the current srsLTE and bladeRF builds.. I don't believe @xavierarteaga patch is still applicable. I get the same error for srsUE as @allanjinhui :

Opening bladeRF...
Set RX sampling rate 1.92 Mhz, filter BW: 1.92 Mhz
Waiting PHY to initialize ... done!
Attaching UE...
Searching cell in DL EARFCN=3400, f_dl=2685.0 MHz, f_ul=2565.0 MHz
set RX frequency to 2684999998
set TX frequency to 2564999998
Set RX sampling rate 1.92 Mhz, filter BW: 1.92 Mhz
/srsLTE/lib/src/phy/rf/rf_blade_imp.c.428: RX failed: Operation timed out; nsamples=9600;

/srsLTE/lib/src/phy/ue/ue_sync.c.728: Error receiving samples

/srsLTE/lib/src/phy/ue/ue_cell_search.c.309: Error calling srslte_ue_sync_work()

/srsLTE/lib/src/phy/ue/ue_cell_search.c.268: Error searching cell

/srsLTE/lib/src/phy/rf/rf_blade_imp.c.428: RX failed: Operation timed out; nsamples=1920;

SYNC:  Receiving from radio while in IDLE_RX
/srsLTE/lib/src/phy/rf/rf_blade_imp.c.428: RX failed: An unexpected error occurred; nsamples=1920;

SYNC:  Receiving from radio while in IDLE_RX
/srsLTE/lib/src/phy/rf/rf_blade_imp.c.428: RX failed: An unexpected error occurred; nsamples=1920;

SYNC:  Receiving from radio while in IDLE_RX
/srsLTE/lib/src/phy/rf/rf_blade_imp.c.428: RX failed: An unexpected error occurred; nsamples=1920;

lcr1017, 2020-03-31:
same problem

closed 2023-11-30 by rthomp10

rthomp10, 2023-11-30:
This issue should be further diagnosed within SRS-LTE before bringing it over to the bladeRF issue tracker. Narrowing the issue down to a reproducible set of steps outside of SRS-LTE is when the issue should be reopened here.