Nuand/bladeRF #752: [hdl] build on windows

vitoller, 2019-08-26:
I am having difficluties Building BladeRF-micro on a Windows machine. As I could not figure out why the build Fails, I hoped that you can help.

* Windows 10
* Quartus Prime 17.1

Steps to reproduce:

1. git clone --recursive https://github.com/Nuand/bladeRF.git

2. As the NiosII Environment did not come with a cmake, I had to install one manually. I choose to download the one from http://cmake.org - Is there a better Version to use?

3. in hdl/quartus execute: **./build_bladerf.sh -b bladeRF-micro -s A4 -r hosted**

This eventually Fails with:

```
##########################################################################
    Building BSP and bladeRF-micro application...
##########################################################################

Skipping creating Nios BSP, settings.bsp already exists
[BSP build complete]
makefile:81: warning: overriding commands for target `../../../../../quartus/work/bladerf-micro-A4-hosted/libad936x/libad936x.a'
../../../common/bladerf/software/bladeRF_nios/Makefile:937: warning: ignoring old commands for target `../../../../../quartus/work/bladerf-micro-A4-hosted/libad936x/libad936x.a'
Info: Building ../../../../../quartus/work/bladerf-micro-A4-hosted/bladeRF_nios_bsp
C:/intelFPGA/17.1/nios2eds/bin/gnu/H-x86_64-mingw32/bin/make --no-print-directory -C ../../../../../quartus/work/bladerf-micro-A4-hosted/bladeRF_nios_bsp
[BSP build complete]
Info: Generating libad936x Makefile
-- QUARTUS_ROOTDIR = C:/intelFPGA/17.1/quartus
-- libad936x: Configuring build...
-- libad936x: building with software mul/mulx/div operations
-- FindPatcher: found C:/intelFPGA/17.1/quartus/bin64/cygwin/bin/patch.exe
-- libad936x: Applying patch: 0001-0bba46e-nuand-modifications.patch
CMake Error at CMakeLists.txt:200 (message):
  Failed to apply 0001-0bba46e-nuand-modifications.patch:

-- Configuring incomplete, errors occurred!
See also "C:/Users/eyermann/Documents/AMS-Fehlerortung/bladeRF/hdl/quartus/work/bladerf-micro-A4-hosted/libad936x/CMakeFiles/CMakeOutput.log".
See also "C:/Users/eyermann/Documents/AMS-Fehlerortung/bladeRF/hdl/quartus/work/bladerf-micro-A4-hosted/libad936x/CMakeFiles/CMakeError.log".
make: *** [../../../../../quartus/work/bladerf-micro-A4-hosted/libad936x/Makefile] Error 1
```

Attached are the CMake-Log files. 

[CMakeOutput.log](https://github.com/Nuand/bladeRF/files/3539606/CMakeOutput.log)
[CMakeError.log](https://github.com/Nuand/bladeRF/files/3539607/CMakeError.log)

rghilduta, 2019-08-26:
The bladeRF repository should not be cloned with "--recursive" ; it might be worth starting from a fresh clone.

closed 2019-08-26 by rghilduta

vitoller, 2019-08-26:
The  [README.md](https://github.com/Nuand/bladeRF/blob/master/README.md)
contains this line:  `git clone --recursive https://github.com/Nuand/bladeRF.git`

Maybe it should be updated?

I used the srouce from:
[https://github.com/Nuand/bladeRF/archive/2019.07.tar.gz](https://github.com/Nuand/bladeRF/archive/2019.07.tar.gz)

However, when I build it, it eventually fails:

[BSP build complete]
make: *** No rule to make target `../../../../../../thirdparty/analogdevicesinc/no-OS/ad9361/sw/*', needed by `../../../../../quartus/work/bladerf-micro-A4-hosted/libad936x/Makefile'.  Stop.

How would I get the correct content of "thirdparty/analogdevicesinc/no-OS"?

robertghilduta, 2019-08-27:
> Thank you for the Explanation.
> I did:
>    cd bladeRF/hdl/quartus/
> The build did not complete with this message:

Ah try going into the bladeRF repository's host/build/ and running cmake
from there first, then once cmake finishes successfully go continue in
hdl/quartus/

vitoller, 2019-08-27:
Thank you for the Explanation. 
I did:

 * git clone https://github.com/Nuand/bladeRF.git

 * cd bladeRF/hdl/quartus/

 * ./build_bladerf.sh -b bladeRF-micro -s A4 -r hosted

The build did not complete with this message:

```
[BSP build complete]
make: *** No rule to make target `../../../../../../thirdparty/analogdevicesinc/no-OS/ad9361/sw/*', needed by `../../../../../quartus/work/bladerf-micro-A4-hosted/libad936x/Makefile'.  Stop.
```

vitoller, 2019-08-27:
OK, I get it.

Unfortunately I do not have a C Compiler installed, which I think is required to run this cmake file.

Is there another way to get the correct content of this Folder?

vitoller, 2019-08-27:
Our IT is quite strict About installing Windows Compilers. 

I am using the NIOS II Shell - which compiles Code for the NIOS processor on the FPGA just fine.

Building hdl/quartus from **2018.10-rc1** worked like a charm

vitoller, 2019-10-08:
OK, I managed to do some more work on this.

1. I think the patch Fails due to DOS / Unix line endings. 
In "hdl/fpga/ip/analogdevicesinc/no_OS/CMakeLists.txt" I changed:

```
        COMMAND ${Patcher_EXECUTABLE} -p3
to:
        COMMAND ${Patcher_EXECUTABLE} -p3 --binary -l
```

that seems to work.

2. I am having difficulties with cmake in the NIOS II Command Shell.
(C:... versus /cygdrive/c)
Therefore I do have two Questions:
* Which Shell do you use? 
* Which cmake have you installed?

Did you manage to install cygwin-cmake? How did you do that?

vbursucianu, 2020-01-27:
Currently facing the same issues on Windows.
I'm calling this line using the Altera Nios2 Command Shell:
`./build_bladerf.sh -b bladeRF-micro -r hosted -s A9`
and it crashes with this line:
```
[BSP build complete]
make: *** No rule to make target `../../../../../../thirdparty/analogdevicesinc/no-OS/ad9361/sw/*', needed by `../../../../../quartus/work/bladerf-micro-A9-hosted/libad936x/Makefile'.  Stop.
```
I've tested the last release : 2019.07 (2019-07-31).

Is there any final solution?

vbursucianu, 2020-01-27:
Solved it. I recloned the project using :
`git clone --recursive https://github.com/Nuand/bladeRF.git`

vbursucianu, 2020-01-30:
I had the same issue on Windows 10 and this patch:
```
        COMMAND ${Patcher_EXECUTABLE} -p3
to:
        COMMAND ${Patcher_EXECUTABLE} -p3 --binary -l
```
in "hdl/fpga/ip/analogdevicesinc/no_OS/CMakeLists.txt"

It solved my problem!

aamrhar-lassena, 2022-02-17:
Thank you @vbursucianu! you are a life saver. I wasted days on this issue. Your solution worked.