Nuand/bladeRF #755: bladerf_get_gain_modes() does not work anymore in 2019.07

f4exb, 2019-09-28:
Hello,

I am trying to implement latest vesion 2019.07 of libbladerf and this call does not work anymore:

```c++
        int n = bladerf_get_gain_modes(m_dev, BLADERF_CHANNEL_RX(0), 0);

        if (n < 0)
        {
            qCritical("DeviceBladeRF2::getGainModesRx: Failed to get the number of Rx gain modes: %s", bladerf_strerror(n));
            return 0;
        }
```
It prints out:
```
DeviceBladeRF2::getGainModesRx: Failed to get the number of Rx gain modes: Invalid operation or parameter
```
What is the new way to call bladerf_get_gain_modes ?

f4exb, 2019-09-28:
Edit: it appears that there is no need to make a call to get only the number of gains. Gains could be queried directly. However the documentation of bladerf_get_gain_modes still says:
```
This function may be called with NULL for modes to determine the number of gain modes supported.
```
See: https://nuand.com/libbladeRF-doc/v2.2.1/group___f_n___g_a_i_n.html#ga5350f1868a06dc92bef4ea0f07914f42

rthomp10, 2026-08-28:
Fixed since commit 9f2f09f

closed 2026-08-28 by rthomp10