Nuand/bladeRF #756: Failed to open bladeRF device with osmocom

wilecoyote2019, 2019-10-02:
Hi!
I'm trying to use the bladeRf xa4 on ubuntu 18.04.03 LTS. 

It came with gnuradio 3.7.11 and gr-osmosdr 0.1.4.

I installed libbladerf  from git.

I can access the bladerf by bladerf-cli -i : 

oam@oam-G3-3579:~/bladeRF/host/build$ bladeRF-cli -i 
bladeRF> info

  Board:                    Nuand bladeRF 2.0 (bladerf2)
  Serial #:                 
  VCTCXO DAC calibration:   0x2023
  FPGA size:                49 KLE
  FPGA loaded:              yes
  Flash size:               32 Mbit
  USB bus:                  2
  USB address:              4
  USB speed:                SuperSpeed
  Backend:                  libusb
  Instance:                 0

bladeRF> version

  bladeRF-cli version:        1.8.0-git-f03d8433
  libbladeRF version:         2.2.0-2018.12-rc3-2-ppabionic

  Firmware version:           2.3.2
  FPGA version:               0.10.2 (configured by USB host)

But when I try to access by osmocom, it cannot open the device :
sudo osmocom_fft -v -a bladerf=0 -s 2500000 -f 446000000
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

Warning: failed to XInitThreads()
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.11
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Opening nuand bladeRF with device identifier string: "*:instance=0"

FATAL: [bladeRF source] Failed to open bladeRF device *:instance=0

Trying to fill up 1 missing channel(s) with null source(s).
This is being done to prevent the application from crashing
due to gnuradio bug #528.

Source has no sample rates (wrong device arguments?).

Thanks for your help!

rghilduta, 2019-10-03:
Is bladeRF-cli closed when you try to use the bladeRF in another program? Please note that only one program at a time can use the bladeRF.

wilecoyote2019, 2019-10-03:
Hi,
Yes bladerf-cli was closed.

vbursucianu, 2020-02-01:
Works for me using : 
gr-osmosdr 0.1.5 (0.1.5) and gnuradio 3.7.13.5
Did you install gr-osmosdr from source?
```
git clone git://git.osmocom.org/gr-osmosdr
cd gr-osmosdr/
mkdir build
cd build/
cmake ../
make
sudo make install
sudo ldconfig
```
When running cmake you should see `nuand bladeRF` inside the enabled component list:
```
-- ######################################################
-- # Gnuradio enabled components                         
-- ######################################################
--   * Python support
--   * Osmocom IQ Imbalance Correction
--   * FUNcube Dongle
--   * FUNcube Dongle Pro+
--   * IQ File Source & Sink
--   * RTLSDR TCP Client
--   * nuand bladeRF
--   * RFSPACE Receivers
--   * SoapySDR support
--   * Red Pitaya SDR
```