Nuand/bladeRF #757: Error probing device: SoapySDR::Device::make() no match

wudave, 2019-10-13:
I am a newbie in SDR.  Recently purchased bladeRF 2.0 micro xA4.  Following the instructions to install SoapySDR bladeRF plugin from https://github.com/pothosware/SoapyBladeRF/wiki, everything seemed to be installing well but when tried to detect bladeRF, got below error.  Does SoapySDR support bladeRF 2.0 micro xA4?  Did I miss any critical steps?

$ SoapySDRUtil --probe="driver=bladerf"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=bladerf
Error probing device: SoapySDR::Device::make() no match

Here are bladeRF info.
bladeRF> version

  bladeRF-cli version:        1.8.0-MacPorts-20191001-f03d8433
  libbladeRF version:         2.2.1-MacPorts-20191001-f03d8433

  Firmware version:           2.3.2
  FPGA version:               0.11.0 (configured by USB host)

bladeRF> info

  Board:                    Nuand bladeRF 2.0 (bladerf2)
  Serial #:                 f4897f2dd81745fca98a93f0ed81571a
  VCTCXO DAC calibration:   0x1fb1
  FPGA size:                49 KLE
  FPGA loaded:              yes
  Flash size:               32 Mbit
  USB bus:                  1
  USB address:              2
  USB speed:                SuperSpeed
  Backend:                  libusb
  Instance:                 0

vbursucianu, 2020-02-01:
Got the same issue....i'm looking for a solution

vbursucianu, 2020-02-01:
The solution for me was to install the Soapy Plugin for Bladerf:
https://github.com/pothosware/SoapyBladeRF/wiki
```
git clone https://github.com/pothosware/SoapyOsmo.git
cd SoapyOsmo
mkdir build
cd build
cmake ..
make
sudo make install
```
When calling : `SoapySDRUtil --find`
```
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Found device 0
  backend = libusb
  device = 0x02:0x04
  driver = bladerf
  instance = 0
  label = BladeRF #0 [310b77a1..95529a6c]
  serial = 310b77a1fc67404780de2c5395529a6c
```

MartinPolytech, 2021-02-03:
Hello,

We have the same error with et HackRF however we tried installing the plugins as you said but the problem is still not fixed. 

When calling SoapySDRUtil, we can see that the drivers are installed correctty :

```
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Found device 0
  device = HackRF One
  driver = hackrf
  label = HackRF One #0 88869dc3352821b
  part_id = a000cb3c006a4767
  serial = 0000000000000000088869dc3352821b
  version = 2017.02.1
```
Do you have any insights on how we could fix our issue ? 

Have a good day :)

surangis, 2021-05-13:
you have to connect the BladeRF through the DEVICE tab on the Ubuntu (if its ubuntu) , i got the same error, then when i connect thorough Devices, it worked

surangis, 2021-05-13:
My issue is I am still very new to SoapySDR, I have installed SoapySDR and then SoapyRTLSDR. i can check using the command SOapySDRUtil --probe, it identifies the rtl drivers and everything is fine.
my question is now how can i connect and load the samples from the SDR to a file using SOapySDR. 
is there is command line. like for eg- rtl_sdr to read the samples.

yangfan852219770, 2021-05-26:
I have the same problem with you, but my device is LimeSDR Mini. When I ran the program in physical machine Ubuntu 20.04 LTS, not in the VMware, it worked normally. I don't know why.

patja, 2022-01-01:
I am also experiencing this problem using the Python API and an Airspy HF+.  

SoapySDRUtil --probe and --find both see the AirspyHF.

I saw a reference elsewhere to the possibility that it is a problem with inconsistent case expectations.  driver AirspyHF is what is reported by --probe but driver airspyhf is what is reported by --info

mhgue, 2023-01-13:
Using Ubuntu 22.04.1 LTS with gnuradio installed as described in https://github.com/gnuradio/pybombs I had the same Problem with `SoapySDRUtil --find` finding no device (neither RTL-SDR nor HackRF). Installing soapysdr-tools from APT is working fine. So my solution was:
```
# Remove soapy library from within prefix-3.10
cd ~/prefix-3.10/src/soapysdr/build
make uninstall
# Install developer version with APT
apt-get install libsoapysdr-dev
# Recompile GNU radio
cd ~/prefix-3.10/src/gnuradio/build
make uninstall
make clean
make install
```
Recompiling gnuradio is needed, because it otherwise tries to use `libSoapySDR.so.0.8`**-2** instead of `libSoapySDR.so.0.8`, and refused using the APT version even if linked and found. After recompilation the ATP version is used and `gnuradio-companion` works fine with RTL-SDR and HackRF even simultaneously.

austinfong, 2023-01-16:
> Using Ubuntu 22.04.1 LTS with gnuradio installed as described in https://github.com/gnuradio/pybombs I had the same Problem with `SoapySDRUtil --find` finding no device (neither RTL-SDR nor HackRF). Installing soapysdr-tools from APT is working fine. So my solution was:
> 
> ```
> # Remove soapy library from within prefix-3.10
> cd ~/prefix-3.10/src/soapysdr/build
> make uninstall
> # Install developer version with APT
> sudo apt-get install libsoapysdr-dev
> # Recompile GNU radio
> cd ~/prefix-3.10/src/gnuradio/build
> make uninstall
> make clean
> make install
> ```
> 
> Recompiling gnuradio is needed, because it otherwise tries to use `libSoapySDR.so.0.8`**-2** instead of `libSoapySDR.so.0.8`, and refused using the APT version even if linked and found. After recompilation the ATP version is used and `gnuradio-companion` works fine with RTL-SDR and HackRF even simultaneously.

Hi, MG,

Could you help find and share the file "libSoapySDR.so.0.8-2" with me (brianfong@qq.com)?
I am installing this SoaphSDR to CentOS in my pc.
When I run command #SoaphySDRUtil --info
It shows below:
SoaphSDRUtil: error while loading shared libraries: libSoapySDR.so.0.8-2: cannot open shared ....

It looks this file is missing in my pc.
Or I also will try to also get gnuradio installed first. 
Thanks a lot for any suggestion about it.

mhgue, 2023-01-16:
Hi @austinfong,
the error you get means that your tools binary is looking for lib and API version 0.8-2 while the paths in your ld config do not provide it. I do not know the exact difference between the 0.8 version and the 0.8-2 version as used by `gnuradio` if installed with `pybombs`. As far as I found, the 0.8-2 doesn't find the devices, so it makes no sense to send you a copy of my version.

First of all you shall check if you have the missing library somewhere `find / -type f -name libSoapySDR.so.0.8-2`. If you can find it you should add the path to your ld config or try with environment variable `LD_LIBRARY_PATH`. 

Is your `SoaphySDRUtil` from a system package or self compiled? You should check if you have two versions of `SoaphySDRUtil` using differnet library versions `find / -type f -name SoaphySDRUtil` If a binary doesn't fit to the library its has something to do with mixing installations and wrong paths.

Hop it will help.

austinfong, 2023-01-16:
您的来信已收悉，我会尽快拜读并回复，非常感谢！Brian/方