Nuand/bladeRF #762: GNURadio 3.8 Support

sodaBitters, 2019-11-21:
gr-osmosdr is now deprecated and unmaintained. 

I've done quite a bit of experimentation with various packages to get the BladeRF2.0 up and running, and the only way I was able to get the radio to work with up-to-date code is via PYBOMBS:
- pybombs initialize a prefix with latest GNURadio master branch (3.8)
- install bladeRF, gr-soapysdr, and gr-soapyuhd packages
- use a UHD SOURCE block, and set device-args to "driver=bladerf"

This allows access the radio, but it doesn't seem very stable... there are many layers of abstraction. 

Relatedly, there are .grc files in the **/host/examples/gnuradio/** directory that are incompatible with the latest version of gnuradio-companion. They also depend on the deprecated gr-osmosdr package.

sodaBitters, 2019-11-21:
Here is a message from the maintainer of gr-osmosdr, he's included a list of gr-osmosdr forks that may have 3.8 support:
[http://lists.osmocom.org/pipermail/osmocom-sdr/2019-September/001983.html](url)

rghilduta, 2019-11-22:
One option for the bladeRF 2.0 (as well as the bladeRF 1.0) would be to build GNURadio 3.7 by trying
> pybombs config --package gnuradio gitrev maint-3.7
Alternatively, to try gr-osmosdr with GNURadio 3.8, use the default GNURadio 3.8 recipe ( don't specify a gitrev for --package gnuradio ) but point the build script to experimental gnuradio-3.8-dev branch in the experimental fork of gr-osmosdr by trying:
> pybombs config --package gr-osmosdr gitrev gnuradio-3.8-dev

rghilduta, 2019-11-22:
One option for the bladeRF 2.0 would be to build GNURadio 3.7 by trying

> pybombs config --package gnuradio gitrev maint-3.7

Alternatively, to the bladeRF 2.0 try gr-osmosdr with GNURadio 3.8, use the default GNURadio 3.8 recipe ( don't specify a gitrev for --package gnuradio ) but point the build script to experimental gnuradio-3.8-dev branch in the experimental fork of gr-osmosdr by trying:

> pybombs config --package gr-osmosdr gitrev gnuradio-3.8-dev

rghilduta, 2019-11-22:
One option for the bladeRF 2.0 would be to build GNURadio 3.7 by trying

> pybombs config --package gnuradio gitrev maint-3.7

Alternatively, to the bladeRF 2.0 try gr-osmosdr with GNURadio 3.8, use the default GNURadio 3.8 recipe ( don't specify a gitrev for --package gnuradio ) but point the build script to experimental gnuradio-3.8-dev branch in the experimental fork of gr-osmosdr by trying:

> pybombs config --package gr-osmosdr gitrev dev-gr-3.8

rghilduta, 2019-11-22:
One option for the bladeRF 2.0 would be to build GNURadio 3.7 by trying

> pybombs config --package gnuradio gitrev maint-3.7

Alternatively, to the bladeRF 2.0 try gr-osmosdr with GNURadio 3.8, use the default GNURadio 3.8 recipe ( don't specify a gitrev for --package gnuradio ) but point the build script to experimental dev-gr-3.8 branch in the experimental fork of gr-osmosdr by trying:

> pybombs config --package gr-osmosdr gitrev dev-gr-3.8

sodaBitters, 2019-11-22:
Robert - thanks for the response! I didn't realize Nuand had its own gr-osmosdr fork.

After installing gnuradio and bladeRF packages in PyBOMBS, I'm compiling it by modifying the recipe file for gr-osmosdr to yours:
```category: common
depends:
- uhd
- rtl-sdr
- gnuradio
- bladeRF
- soapysdr
description: Interface API independent of the underlying radio hardware
gitbranch: dev-gr-3.8
inherit: cmake
# Let's always build this from source, not binaries
source: https://github.com/Nuand/gr-osmosdr.git
```
The cmake command nearly configures, but is held up by this error:
```
-- ######################################################
-- # gr-osmosdr enabled components                         
-- ######################################################
--   * Python support
--   * Osmocom RTLSDR
--   * Ettus USRP Devices
--   * nuand bladeRF
--   * RFSPACE Receivers
--   * SoapySDR support
--   * Red Pitaya SDR
-- 
-- ######################################################
-- # gr-osmosdr disabled components                        
-- ######################################################
--   * Osmocom IQ Imbalance Correction
--   * sysmocom OsmoSDR
--   * IQ File Source & Sink
--   * RTLSDR TCP Client
--   * Osmocom MiriSDR
--   * HackRF & rad1o Badge
--   * AIRSPY Receiver
--   * FreeSRP support
-- 
-- Building for version: 46204cea / 0.1.5git
-- Using install prefix: /home/eddie/gr_prefix
-- Configuring done
CMake Error at lib/CMakeLists.txt:272 (ADD_LIBRARY):
  Target "gnuradio-osmosdr" links to target "gnuradio::gnuradio-blocks" but
  the target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at /home/eddie/gr_prefix/lib64/cmake/gnuradio/UseSWIG.cmake:573 (add_library):
  Target "osmosdr_swig" links to target "gnuradio::gnuradio-blocks" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?
Call Stack (most recent call first):
  /home/eddie/gr_prefix/lib64/cmake/gnuradio/GrSwig.cmake:133 (swig_add_library)
  swig/CMakeLists.txt:53 (GR_SWIG_MAKE)
```

I'll continue to troubleshoot; please let me know if you can think of anything that would cause this.

sodaBitters, 2019-11-22:
To fix it, I had to change [CMakeLists.txt line 170](https://github.com/Nuand/gr-osmosdr/blob/46204cea1dc5d338c060d568490f84a9cb5dee62/CMakeLists.txt#L170) from:
`find_package(gnuradio-blocks PATHS ${GR_PREFIX}/lib/cmake/gnuradio/)`
to:
`find_package(gnuradio-blocks PATHS ${GR_PREFIX}/lib64/cmake/gnuradio/)`

and the source builds successfully from GNURadio 3.8 master branch using PyBOMBS.

sodaBitters, 2019-11-22:
I'm not sure if this is something you'd like to update in the CMake file, but I'll close the issue here since this problem is technically related to nuand/gr-osmosdr repository. Thanks again, Robert.

closed 2019-11-22 by sodaBitters