Nuand/bladeRF #778: BladeRF becomes unusable after changing sample rate

andrepuschmann, 2020-04-25:
Hey,
I am using the bladeRF 2.0 and I am seeing difficulties when changing the sample rates during operation. This happens with Rx only as well as with Rx/Tx. I've also tried to stop the Rx streaming operation before changing the rate but it doesn't seem to help. I remember this actually working before so I was wondering if that might be a regression or if I am doing something wrong here?

The issue can easily be reproduced with srsLTE and the PDSCH example where we scan for the LTE cell at 1.92 Msamps/s before changing to the actual cell bandwidth, 11.52 Msamps/s for example for a 10 MHz LTE cell. See log below. If that happens, I actually need to `kill -9` the application.

I've swapped the cable and also tried on two different machines, different USB ports, etc. 

Normal streaming is ok, as long as I don't change the sample rate.

Any ideas what might be causing the issue?

Thanks
Andre

```
$ ./lib/examples/pdsch_ue -f 816e6 -I soapy
Opening RF device with 1 RX antennas...
Soapy has found device #0: backend=libusb, device=0x02:0x1B, driver=bladerf, instance=0, label=BladeRF #0 [ab356e9a..f965f068], serial=ab356e9ac74547478d292539f965f068,
Selecting Soapy device: 0
[INFO] bladerf_open_with_devinfo()
[INFO] bladerf_get_serial() = ab356e9ac74547478d292539f965f068
[INFO] setSampleRate(Rx, 0, 4.000000 MHz), actual = 4.000000 MHz
[INFO] setSampleRate(Tx, 0, 4.000000 MHz), actual = 4.000000 MHz
Setting up Rx stream with 1 channel(s)
[INFO] numBuffs=32
[INFO] bufSize=4096
[INFO] numXfers=16
Setting up Tx stream with 1 channel(s)
[INFO] numBuffs=32
[INFO] bufSize=4096
[INFO] numXfers=16
[INFO] setSampleRate(Rx, 0, 1.920000 MHz), actual = 1.920000 MHz
[INFO] setSampleRate(Tx, 0, 1.920000 MHz), actual = 1.920000 MHz
Available device sensors:
 - RFIC_TEMP
Available sensors for Rx channel 0:
 - PRE_RSSI
 - SYM_RSSI
State of gain elements for Rx channel 0 (AGC supported):
 - full: 69.00 dB
State of gain elements for Tx channel 0 (AGC not supported):
 - dsa: -10.00 dB
Rx antenna set to RX
Tx antenna set to TX
Starting AGC thread...
Tunning receiver to 816.000 MHz
Searching for cell...
[INFO] setSampleRate(Rx, 0, 1.920000 MHz), actual = 1.920000 MHz
0 Found Cell_id:   0 FDD, CP: Normal  , DetectRatio= 0% PSR=0.00, Power=-inf dBm
*Found Cell_id: 190 FDD, CP: Normal  , DetectRatio=100% PSR=5.65, Power=39.7 dBm
 Found Cell_id:   0 FDD, CP: Normal  , DetectRatio= 0% PSR=0.00, Power=-inf dBm
Decoding PBCH for cell 190 (N_id_2=1)
[INFO] setSampleRate(Rx, 0, 1.920000 MHz), actual = 1.920000 MHz
00000000000000Setting sampling rate 11.52 MHz
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:305] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:305] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:305] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:305] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:305] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:305] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
```

I am using the device through Soapy, all compiled from sources.
```
$ SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.0-gf722f9ce
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.8
Module found: /usr/local/lib/SoapySDR/modules0.8/libLMS7Support.so    (20.01.0-0854a51e)
Module found: /usr/local/lib/SoapySDR/modules0.8/libbladeRFSupport.so (0.4.1-1c1e8aa)
```

I am also using the latest bladeRf driver and FPGA I believe.
```
$ bladeRF-cli -e version
[INFO @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:309] Waiting for device to become ready...

  bladeRF-cli version:        1.8.0-git-45521019
  libbladeRF version:         2.2.1-git-45521019

  Firmware version:           2.3.2
  FPGA version:               0.11.0 (configured from SPI flash)

```

robertghilduta, 2020-04-27:
Quick question, do you know if the libbladeRF tuning mode is set to host or
FPGA?
If it's currently FPGA, please try to set it to host.

andrepuschmann, 2020-04-27:
Hey Robert, thanks for the hint. I didn't know about those tuning modes but it turned out that it was set to host by default even though it should be FPGA according to [this](https://nuand.com/libbladeRF-doc/v2.2.1/group___f_n___t_u_n_i_n_g___m_o_d_e.html) with the FPGA/lib version I am using, shouldn't it?

So `export BLADERF_DEFAULT_TUNING_MODE=fpga` does fix the issue for the Rx only app like `pdsch_ue`. But interestingly, the full LTE UE which apparently does Rx and Tx works better with `host` mode albeit printing errors.
```
00000000[INFO] setSampleRate(Rx, 0, 1.920000 MHz), actual = 1.920000 MHz
000000000000000000[INFO] setSampleRate(Tx, 0, 1.920000 MHz), actual = 1.920000 MHz
00.
Found Cell:  Mode=FDD, PCI=190, PRB=50, Ports=2, CFO=0.2 KHz
0[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:305] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
RF status: O=29, U=0, L=1
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:305] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
[ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1089] Transfer timed out for buffer 0x560044d0cb20
000000000000000[INFO] setSampleRate(Rx, 0, 11.520000 MHz), actual = 11.520000 MHz
00000000000000000[INFO] setSampleRate(Tx, 0, 11.520000 MHz), actual = 11.520000 MHz
```

rghilduta, 2020-06-22:
I am still unable to replicate this exactly even after trying with and without the native bladeRF driver, I am wondering, is everything is running on Superspeed ("info" in bladeRF-cli)?

andrepuschmann, 2020-06-22:
Hey Robert,
if you have a chance, can you try to run our srsLTE radio benchmark tool and see if that works for you? It's included in srsLTE 20.04. Try `./lib/src/radio/test/benchmark_radio  -s 11.52e6 -f 801.3e6 -t 10  -x -y` for example.

rghilduta, 2020-06-22:
Ah, that seems able to trigger an issue which is likely either in libbladeRF or the bladeRF HDL. The first failure seems to happen just as the bladeRF's sample rate is changed from 11.52MHz to 1.92MHz. Let me try to see if I can get a better understanding of what's happening within libbladeRF.

andrepuschmann, 2020-06-22:
Sure, thanks for looking at this issue Robert. The `-y` parameter "simulates" changes from 11.52MHz (or any other configured value) to 1.92MHz (lowest LTE bandwidth used for cell search). The purpose it to mimic the UE going into cell search and then back to the actual cell bandwidth. While this isn't done every seconds in the normal UE, it still serves as a good radio (API) benchmark.

alphafox02, 2020-07-14:
Believe I'm having the same issues while trying to setup srsLTE using 2x Micro xA4. Here's a paste from the test you requested above and I'll work backwards from there.

20.04.1

`./benchmark_radio -d bladeRF -s 11.52e6 -f 801.3e6 -t 10 -x -y
Instantiating objects and allocating memory...
Initialising instances...
Opening 1 channels in RF device=bladeRF with args=default
Opening bladeRF...
Set RX sampling rate 1.92 Mhz, filter BW: 1.92 Mhz

Warning: TX gain was not set. Using open-loop power control (not working properly)

set RX frequency to 801299966
Set RX sampling rate 11.52 Mhz, filter BW: 9.22 Mhz
Setting manual TX/RX offset to 0 samples
Start capturing 10000 frames of 11520 samples...
Changing sampling rate to 11.52 Msamps/s
Set RX sampling rate 11.52 Mhz, filter BW: 9.22 Mhz
Setting manual TX/RX offset to 0 samples
Changing sampling rate to 1.92 Msamps/s
Set RX sampling rate 1.92 Mhz, filter BW: 1.92 Mhz
Setting manual TX/RX offset to 0 samples
/usr/src/srsLTE/lib/src/phy/rf/rf_blade_imp.c.434: RX failed: Operation timed out; nsamples=11520;                                                        
                                                                             
/usr/src/srsLTE/lib/src/phy/rf/rf_blade_imp.c.434: RX failed: Operation timed out; nsamples=11520;                                                        
                                                                             
/usr/src/srsLTE/lib/src/phy/rf/rf_blade_imp.c.434: RX failed: Operation timed out; nsamples=11520;                                                        
                                                                             
/usr/src/srsLTE/lib/src/phy/rf/rf_blade_imp.c.434: RX failed: Operation timed out; nsamples=11520;                                                        
                                                                             
/usr/src/srsLTE/lib/src/phy/rf/rf_blade_imp.c.434: RX failed: Operation timed out; nsamples=11520;                                                        
                                                                             
/usr/src/srsLTE/lib/src/phy/rf/rf_blade_imp.c.434: RX failed: Operation timed out; nsamples=11520;
                                                                                                                                                                                             
/usr/src/srsLTE/lib/src/phy/rf/rf_blade_imp.c.434: RX failed: Operation timed out; nsamples=11520;                                                                                           
                                                                                                                                                                                             
/usr/src/srsLTE/lib/src/phy/rf/rf_blade_imp.c.434: RX failed: Operation timed out; nsamples=11520;                                                                                           
                                                                                                                                                                                             
/usr/src/srsLTE/lib/src/phy/rf/rf_blade_imp.c.434: RX failed: Operation timed out; nsamples=11520;                                                                                           
                                                                                                                                                                                             
/usr/src/srsLTE/lib/src/phy/rf/rf_blade_imp.c.434: RX failed: Operation timed out; nsamples=11520;                                                                                           
                                                                                                                                                                                             
/usr/src/srsLTE/lib/src/phy/rf/rf_blade_imp.c.434: RX failed: Operation timed out; nsamples=11520;`

`bladeRF-cli -e version                                                                                                                   
                                                                                                                                                                                             
  bladeRF-cli version:        1.8.0-git-c8320f71-ppafocal                                                                                                                                    
  libbladeRF version:         2.2.1-git-c8320f71-ppafocal                                                                                                                                    
                                                                                                                                                                                             
  Firmware version:           2.2.0-git-47f93fe1                                                                                                                                             
  FPGA version:               0.11.0       `

`######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.7.2-1
API Version: v0.7.1
ABI Version: v0.7
Install root: /usr
Search path:  /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7
[... cut ...]
Module found: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/libHackRFSupport.so  (0.3.3)
Module found: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/libLMS7Support.so    (20.01.0)
Module found: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/libRedPitaya.so      (0.1.1)
Module found: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/libairspySupport.so  (0.1.2)
Module found: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/libaudioSupport.so   (0.1.1)
Module found: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/libbladeRFSupport.so (0.4.1)
`

sparklespdx, 2020-09-05:
I’m having the same issue with the bladerf A4 using master branch from bladerf and srsLTE and the newest firmware and bitstream. Is there any movement on this issue? Would be happy to try a branch.

sparklespdx, 2020-09-22:
I have found further information about this issue.

I set the default sample rate to 11.52MHz in `rf_bladerf_imp.c`. This causes srsUE to set the sample rate to 11.52MHz when the device initializes, then sets it to 1.92MHz, searches for a cell, and then sets it to 11.52MHz again when it finds one.

It seems that the radio functions fine when switching down from 11.52Mhz to 1.92Mhz, but going the other direction it starts timing out.

The version of srsLTE I am using will switch back to cell searching once the device has been in an idle state for long enough, and what's really interesting is when the radio is switched back down to 1.92MHz, it works fine and can receive again. Then when it is switched back up to 11.52MHz, it starts timing out.

```
Opening 1 RF devices with 1 RF channels...
Opening bladeRF...
Set RX sampling rate 11.52 Mhz, filter BW: 9.22 Mhz

Warning: TX gain was not set. Using open-loop power control (not working properly)

Waiting PHY to initialize ... Warning: Failed to create thread with real-time priority. Creating it with normal priority: Success
Warning: Failed to create thread with real-time priority. Creating it with normal priority: Success
Warning: Failed to create thread with real-time priority. Creating it with normal priority: Success
Warning: Failed to create thread with real-time priority. Creating it with normal priority: Resource temporarily unavailable
done!
Attaching UE...
Searching cell in DL EARFCN=3400, f_dl=1935.0 MHz, f_ul=1855.0 MHz
set RX frequency to 1935000062
set TX frequency to 1855000062
Set RX sampling rate 1.92 Mhz, filter BW: 1.92 Mhz
.
.
Found Cell:  Mode=FDD, PCI=393, PRB=50, Ports=4, CFO=1.4 KHz
Set RX sampling rate 11.52 Mhz, filter BW: 9.22 Mhz
SYNC:  Receiving from radio while in IDLE_RX
SYNC:  Receiving from radio while in IDLE_RX
Set RX sampling rate 1.92 Mhz, filter BW: 1.92 Mhz
.
Found Cell:  Mode=FDD, PCI=393, PRB=50, Ports=4, CFO=1.1 KHz
Set RX sampling rate 11.52 Mhz, filter BW: 9.22 Mhz
SYNC:  Receiving from radio while in IDLE_RX
SYNC:  Receiving from radio while in IDLE_RX
Set RX sampling rate 1.92 Mhz, filter BW: 1.92 Mhz
.
Found Cell:  Mode=FDD, PCI=393, PRB=50, Ports=4, CFO=0.6 KHz
Set RX sampling rate 11.52 Mhz, filter BW: 9.22 Mhz
SYNC:  Receiving from radio while in IDLE_RX
SYNC:  Receiving from radio while in IDLE_RX
Set RX sampling rate 1.92 Mhz, filter BW: 1.92 Mhz
.
Found Cell:  Mode=FDD, PCI=393, PRB=50, Ports=4, CFO=1.2 KHz
```

sparklespdx, 2020-09-27:
More interesting data; I am using radio_benchmark for troubleshooting. It seems that if the radio is told to change the sample rate to exactly the same sample rate it already has, it still encounters the error (i.e. changing from 1.92e6 to 1.92e6 triggers the bug.

```
[josh@lappy-mk6 build]$ sudo bash -c "export BLADERF_LOG_LEVEL=debug; export BLADERF_DEFAULT_TUNING_MODE=host; ./lib/src/radio/test/benchmark_radio -d bladerf -s 1.92e6 -f 801.3e6 -t 10 -x -y"
[DEBUG @ host/libraries/libbladeRF/src/init_fini.c:90] libbladeRF 2.2.1-git-b02acdd2-dirty: initializing
Instantiating objects and allocating memory...
Initialising instances...
Opening 1 channels in RF device=bladerf with args=default
Opening bladeRF...
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/libusb.c:189] Bus 002 Device 109: Nuand bladeRF 2.0, serial 8a952f477f514a11a049743f0c98a38b
[DEBUG @ host/libraries/libbladeRF/src/driver/ina219.c:70] Configuration register: 0x019f
[DEBUG @ host/libraries/libbladeRF/src/driver/ina219.c:82] Calibration register: 0xa000
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/common.c:263] Default tuning mode: Host
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:2276] bladerf2_set_tuning_mode: New tuning mode: Host
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:2311] bladerf2_set_tuning_mode: Releasing FPGA RFIC control
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/rfic_fpga.c:251] _rfic_fpga_deinitialize: deinitializing
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:2326] bladerf2_set_tuning_mode: Initializing Host RFIC control
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/rfic_host.c:84] _rfic_host_initialize: initializating
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/rfic_host.c:146] _rfic_host_initialize: initialization complete
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:2558] trim DAC is enabled
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:2591] saving current trim DAC value: 0x1f6b
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:2558] trim DAC is disabled
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:2595] restoring old trim DAC value: 0x1f6b
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:207] _bladerf2_initialize: complete
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:2276] bladerf2_set_tuning_mode: New tuning mode: FPGA
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:2311] bladerf2_set_tuning_mode: Releasing Host RFIC control
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/rfic_host.c:156] _rfic_host_deinitialize: deinitializing
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:2326] bladerf2_set_tuning_mode: Initializing FPGA RFIC control
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/rfic_fpga.c:233] _rfic_fpga_initialize: initializing
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1053] bladerf2_set_sample_rate: enabling 4x decimation/interpolation filters
Set RX sampling rate 1.92 Mhz, filter BW: 1.92 Mhz

Warning: TX gain was not set. Using open-loop power control (not working properly)

set RX frequency to 801299966
Set RX sampling rate 1.92 Mhz, filter BW: 1.92 Mhz
Setting manual TX/RX offset to 0 samples
Start capturing 10000 frames of 1920 samples...
Changing sampling rate to 1.92 Msamps/s
Set RX sampling rate 1.92 Mhz, filter BW: 1.92 Mhz
Setting manual TX/RX offset to 0 samples
Changing sampling rate to 1.92 Msamps/s
Set RX sampling rate 1.92 Mhz, filter BW: 1.92 Mhz
Setting manual TX/RX offset to 0 samples
/home/josh/git/srsLTE/lib/src/phy/rf/rf_blade_imp.c.439: RX failed: Operation timed out; nsamples=1920;
```

dillona, 2020-10-02:
This happens to me with bitstream v0.10.2 but not v0.9.0 (on xA4 from [Nuand](http://www.nuand.com/fpga_images/))

alphafox02, 2020-10-02:
Thanks for the updates. Is it possible to downgrade the bitstream when using the PPA/20.04? Still learning the ins and outs of the bladerf. I’ll do some research and see if I can find out more and try and run it again as the UE.

alphafox02, 2021-02-05:
In recent testing with libbladerf 2.4 from git, latest firmware and fpga , my xA4 and xA9 seems fine in the role of the enodeb or ue with srsLTE (can’t recall if uhd or bladerf driver arg worked better). Also works great with croc hunter when configured to use only the srsLTE bladerf driver.

sbourdeauducq, 2021-02-27:
I can confirm that the problem disappeared after upgrading to the latest FX3 firmware (2021-01-12 - v2.4.0), FPGA (2021-01-12 - v0.12.0) and libbladeRF (git e09200c38) as of today.

yesimxev, 2021-04-18:
> I can confirm that the problem disappeared after upgrading to the latest FX3 firmware (2021-01-12 - v2.4.0), FPGA (2021-01-12 - v0.12.0) and libbladeRF (git [e09200c](https://github.com/Nuand/bladeRF/commit/e09200c38e1b273ebe43cd789839e66b0506aead)).

Unfortunately, the problem does not disappear for me after upgrading to latest, or downgrading to each fpga from v0.12.0 to v0.8.0
```
  bladeRF-cli version:        1.8.0-git-e09200c3
  libbladeRF version:         2.4.0-git-e09200c3

  Firmware version:           2.4.0-git-a3d5c55f
  FPGA version:               0.9.0 (configured by USB host)
```

us-maker, 2021-11-30:
Has any progress been made to this ?  I have tested the most recent FPGA/Firmware/master branch combo and the benchmark still fails with gaps and overruns

bladeRF-cli version:        1.8.0-git-d1c38277
  libbladeRF version:         2.4.1-git-d1c38277

  Firmware version:           2.4.0-git-a3d5c55f
  FPGA version:               0.14.0 (configured from SPI flash)

Runnign on Ubuntu 20.04LTS // VMware with usb 3.0 DRIVERS USB Speed clocked at ~7700-8000 kb/s (with usbtop)

USB Port was also compared to bare metail Ubuntu 20.04 with exact same numbers on a 1 Gig file creation:
dd if=/dev/zero of=/media/user/Verbatim/sparse_file bs=1M count=1024
1024+0 records in
1024+0 records out
1073741824 bytes (1.1 GB, 1.0 GiB) copied, 55.122 s, 19.5 MB/s

USBTOP reports ~100e3 throughput on this port during file creation

sudo bash -c "export BLADERF_LOG_LEVEL=debug; export BLADERF_DEFAULT_TUNING_MODE=host; ./lib/src/radio/test/benchmark_radio -d bladerf -s 1.92e6 -f 801.3e6 -t 10 -x -y"
[DEBUG @ host/libraries/libbladeRF/src/init_fini.c:90] libbladeRF 2.4.1-git-d1c38277: initializing
Instantiating objects and allocating memory...
Initialising instances...
Opening 1 channels in RF device=bladerf with args=default
Available RF device list: bladeRF 
Opening bladeRF...
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/libusb.c:189] Bus 004 Device 013: Nuand bladeRF 2.0, serial 92bfacb990d74ded85cf32a954e35f9f
[DEBUG @ host/libraries/libbladeRF/src/driver/ina219.c:70] Configuration register: 0x019f
[DEBUG @ host/libraries/libbladeRF/src/driver/ina219.c:82] Calibration register: 0xa000
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/common.c:270] Default tuning mode: Host
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:2287] bladerf2_set_tuning_mode: New tuning mode: Host
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:2337] bladerf2_set_tuning_mode: Initializing Host RFIC control
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/rfic_host.c:85] _rfic_host_initialize: initializating
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/rfic_host.c:154] _rfic_host_initialize: initialization complete
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:2569] trim DAC is enabled
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:2602] saving current trim DAC value: 0x1bd7
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:2569] trim DAC is disabled
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:2606] restoring old trim DAC value: 0x1bd7
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:207] _bladerf2_initialize: complete
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1062] bladerf2_set_sample_rate: enabling 4x decimation/interpolation filters
Set RX sampling rate 1.92 Mhz, filter BW: 1.92 Mhz

Warning: TX gain was not set. Using open-loop power control (not working properly)

set RX frequency to 801299966
Set RX sampling rate 1.92 Mhz, filter BW: 1.92 Mhz
Setting manual TX/RX offset to 0 samples
Start capturing 10000 sub-frames of 1920 samples (approx. 10s) ...
Changing sampling rate to 1.92 Msamps/s
Set RX sampling rate 1.92 Mhz, filter BW: 1.92 Mhz
[repeat 9 times]
Finished streaming with 18 gaps, 0 late timestamps, 74 overflows, 0 underflow...
Tearing down...
Failed!
[DEBUG @ host/libraries/libbladeRF/src/init_fini.c:98] libbladeRF 2.4.1-git-d1c38277: deinitializing

yesimxev, 2021-11-30:
My issue is gone since I'm using srsRAN, however I'm not sure with @andrepuschmann tried that or the latest updates from Nuand

us-maker, 2021-12-01:
I have 2 BladeRF xA4 Micro's, used one as spectrum analyzer with gqrx on a cabled connection from the Radio under test(with benchmark_radio executing ) and saw no transmit.  Does benchmark_radio @andrepuschmann  actually transmit with -x?  I tested same frequency with a simple FM modulation (same freq/ sample rate) through GRC and received a good xmit.

us-maker, 2021-12-01:
Can you provide the versions and physical setup you are using?  Thanks for
commenting.

yesimxev, 2022-03-17:
Please flash firmware v2.4.1 and load/flash fpga v0.14.0. They have several fixes, anyway I noticed you try with the latest lib. Fw and libs should always match

devmuroid, 2022-03-18:
I have loaded v0.14.0 fpga but used the 2.4.1 version mentioned above with the latest firmware (latest release is 2.4.0 as far as I see) according to the releases page on [https://github.com/Nuand/bladeRF/releases](url) 

```
* FPGA bitstream                v0.14.0
* FX3 firmware                  v2.4.0
* libbladeRF                    v2.4.1
* bladeRF-cli                   v1.8.0
* MATLAB & Simulink bindings    v1.0.5
* Python bindings               v1.2.1
```

The srsenb starting log is as follows:

```
Opening 1 channels in RF device=soapy with args=default
Soapy has found device #0: backend=libusb, device=0x02:0x11, driver=bladerf, instance=0, label=BladeRF #0 [9ffbb05f..e1e2c2df], serial=9ffbb05ff656470d9fc59235e1e2c2df, 
Selecting Soapy device: 0
[INFO] bladerf_open_with_devinfo()
[INFO] bladerf_get_serial() = 9ffbb05ff656470d9fc59235e1e2c2df
[INFO] setSampleRate(Rx, 0, 4.000000 MHz), actual = 4.000000 MHz
[INFO] setSampleRate(Tx, 0, 4.000000 MHz), actual = 4.000000 MHz
Setting up Rx stream with 1 channel(s)
Setting up Tx stream with 1 channel(s)
[INFO] setSampleRate(Rx, 0, 1.920000 MHz), actual = 1.920000 MHz
[INFO] setSampleRate(Tx, 0, 1.920000 MHz), actual = 1.920000 MHz
Available device sensors: 
 - RFIC_TEMP
Available sensors for Rx channel 0: 
 - PRE_RSSI
 - SYM_RSSI
State of gain elements for Rx channel 0 (AGC supported):
 - full: 71.00 dB
State of gain elements for Tx channel 0 (AGC not supported):
 - dsa: -10.00 dB
Rx antenna set to RX
Tx antenna set to TX

==== eNodeB started ===
Type <t> to view trace
[INFO] setSampleRate(Rx, 0, 5.760000 MHz), actual = 5.760000 MHz
[INFO] setSampleRate(Tx, 0, 5.760000 MHz), actual = 5.760000 MHz
Setting frequency: DL=2680.0 Mhz, UL=2560.0 MHz for cc_idx=0 nof_prb=25
0000000
```

As note, when the TX port is connected to a spectrum analyzer, we see notches at the ends of the LTE signal bandwidth. (while these notches are not there with other SDR cards such as Ettus B200 mini)

Any help is appreciated about the configuration of correctly working srsenb with bladeRFxA9. 

Thanks in advance

yesimxev, 2022-03-18:
You are right, I forgot actually the latest version is 2.4.0. Is your issue gone? As far as I remember, according to@rghilduta , the bandwitdth very end can be ignored if it doesn't affect usage. Can't remember if it can be trimmed or fine tuned

knt141, 2022-03-22:
Hello, experiencing the same issue as @devmuroid:

### Versions
Card: bladeRF 2.0 xA4
bladeRF-cli version:        1.8.0-git-3a411c87
libbladeRF version:         2.4.1-git-4d9b0b87

Firmware version:           2.4.0-git-a3d5c55f
FPGA version:               0.14.0 (configured from SPI flash)

### Benchmark log (FPGA tuning mode, device=bladeRF)
```
$ BLADERF_DEFAULT_TUNING_MODE=fpga ./lib/src/radio/test/benchmark_radio -s 11.52e6 -f 801.3e6 -t 10 -x -y -d bladeRF
Instantiating objects and allocating memory...
Initialising instances...
Opening 1 channels in RF device=bladeRF with args=default
Available RF device list: soapy  bladeRF 
Opening bladeRF...
[INFO @ host/libraries/libbladeRF/src/helpers/version.c:79] Firmware version (v2.4.0) is newer than entries in libbladeRF's compatibility table. Please update libbladeRF if problems arise.
[INFO @ host/libraries/libbladeRF/src/helpers/version.c:103] FPGA version (v0.14.0) is newer than entries in libbladeRF's compatibility table. Please update libbladeRF if problems arise.
Set RX sampling rate 1.92 Mhz, filter BW: 1.92 Mhz

Warning: TX gain was not set. Using open-loop power control (not working properly)

set RX frequency to 801299966
Set RX sampling rate 11.52 Mhz, filter BW: 9.22 Mhz
Setting manual TX/RX offset to 0 samples
Start capturing 10000 sub-frames of 11520 samples (approx. 10s) ...
Changing sampling rate to 11.52 Msamps/s
Set RX sampling rate 11.52 Mhz, filter BW: 9.22 Mhz
Setting manual TX/RX offset to 0 samples
Changing sampling rate to 1.92 Msamps/s
Set RX sampling rate 1.92 Mhz, filter BW: 1.92 Mhz
Setting manual TX/RX offset to 0 samples
/home/user/projects/radio/srsRAN/lib/src/phy/rf/rf_blade_imp.c:534: TX failed: Invalid operation or parameter
Changing sampling rate to 11.52 Msamps/s
Set RX sampling rate 11.52 Mhz, filter BW: 9.22 Mhz
Setting manual TX/RX offset to 0 samples
Changing sampling rate to 1.92 Msamps/s
Set RX sampling rate 1.92 Mhz, filter BW: 1.92 Mhz
Setting manual TX/RX offset to 0 samples
Changing sampling rate to 11.52 Msamps/s
Set RX sampling rate 11.52 Mhz, filter BW: 9.22 Mhz
Setting manual TX/RX offset to 0 samples
Changing sampling rate to 1.92 Msamps/s
Set RX sampling rate 1.92 Mhz, filter BW: 1.92 Mhz
Setting manual TX/RX offset to 0 samples
Changing sampling rate to 11.52 Msamps/s
Set RX sampling rate 11.52 Mhz, filter BW: 9.22 Mhz
Setting manual TX/RX offset to 0 samples
Changing sampling rate to 1.92 Msamps/s
Set RX sampling rate 1.92 Mhz, filter BW: 1.92 Mhz
Setting manual TX/RX offset to 0 samples
Changing sampling rate to 11.52 Msamps/s
Set RX sampling rate 11.52 Mhz, filter BW: 9.22 Mhz
Setting manual TX/RX offset to 0 samples
Changing sampling rate to 1.92 Msamps/s
Set RX sampling rate 1.92 Mhz, filter BW: 1.92 Mhz
Setting manual TX/RX offset to 0 samples
Finished streaming with 114 gaps, 1 late timestamps, 2390 overflows, 0 underflow...
Tearing down...
Failed!
```

### Benchmark log (Host tuning mode, device=bladeRF)
```
 BLADERF_DEFAULT_TUNING_MODE=host ./lib/src/radio/test/benchmark_radio -s 11.52e6 -f 801.3e6 -t 10 -x -y -d bladeRF
Instantiating objects and allocating memory...
Initialising instances...
Opening 1 channels in RF device=bladeRF with args=default
[... 73 lines cut ...]
 - SYM_RSSI
State of gain elements for Rx channel 0 (AGC supported):
State of gain elements for Tx channel 0 (AGC not supported):
Rx antenna set to RX
Tx antenna set to TX
RF device 'soapy' successfully opened

Warning: TX gain was not set. Using open-loop power control (not working properly)

[INFO] setSampleRate(Rx, 0, 11.520000 MHz), actual = 11.520000 MHz
[INFO] setSampleRate(Tx, 0, 11.520000 MHz), actual = 11.520000 MHz
Setting manual TX/RX offset to 0 samples
Start capturing 10000 sub-frames of 11520 samples (approx. 10s) ...
Changing sampling rate to 11.52 Msamps/s
[INFO] setSampleRate(Rx, 0, 11.520000 MHz), actual = 11.520000 MHz
[INFO] setSampleRate(Tx, 0, 11.520000 MHz), actual = 11.520000 MHz
Setting manual TX/RX offset to 0 samples
00000000000000Changing sampling rate to 1.92 Msamps/s
[INFO] setSampleRate(Rx, 0, 1.920000 MHz), actual = 1.920000 MHz
[INFO] setSampleRate(Tx, 0, 1.920000 MHz), actual = 1.920000 MHz
Setting manual TX/RX offset to 0 samples
/home/user/projects/radio/srsRAN/lib/src/phy/rf/rf_soapy_imp.c:1001: Couldn't write all samples after 100 trials.
/home/user/projects/radio/srsRAN/lib/src/phy/rf/rf_soapy_imp.c:1001: Couldn't write all samples after 100 trials.
... continues for ages.
```

grmono, 2022-08-27:
Is their any update on this issue or any combination of firmware and libbladerf version that can support the srsue ?

yesimxev, 2022-08-28:
Usb3.0, latest fpga, libs and srsran did the trick for others

grmono, 2022-08-28:
> Usb3.0, latest fpga, libs and srsran did the trick for others

Unfortunately that doesn't work for me, i am using USB3 for sure on ubuntu 22.04 with the latest fpga / bitstream firmware and libbladerf library my bladerf board is the xa4 and i am also using the latest version of srsran.

The issue is exactly as described here , pdsch_ue is working fine for me but srsue is stuck once it tries to switch down to 1.92Mhz also the benchamrk test fails with all drivers (soapy,bladerf), i have tried the following so far i made changes to srsue rf_blade_impl.c to change 1.92Mhz to 11.52e6 and also set srate in ue.conf to 11.52e6 this seems to at least help the UE to proceed and attach but this doesn't work on all earfcns only on specific earfcns such as 2850 but it fails on 1350/6400 etc.

Also even tho the UE seems to attach and go to RRC IDLE in the ue_mac.pcap the ue doesn't actually capture correct info since the packets in the pcap are garbage and cannot be analyzed correctly, i think the only way for srsue to actually work correctly is to actually set the 1.92Mhz rate as originally intended but this bug still seems to apply for the bladerf lib since if the rate is set from 11.52Mhz to 1.92Mhz the boards times out and is unresponsive.

alphafox02, 2022-08-28:
The one person that told me the xa4 worked as a UE also said they added this to the arguments --rf.srate=11520000

I’ll follow up and see what else they used.

grmono, 2022-08-29:
> The one person that told me the xa4 worked as a UE also said they added this to the arguments --rf.srate=11520000
> 
> I’ll follow up and see what else they used.

I tried this aswell and this works only for specific earfcns such as the 2850 that work with this rate but it does not work on other earfcns like the 1350/6400 etc, regardless the issue on the bladerf that causes the timeout when switched to a lower rate from a higher rate is still applicable at least for the xa4 board that i can test on

aaalbanese, 2022-11-22:
Hey, I have the same issue with a bladeRF 2.0 micro xA5. 

```
  bladeRF-cli version:        1.8.0-git-21084112-dirty
  libbladeRF version:         2.4.1-git-21084112-dirty

  Firmware version:           2.4.0-git-a3d5c55f
  FPGA version:               0.14.0
```
I tried many of the proposed solutions above. Though it looks like downgrading the FPGA image might do the trick, I cannot do it as the xA5 and the xA4 don't share the same FPGA. 

```
export BLADERF_DEFAULT_TUNING_MODE=host
sudo ./lib/src/radio/test/benchmark_radio -d bladerf -s 1.92e6 -f 801.3e6 -t 10 -x -y
```

See the output below.

```
Instantiating objects and allocating memory...
Initialising instances...
Opening 1 channels in RF device= with args=default
Available RF device list: UHD  soapy  bladeRF 
Trying to open RF device 'UHD'
[INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_4.3.0.0-6-g5aa6bc44
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=1, args: 
[INFO] [UHD RF] RF UHD Generic instance constructed
/home/nuc/srsRAN21.10-master/lib/src/phy/rf/rf_uhd_imp.cc:930: uhd_init failed, freeing...
[ERROR] [UHD RF] LookupError: KeyError: No devices found for ----->
Empty Device Address
Unable to open RF device 'UHD'
Trying to open RF device 'soapy'
Soapy has found device #0: backend=libusb, device=0x04:0x08, driver=bladerf, instance=0, label=BladeRF #0 [7a3c782d..ef922e52], serial=7a3c782d209b4d5e93d03d03ef922e52, 
Selecting Soapy device: 0
[INFO] bladerf_open_with_devinfo()
[INFO] bladerf_get_serial() = 7a3c782d209b4d5e93d03d03ef922e52
[INFO] setSampleRate(Rx, 0, 4.000000 MHz), actual = 4.000000 MHz
[INFO] setSampleRate(Tx, 0, 4.000000 MHz), actual = 4.000000 MHz
Setting up Rx stream with 1 channel(s)
Setting up Tx stream with 1 channel(s)
[INFO] setSampleRate(Rx, 0, 1.920000 MHz), actual = 1.920000 MHz
[INFO] setSampleRate(Tx, 0, 1.920000 MHz), actual = 1.920000 MHz
Available device sensors: 
 - RFIC_TEMP
Available sensors for Rx channel 0: 
 - PRE_RSSI
 - SYM_RSSI
State of gain elements for Rx channel 0 (AGC supported):
 - full: 71.00 dB
State of gain elements for Tx channel 0 (AGC not supported):
 - dsa: -10.00 dB
Rx antenna set to RX
Tx antenna set to TX
RF device 'soapy' successfully opened

Warning: TX gain was not set. Using open-loop power control (not working properly)

[INFO] setSampleRate(Rx, 0, 11.520000 MHz), actual = 11.520000 MHz
[INFO] setSampleRate(Tx, 0, 11.520000 MHz), actual = 11.520000 MHz
Setting manual TX/RX offset to 0 samples
Start capturing 10000 sub-frames of 11520 samples (approx. 10s) ...
Changing sampling rate to 11.52 Msamps/s
[INFO] setSampleRate(Rx, 0, 11.520000 MHz), actual = 11.520000 MHz
[INFO] setSampleRate(Tx, 0, 11.520000 MHz), actual = 11.520000 MHz
Setting manual TX/RX offset to 0 samples
00Changing sampling rate to 1.92 Msamps/s
[INFO] setSampleRate(Rx, 0, 1.920000 MHz), actual = 1.920000 MHz
[ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1089] Transfer timed out for buffer 0x7f1ed4070f40
[INFO] setSampleRate(Tx, 0, 1.920000 MHz), actual = 1.920000 MHz
[... 29 lines cut ...]
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
```

Any ideas?   

Thanks!

Antonio

skesir, 2023-02-01:
Hey, I have the same issue with bladeRF 2.0 micro xA9.

  bladeRF-cli version:        1.8.0-git-6ad1a34f
  libbladeRF version:         2.4.1-git-6ad1a34f

  Firmware version:           2.4.0-git-a3d5c55f
  FPGA version:               0.14.0 (configured from SPI flash)

Is there any update? Thanks.

JoseSecmotic, 2023-02-23:
I am facing the same issues, both in bladeRF xA9 and xA5 with the latest FPGA and Firmware images.
Only changing the **n_prb** param to 6 or 15 in enb.conf seems to help, but this is not the desirable behavior since its very limiting

davwheat, 2023-03-19:
For my srsRAN setup, I can only run my bladeRF xA5 at 15 RBs (3 MHz) at srsRAN's stock sample rate.

Setting my RB count to 25 (5 MHz paired spectrum) results in overruns of over 1000 when enabling the RF trace in srsRAN. Same with setting the sample rate to `11520000` as per @JoseSecmotic's comment above.

I'm on the latest FPGA (1.5.0) that released in Feb and it seems to make absolutely no difference to my workflow.

I've had this issue since I first ordered my bladeRF and it's extremely frustrating as it means using the SDR for anything except the most boring, unperformant tests isn't possible. This wasn't what I had in mind.

nitinvjain, 2023-11-27:
Hi Rober (@robertghilduta ) and all,
I am trying to setup srsRAN_4G ( 23.04 for srsRAN_4G) and am able to setup pretty easily with Ettus B210 using srsenb and srsue and attach does happen consistently. But same is not the case with bladeRF 2.0 micro xA4. 

Below is the setup for bladerf.
  bladeRF-cli version:        1.8.0-git-unknown
  libbladeRF version:         2.4.1-git-unknown

  Firmware version:           2.4.0-git-a3d5c55f
  FPGA version:               0.14.0 (configured by USB host)

When I try to run 
sudo ./lib/src/radio/test/benchmark_radio -d bladerf -s 1.92e6 -f 801.3e6 -t 10 -x -y or 

sudo ./lib/src/radio/test/benchmark_radio -d bladerf -s 11.52e6 -f 801.3e6 -t 10 -x -y

Overall , It fails anyway but many times, it gets stuck to following too:
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 400 ms

Please do provide any solution.
I even tried modifying lib/src/phy/rf/rf_soapy_imp.c to set below define constant 

#define STOP_STREAM_BEFORE_RATE_CHANGE 1 from

#define STOP_STREAM_BEFORE_RATE_CHANGE 0

But that also does not make any difference.

Please do share any workarounds or fixes for same.

nitinvjain, 2023-11-28:
@andrepuschmann ,
Any idea if any sure shot solutions to above issue. I am still facing it with all latest releases of bladeRF fw + fpga and libbladeRF ?
I am neither able to pass benchmark_radio pass nor UE srsUE attach happens consistently (highly erratic and inconsistent. Happens once in 10 attempts or so).

davwheat, 2023-11-28:
It's been like this since I first received my xA5 and has never been fixed.

The SDR has just been sat in a drawer being totally useless for me.

nitinvjain, 2023-11-28:
Oh my god. I  purchased 3 weeks back bladerf 2.0 micro XA4 and have been struggling to bring it up for srsRAN_4G. Quite a big let down..If it is indeed bladerf FPGA bit load issue or fw and there is some work around to not create the condition for bladerf, it can be attempted in srsRAN bladerf interface files. I can see that there is some macro in sdr interface code in srsRAN  to stop rx stream while changing tx or rx sample rate(comments in the file give impression that it is meant for limesdr). Any clue to set sampling rates once in beginning and not touch it. It means that cell search will operate on higher rates.. srsRAN folks can advice. I will attempt to take on...

Regards Nitin

rghilduta, 2023-11-29:
@nitinvjain please use my @rghilduta Github user or ideally for faster response time please contact us by email at bladeRF@nuand.com . @robertghilduta is just a placeholder.

The `[ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1089] Transfer timed out for buffer ` issue unfortunately is a catch all symptom. It may potentially be best to retire this issue due to the initial symptom being addressed (FPGA timestamp skips). Subsequent occurrences of the "Transfer time out" error can be caused by many divergent issues, for example: a USB 2.0 connection, USB 3.0 bandwidth saturation, CPU saturation, FX3/libbladeRF/FPGA mismatch (usually caused by apt-get'ing less recent libraries), etc. These issues are generally system or setup specific so we encourage email support ( bladeRF@nuand.com ).
We've even added a new holistic testing suite meant to catch bugs similar that led to the initial #778 bug, please take a look at the [commits dated October 31st, 2023](https://github.com/Nuand/bladeRF/commits/d0fbe0569059f10fcc71095409284586b00fe59b).

closed 2023-11-29 by rghilduta

rghilduta, 2023-11-29:
For posterity sake, please also ensure to the bladerf interface directly (and the soapy one to access the bladeRF).
@nitinvjain I believe this applies in your case as well.

alphafox02, 2023-11-29:
If it helps in the short term, slap DragonOS FocalX on a usb stick. Boot live, and pass the bladerf argument via command line with sudo srsenb. I use the bladerf as the 4g enb and the b205mini as a ue. 

https://m.youtube.com/watch?v=ZNRigxX17r0&feature=youtu.be

metasim, 2024-10-18:
@rghilduta When you closed this, what was the solution/fix to this problem? Maybe I'm missing a detail in the conversation. I'm having the same problem with this version:

```
❯ bladeRF-cli -e version

  bladeRF-cli version:        1.9.0-git-41ef6346-dirty
  libbladeRF version:         2.5.0-git-41ef6346-dirty

  Firmware version:           2.4.0-git-a3d5c55f
  FPGA version:               0.15.0 (configured from SPI flash)
```