Nuand/bladeRF #779: i'm using bladeRF x40

EMAF117, 2020-05-01:
root@raspberrypi:/tmp/bladeRF-master/host/build# cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr/local -DINSTALL_UDEV_RULES=ON ../
-- The C compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check if the system is big endian
-- Searching 16 bit integer
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of unsigned short
-- Check size of unsigned short - done
-- Using unsigned short
-- Check if the system is big endian - little endian
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29") 
-- Checking libusb version...
-- libusb version: 1.0.22

-- Looking for clock_gettime in c
-- Looking for clock_gettime in c - found
-- libad936x: Configuring build...
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- FindPatcher: found /usr/bin/patch
-- libad936x: Applying patch: 0001-0bba46e-nuand-modifications.patch
CMake Error at common/thirdparty/ad936x/CMakeLists.txt:195 (message):
  Failed to apply 0001-0bba46e-nuand-modifications.patch:

-- Configuring incomplete, errors occurred!
See also "/tmp/bladeRF-master/host/build/CMakeFiles/CMakeOutput.log".
See also "/tmp/bladeRF-master/host/build/CMakeFiles/CMakeError.log".

EMAF117, 2020-05-01:
any help please

Nuand, 2020-05-05:
Ensure the bladeRF.git installation directory is clean, and try again .
For installation issues, please email bladeRF@nuand.com or use the forums.