Nuand/bladeRF #780: BladeRF 2.0 micro xA4 do not transmit

Furtad0, 2020-05-08:
I have two machines, a MAC OS where I have gnuradio (version 3.7.13.5)  installed through Mac ports. Another I have Ubuntu 18.04 with gnuradio (version 3.7.14.0) installed through pyboms. In terms of reception I have no problems. On both machines, I can successfully receive RF signals with the BladeRF xA4 boards. However, in terms of Tx I have some issues.

On the MAC OS machine, when I set the "RF Gain" of the osmocom_sink different of zero, it results in the case where the board does not transmit anything (I verified this with a gqrx + BladeRF XA4). When the osmocom_sink "RF Gain" is equal to zero, I have signal transmission as expected.

Given this, I thought it was also the case for the osmocom_sink configuration on the machine with Ubuntu. But no, I already tried several TX gains (RF, IF and BB), and I never have transmission from the bladeRF xA4. Could you clarify what the problem is? How should I set up TX gains in the osmocom_sink?

NOTE: Although with the osmocom_sink no transmission is achieved, when I use de osmocom_siggen the expected transmissions occur.

rghilduta, 2020-06-22:
Out of curiosity is osmocom_siggen able to function properly with the bladeRF 2.0 micro xA4 on the Ubuntu system? What RF/IF/BB values have you tried, have you tried "60"?

Furtad0, 2020-07-07:
Yes, I have tried with 60 on the RF/IF/BB and the issue continues. Since then I moved for the 3.8 Gnuradio. For now everything is working as expected in Gnuradio 3.8.

rghilduta, 2023-02-28:
Closing thread to clean up issues. Latest version of gr-osmosdr and libbladeRF should perform extra bounds checks. Also, please consider using gr-bladeRF: https://github.com/Nuand/gr-bladeRF

closed 2023-02-28 by rghilduta