Nuand/bladeRF #781: Build error on Ubuntu 18.04 with GCC 10 - Implicit enum conversion

TehWan, 2020-05-25:
Trying to build the `host` project on Ubuntu 18.04.1 LTS x64 using GCC 10.1, I get the following error:
```
/source/SDR/bladeRF/host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c: In function ‘bladerf2_open’:
/source/SDR/bladeRF/host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:283:38: error: implicit conversion from ‘enum <anonymous>’ to ‘enum <anonymous>’ [-Werror=enum-conversion]
  283 |     dev->flash_arch->status          = STATE_UNINITIALIZED;
      |                                      ^
cc1: all warnings being treated as errors
libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/build.make:186: recipe for target 'libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/board/bladerf1/bladerf1.c.o' failed
```

I have tried on v2.2.0, v2.2.1, and on master.

Here are the system specs:
CMake
```
$ cmake --version
cmake version 3.17.2

CMake suite maintained and supported by Kitware (kitware.com/cmake).
```

GCC
```
$ gcc --version
gcc (Ubuntu 10.1.0-2ubuntu1~18.04) 10.1.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

uname
```
$ uname -a
Linux workpc 5.3.0-53-generic #47~18.04.1-Ubuntu SMP Thu May 7 13:10:50 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

TehWan, 2020-05-25:
Same issue in bladerf1.c:
```
/source/SDR/bladeRF/host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c: In function ‘bladerf1_open’:
/source/SDR/bladeRF/host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:841:38: error: implicit conversion from ‘enum <anonymous>’ to ‘enum <anonymous>’ [-Werror=enum-conversion]
  841 |     dev->flash_arch->status          = STATE_UNINITIALIZED;
```

TehWan, 2020-05-25:
It seems that GCC 10 gives out more warnings than previous compilers -- for the better. However, this project, by default, treats warnings as errors. To be able to build it, I disabled the `enum-conversion` warning:
```
cmake .. -DCMAKE_C_FLAGS="-Wno-enum-conversion"
```

This is a local fix, but ideally needs a permanent patch solution.

Linux kernel had the same issue here: https://github.com/ClangBuiltLinux/linux/issues/138

They fixed it by using an `int` instead of an anonymous `enum`.

rghilduta, 2020-05-26:
GCC 10 appears to be a bit more cautious about certain declarations. Please try master again on a clean repository after fetching.

rghilduta, 2020-05-26:
GCC 10 appears to be a bit more picky about certain enum declarations. Please try master again on a clean repository after fetching.

rghilduta, 2020-05-26:
Please post/re-open again if this issue still persists.

closed 2020-05-26 by rghilduta

TehWan, 2020-05-30:
Works great, thanks!