Nuand/bladeRF #786: MATLab (R2020a) not working on Linux

sargun, 2020-07-03:
Trying to use the BladeRF on Matlab on Ubuntu Linux 20.04 unfortunately didn't go as smoothly as I hoped. 

I tried to build the thunk, and I got the following warnings:

```

>> bladeRF.build_thunk();
  Build the MATLAB thunk library for use with the bladeRF MATLAB wrapper
 
  bladeRF.build_thunk();
 
  This function is intended to provide developers with the means to
  regenerate the libbladeRF_thunk_<arch>.<library extension> files.
  Users of pre-built binaries need not concern themselves with this
  function.
 
  Use of this function requires that:
    - The system contains a MATLAB-compatible compiler
    - Any required libraries, DLLs, headers, etc. are in the search path or current working directory.
 
  For Windows users, the simplest approach is to copy the following
  to this directory:
    - libbladeRF.h
    - bladeRF1.h
    - bladeRF2.h
    - bladeRF.dll
    - libusb-1.0.dll and/or CyUSB.dll
    - pthreadVC2.dll
 
  Linux users will need to copy libbladeRF.h, bladeRF1.h, bladeRF2.h to this directory
  and add the following #define to libbladeRF.h, after "#define BLADERF_H_" or "#define LIBBLADERF_H"
 
  #define MATLAB_LINUX_THUNK_BUILD_

Warning: Could not find required #define in libbladeRF.h . Please review instructions. 
> In bladeRF.build_thunk (line 287) 
>> bladeRF.build_thunk();
Warning: The data type 'FcnPtr' used by function bladerf_init_stream does not exist. 
> In loadlibrary
  In bladeRF.build_thunk (line 290) 
Warning: The data type 'voidPtrPtrPtr' used by function bladerf_init_stream does not exist. 
> In loadlibrary
  In bladeRF.build_thunk (line 290) 
Warning: Encountered the following warnings while loading libbladeRF:

No match found for enum value expression sizeof in ( 1024 / ( 8 * sizeof ( unsigned long int
))) expression ignored.
Found on line 649 of input from line 8 of file
/usr/include/x86_64-linux-gnu/bits/types/__sigset_t.h

No match found for enum value expression int in  1024 / ( 8 * ( int ) sizeof ( __fd_mask ))
expression ignored.
Found on line 707 of input from line 70 of file /usr/include/x86_64-linux-gnu/sys/select.h

Failed to parse type 'union { char *__size; int __align ; } pthread_mutexattr_t' original
input 'union { char __size [ 4 ]; int __align ; } pthread_mutexattr_t'
Found on line 869 of input from line 36 of file
/usr/include/x86_64-linux-gnu/bits/pthreadtypes.h

Failed to parse type 'union { char *__size; int __align ; } pthread_condattr_t' original input
'union { char __size [ 4 ]; int __align ; } pthread_condattr_t'
Found on line 878 of input from line 45 of file
/usr/include/x86_64-linux-gnu/bits/pthreadtypes.h

Type 'intunionpthread_attr_t' was not found.  Defaulting to type error.

Found on line 895 of input from line 62 of file
/usr/include/x86_64-linux-gnu/bits/pthreadtypes.h

Failed to parse type 'union { struct __pthread_mutex_s __data ; char *__size; long int __align
; } pthread_mutex_t' original input 'union { struct __pthread_mutex_s __data ; char __size [
40 ]; long int __align ; } pthread_mutex_t'
Found on line 905 of input from line 72 of file
/usr/include/x86_64-linux-gnu/bits/pthreadtypes.h

Failed to parse type 'union { struct __pthread_cond_s __data ; char *__size; __extension__
long long int __align ; } pthread_cond_t' original input 'union { struct __pthread_cond_s
__data ; char __size [ 48 ]; __extension__ long long int __align ; } pthread_cond_t'
Found on line 913 of input from line 80 of file
/usr/include/x86_64-linux-gnu/bits/pthreadtypes.h

Failed to parse type 'union { struct __pthread_rwlock_arch_t __data ; char *__size; long int
__align ; } pthread_rwlock_t' original input 'union { struct __pthread_rwlock_arch_t __data ;
char __size [ 56 ]; long int __align ; } pthread_rwlock_t'
Found on line 924 of input from line 91 of file
/usr/include/x86_64-linux-gnu/bits/pthreadtypes.h

Failed to parse type 'union { char *__size; long int __align ; } pthread_rwlockattr_t'
original input 'union { char __size [ 8 ]; long int __align ; } pthread_rwlockattr_t'
Found on line 930 of input from line 97 of file
/usr/include/x86_64-linux-gnu/bits/pthreadtypes.h

Failed to parse type 'union { char *__size; long int __align ; } pthread_barrier_t' original
input 'union { char __size [ 32 ]; long int __align ; } pthread_barrier_t'
Found on line 945 of input from line 112 of file
/usr/include/x86_64-linux-gnu/bits/pthreadtypes.h

Failed to parse type 'union { char *__size; int __align ; } pthread_barrierattr_t' original
input 'union { char __size [ 4 ]; int __align ; } pthread_barrierattr_t'
Found on line 951 of input from line 118 of file
/usr/include/x86_64-linux-gnu/bits/pthreadtypes.h

Type 'unsignedlonglong__extension__' was not found.  Defaulting to type error.

Found on line 1060 of input from line 498 of file /usr/include/stdlib.h

Type 'bladerf_quick_tunePtr' was not found.  Defaulting to type voidPtr.

Found on line 2001 of input from line 1890 of file /home/sargun/tmp/libbladeRF.h

Type 'bladerf_quick_tunePtr' was not found.  Defaulting to type voidPtr.

Found on line 2010 of input from line 1931 of file /home/sargun/tmp/libbladeRF.h

 
> In bladeRF.build_thunk (line 309) 
Warning: Restart program after running bladeRF.build_thunk() 
> In bladeRF.build_thunk (line 312) 
Warning: Could not find current directory in path. 
> In bladeRF.build_thunk (line 314) 
Warning: Consider saving this directory to path by trying: 
> In bladeRF.build_thunk (line 315) 
Warning:     addpath(pwd) ; savepath 
> In bladeRF.build_thunk (line 316) 
```

Upon attempting to load it, I get the following error:

```
Error using calllib
Parameter must be scalar.

Error in bladeRF_XCVR/set.samplerate (line 91)
            [status, ~, ~, actual] = calllib('libbladeRF', 'bladerf_set_rational_sample_rate',
            obj.bladerf.device, obj.module, rate, rate);

Error in bladeRF_XCVR (line 545)
            obj.samplerate = 3e6;

Error in bladeRF (line 565)
            obj.rx = bladeRF_XCVR(obj, 'RX', xb);
```

My default cc is gcc 9:

```
$ cc -v

Using built-in specs.
COLLECT_GCC=cc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/9/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none:hsa
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 9.3.0-10ubuntu2' --with-bugurl=file:///usr/share/doc/gcc-9/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++,gm2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-9 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none,hsa --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 9.3.0 (Ubuntu 9.3.0-10ubuntu2) 
```

rghilduta, 2020-07-03:
Those warnings usually appear when the required #defines are not present in libbladeRF.h , is indicated macro (MATLAB_LINUX_THUNK_BUILD_  ) present the in libbladeRF.h ?

rghilduta, 2020-07-03:
Those warnings usually appear when the required #defines are not present in libbladeRF.h , is the indicated macro (MATLAB_LINUX_THUNK_BUILD_ ) present in libbladeRF.h ?

sargun, 2020-07-03:
I added it, and a CD's (in matlab) to this directory before building the thunk.

```
sargun@sargun-NUC10i7FNH:~/tmp$ dpkg -S /usr/include/libbladeRF.h
libbladerf-dev:amd64: /usr/include/libbladeRF.h
sargun@sargun-NUC10i7FNH:~/tmp$ sha1sum  /usr/include/libbladeRF.h
20689cebf2f9c08c3bc82572261d3b9cc399e859  /usr/include/libbladeRF.h
sargun@sargun-NUC10i7FNH:~/tmp$ grep -n MATLAB_LINUX_THUNK_BUILD_ *
libbladeRF.h:25:#define MATLAB_LINUX_THUNK_BUILD_
libbladeRF.h:81:#elif defined _DOXYGEN_ONLY_ || defined MATLAB_LINUX_THUNK_BUILD_
```

rghilduta, 2020-07-03:
The libbladeRF header files should be in the same directory as where bladeRF.m is and runs from ( host/libraries/libbladeRF_bindings/matlab )

sargun, 2020-07-03:
Move the headers; s ame build error:

```
>> bladeRF.build_thunk
Warning: The data type 'FcnPtr' used by function bladerf_init_stream does not exist. 
> In loadlibrary
  In bladeRF.build_thunk (line 290) 
Warning: The data type 'voidPtrPtrPtr' used by function bladerf_init_stream does not exist. 
> In loadlibrary
  In bladeRF.build_thunk (line 290) 
Warning: Encountered the following warnings while loading libbladeRF:

No match found for enum value expression sizeof in ( 1024 / ( 8 * sizeof ( unsigned long int ))) expression ignored.
Found on line 649 of input from line 8 of file /usr/include/x86_64-linux-gnu/bits/types/__sigset_t.h

No match found for enum value expression int in  1024 / ( 8 * ( int ) sizeof ( __fd_mask )) expression ignored.
Found on line 707 of input from line 70 of file /usr/include/x86_64-linux-gnu/sys/select.h

Failed to parse type 'union { char *__size; int __align ; } pthread_mutexattr_t' original input 'union { char __size [ 4 ]; int __align ; } pthread_mutexattr_t'
Found on line 869 of input from line 36 of file /usr/include/x86_64-linux-gnu/bits/pthreadtypes.h

Failed to parse type 'union { char *__size; int __align ; } pthread_condattr_t' original input 'union { char __size [ 4 ]; int __align ; } pthread_condattr_t'
Found on line 878 of input from line 45 of file /usr/include/x86_64-linux-gnu/bits/pthreadtypes.h

Type 'intunionpthread_attr_t' was not found.  Defaulting to type error.

Found on line 895 of input from line 62 of file /usr/include/x86_64-linux-gnu/bits/pthreadtypes.h

Failed to parse type 'union { struct __pthread_mutex_s __data ; char *__size; long int __align ; } pthread_mutex_t' original input 'union { struct __pthread_mutex_s __data ; char __size [ 40 ]; long int __align ; } pthread_mutex_t'
Found on line 905 of input from line 72 of file /usr/include/x86_64-linux-gnu/bits/pthreadtypes.h

Failed to parse type 'union { struct __pthread_cond_s __data ; char *__size; __extension__ long long int __align ; } pthread_cond_t' original input 'union { struct __pthread_cond_s __data ; char __size [ 48 ]; __extension__ long long
int __align ; } pthread_cond_t'
Found on line 913 of input from line 80 of file /usr/include/x86_64-linux-gnu/bits/pthreadtypes.h

Failed to parse type 'union { struct __pthread_rwlock_arch_t __data ; char *__size; long int __align ; } pthread_rwlock_t' original input 'union { struct __pthread_rwlock_arch_t __data ; char __size [ 56 ]; long int __align ; }
pthread_rwlock_t'
Found on line 924 of input from line 91 of file /usr/include/x86_64-linux-gnu/bits/pthreadtypes.h

Failed to parse type 'union { char *__size; long int __align ; } pthread_rwlockattr_t' original input 'union { char __size [ 8 ]; long int __align ; } pthread_rwlockattr_t'
Found on line 930 of input from line 97 of file /usr/include/x86_64-linux-gnu/bits/pthreadtypes.h

Failed to parse type 'union { char *__size; long int __align ; } pthread_barrier_t' original input 'union { char __size [ 32 ]; long int __align ; } pthread_barrier_t'
Found on line 945 of input from line 112 of file /usr/include/x86_64-linux-gnu/bits/pthreadtypes.h

Failed to parse type 'union { char *__size; int __align ; } pthread_barrierattr_t' original input 'union { char __size [ 4 ]; int __align ; } pthread_barrierattr_t'
Found on line 951 of input from line 118 of file /usr/include/x86_64-linux-gnu/bits/pthreadtypes.h

Type 'unsignedlonglong__extension__' was not found.  Defaulting to type error.

Found on line 1060 of input from line 498 of file /usr/include/stdlib.h

Type 'bladerf_quick_tunePtr' was not found.  Defaulting to type voidPtr.

Found on line 2001 of input from line 1890 of file /home/sargun/bladeRF/host/libraries/libbladeRF_bindings/matlab/libbladeRF.h

Type 'bladerf_quick_tunePtr' was not found.  Defaulting to type voidPtr.

Found on line 2010 of input from line 1931 of file /home/sargun/bladeRF/host/libraries/libbladeRF_bindings/matlab/libbladeRF.h

 
> In bladeRF.build_thunk (line 309) 
Warning: Restart program after running bladeRF.build_thunk() 
> In bladeRF.build_thunk (line 312) 

```

But invoking bladerf works:
```
>> b = bladeRF()

b = 

  bladeRF with properties:

          rx: [1x1 bladeRF_XCVR]
          tx: [1x1 bladeRF_XCVR]
      vctcxo: [1x1 bladeRF_VCTCXO]
        misc: [1x1 bladeRF_Misc]
    loopback: 'NONE'
        info: [1x1 struct]
    versions: [1x1 struct]
          xb: 'None'

>> 
```

So, case closed?

rghilduta, 2020-07-03:
A few warning messages about some types in the libbladeRF headers like that can be expected when running build_thunk.

closed 2020-07-03 by rghilduta