Nuand/bladeRF #787: Binary Save File unable to surpass 4.3 GB

nedonatelli, 2020-07-08:
Configuring the n value in rx config wraps after 2^31 (2.147 bil) i.e. setting n=10G will give the same as n=2G

Setting n=0 works but the overall file being written to will stop at 4.3 GB (uint32 max). 

I've tried a number of different configurations on bandwidth/sample rate and keep hitting the 4.3 GB limit. I'm not receiving any overflows, the last error message being reported is an Invalid Checksum error. Even running with verbosity level set to verbose reports no messages about the invalid checksum

I dug into some of the rx.c and found most variables set to size_t which is 2^64 on my machine. It seems like there's something being assigned as a uint32?

Setting the BW/Sample Rate to 30e6 and n=0 should be able to replicate this in half a minute.

rghilduta, 2020-07-11:
This sounds like a filesystem limit, on ext4 bladeRF-cli should be able to go past 5GB, can you check to see if the filesystem has support for files greater than 4GB per file?

rthomp10, 2023-11-30:
cf181d042794c99be27750b622ef6a952e8a9d8f

closed 2023-11-30 by rthomp10