Nuand/bladeRF #788: bladeRF micro (AD9361) set/get_correction of phase and gain

small-mallet, 2020-07-22:
I am looking to pre-calculate the i/q phase and gain corrections for specific frequencies and load them before acquiring samples. Do the set_correction and get_correction functions work properly for the bladeRF micro? Reading the values using the python wrapper gives values like: -1408(phase), -4992(gain), but the API documentation mentions the values should be [-4096, 4096] (I am guessing that's for the older bladeRF?). I would really appreciate some examples for usage of this functionality.

In python, for RX channel 2, I am using the functions as below:
```
device.get_correction( _bladerf.ChannelLayout['RX_X2'].value, _bladerf.Correction['PHASE'])
device.set_correction( _bladerf.ChannelLayout['RX_X2'].value, _bladerf.Correction['PHASE'], value)
```
#### Issues:
- Any correction I set doesn't seem to have any effect on the output samples (inferred from the image rejection). Should I set_correction **before or after enabling** the channel?
- Does using the get_correction function give me the present internal auto calibrations performed by AD9361?
- The get_correction function seems to return the same values always within an active "session".
- Am I supposed to disable the **automatic quadrature tracking** option of AD9361 or does the set_correction function takes care of it?

small-mallet, 2020-07-22:
One correction and two hacky fixes that seem to get it working (all in bladerf2.c):

- Lines[ bladerf2.c#L1517](https://github.com/Nuand/bladeRF/blob/master/host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c#L1517) and L1537 need to be corrected.

```
AD936X_REG_RX1_INPUT_BC_PHASE_CORR -> AD936X_REG_RX1_INPUT_BC_GAIN_CORR
AD936X_REG_RX2_INPUT_BC_PHASE_CORR -> AD936X_REG_RX2_INPUT_BC_GAIN_CORR
```

- Disable RX Quadrature correction tracking (I inserted this into the set_correction function):

```
/* Disable RX_QUAD Tracking if value*/
ad9361_set_rx_quad_track_en_dis(phy, 0);
```
- Removed the (arbitrary?) bit shifts in set_correction and get_correction functions:

```
data = val << shift; -> data = val; //L1763
data = (value >> shift) & 0xff; -> data = value  & 0xff; //L1902
```

The 8-bit calibration values seem to range from [-128, 127]  with 0 as the starting point. I could try creating a pull request for the corrections but disabling the quadrature tracking might need to be more explicit as it needs an enable/disable function. Please let me know if there is a better way for this.

Abraham403, 2020-09-20:
Can you please elaborate why or what is the advantage of precorrection feature? are you using it with GNCradio? Thanks