Nuand/bladeRF #793: Python3 Binding Not Installing

AgileEngineeringLLC, 2020-10-14:
When I try to install the python3 binding in Ubuntu or Mac OS using:

$ sudo python3 setup.py install

I get this same error:

running install_lib
warning: install_lib: 'build/lib' does not exist -- no Python modules to install

So then I'm unable to "import bladerf" in python (it says module does not exist).

Any suggestions on what I'm doing wrong? Thanks!

rghilduta, 2020-10-14:
Are you running the bladeRF python setup script from host/libraries/libbladeRF_bindings/python ?
Do you see a "Finished processing dependencies for bladerf==1.2.0" at the end of "setup.py install"?
Also, what is your Python version? I just tried it successfully with 3.8.5.

AgileEngineeringLLC, 2020-10-14:
_I just re-installed everything using MacPorts, and now it does seem to execute the setup.py correctly, because I get this:_

Using /Users/donaldkelly/opt/anaconda3/lib/python3.8/site-packages
Finished processing dependencies for bladerf==1.2.0

_But when I type "import bladerf" in Python, I get this:_

Python 3.8.1 (default, Jan  8 2020, 16:15:59) 
Type "copyright", "credits" or "license" for more information.

IPython 7.16.1 -- An enhanced Interactive Python.

import bladerf
Traceback (most recent call last):

  File "<ipython-input-1-2d1d0f79e893>", line 1, in <module>
    import bladerf

  File "<frozen importlib._bootstrap>", line 991, in _find_and_load

  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked

  File "<frozen importlib._bootstrap>", line 655, in _load_unlocked

  File "<frozen importlib._bootstrap>", line 618, in _load_backward_compatible

  File "<frozen zipimport>", line 259, in load_module

  File "/Users/donaldkelly/opt/anaconda3/lib/python3.8/site-packages/bladerf-1.2.0-py3.8.egg/bladerf/__init__.py", line 35, in <module>
    from bladerf import _bladerf

  File "<frozen importlib._bootstrap>", line 991, in _find_and_load

  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked

  File "<frozen importlib._bootstrap>", line 655, in _load_unlocked

  File "<frozen importlib._bootstrap>", line 618, in _load_backward_compatible

  File "<frozen zipimport>", line 259, in load_module

  File "/Users/donaldkelly/opt/anaconda3/lib/python3.8/site-packages/bladerf-1.2.0-py3.8.egg/bladerf/_bladerf.py", line 33, in <module>
    libbladeRF = ffi.dlopen("libbladeRF.so")

  File "/Users/donaldkelly/opt/anaconda3/lib/python3.8/site-packages/cffi/api.py", line 150, in dlopen
    lib, function_cache = _make_ffi_library(self, name, flags)

  File "/Users/donaldkelly/opt/anaconda3/lib/python3.8/site-packages/cffi/api.py", line 832, in _make_ffi_library
    backendlib = _load_backend_lib(backend, libname, flags)

  File "/Users/donaldkelly/opt/anaconda3/lib/python3.8/site-packages/cffi/api.py", line 827, in _load_backend_lib
    raise OSError(msg)

OSError: cannot load library 'libbladeRF.so': dlopen(libbladeRF.so, 2): image not found.  Additionally, ctypes.util.find_library() did not manage to locate a library called 'libbladeRF.so'

_I'm guessing the lib is installed, as I get this when I run the "bladerf-cli -p" command:_

(base) Dons-iMac-4:python donaldkelly$ bladerf-cli -p

  Description:    Nuand bladeRF 2.0
  Backend:        libusb
  Serial:         ae9f773573ef45a3bb1e5f80f09ab6ed
  USB Bus:        20
  USB Address:    18

(base) Dons-iMac-4:python donaldkelly$ 

_Thanks, Don_

rghilduta, 2020-10-14:
Do you know where the libbladeRF object file bladeRF.so was installed?
This appears to be an anaconda3 environment issue, if you installed libbladeRF either with Macports or homebrew, can you check to see if anaconda3 is somehow making Macports or homepath compiled libraries like libbladeRF inaccessible to python3 by changing LD_LIBRARY_PATH?

AgileEngineeringLLC, 2020-10-15:
The _bladerf.py file tries to locate libbladeRF.so, but on my Mac it appears that the lib is called libbladeRF.2.dylib. So when I edit the _bladerf.py file to look for that file, it then allows me to run setup.py, and then "import bladerf" will apparently import in Python. But if I try to run the bladeRF_txrx.py example (testing just rx, no tx), I get 

NoDevError: No devices available

even though I can run "$bladeRF-cli -p" in a terminal and the bladeRF shows up as a device.

rghilduta, 2020-10-15:
Was either homebrew or macports used to compile libbladeRF?
What is the output of "bladerf-tool probe" ?