Nuand/bladeRF #796: Error compiling bladeRF with gcc10

hiperiondev, 2020-10-24:
In fedora 32 with gcc (GCC) 10.2.1 20201016 (Red Hat 10.2.1-6) compilation breaks with:
/var/tmp/SDRANGEL/SRC/bladeRF/host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c: En la función ‘bladerf1_open’:
/var/tmp/SDRANGEL/SRC/bladeRF/host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:841:38: error: implicit conversion from ‘enum <anónimo>’ to ‘enum <anónimo>’ [-Werror=enum-conversion]
841 | dev->flash_arch->status = STATE_UNINITIALIZED;

rghilduta, 2020-11-14:
I cannot seem to recreate this. I tried Fedora Core 33, with GCC version "gcc version 10.2.1 20201016 (Red Hat 10.2.1-6) (GCC) "
Do you know what cmake set the full CFLAGS and gcc options to?

hiperiondev, 2020-11-16:
After pull this error is corrected

closed 2020-11-16 by hiperiondev