Nuand/bladeRF #797: How to access the bladeRf -cli from code written in C language.

serj053, 2020-11-10:
Windows operating system. Trying to access the BladeRf device in C using the code
```
#include <stdio.h>
#include <stdlib.h>

int main()
{
FILE *blrf = popen("bladeRF-cli.exe --interactive", " wr+");
if (blrf = = NULL)
{
printf("Error opening pipe \n");
exit(0);
}

fprintf(blrf, "info");

return 0;
}
```
but  the command line returns an error- _Error opening pipe_  and the  QT environment throws an error -_Invalid parameter passed to C runtime function.Invalid parameter passed to C runtime function._
Please tell me where I am wrong and how to get access to BladeRF management correctly

rghilduta, 2020-11-14:
Please take a look at the libbladeRF documentation, http://www.nuand.com/libbladeRF-doc/v2.2.1/
For examples of using libbladeRF, please take a look here: https://github.com/Nuand/bladeRF/tree/master/host/examples/bladeRF-cli/sync_trx

closed 2020-11-14 by rghilduta

serj053, 2020-11-14:
Thank you very much for the hint, I got lost in the endless expanses of programming .       Sincerely, a beginner programmer.

Closed #797.