Nuand/bladeRF #799: ad9361_init : AD936x initialization error

Abraham403, 2020-11-21:
Bladerf 2 micro is not able to start properly, fpga version used 0.11 (tried 0.11.1 didnt work as well) , firmware version 2.3.2, windows 10 usb driver used (WinUSB (v6.1.7600.16385). all leds light green after using the "bladeRF-cli -v verbose -i -l hostedxA9-latest.rbf" command, so FPGA loads correctly but it seems AD9361 spi isnt initialized properly. i tried on multiple windows pc to eliminate machine error, still same issue same logs

below is the output i get when starting the device

C:\Program Files\bladeRF\x64>bladeRF-cli -v verbose -i -l hostedxA9-latest.rbf
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:642] Using libusb version: 1.0.22.11312
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:519] Found a bladeRF (idx=9)
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/libusb.c:191] Bus 002 Device 001: Nuand bladeRF 2.0, serial 6627b218197f4ad68d89623a87608bcb
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:409] Mutex name: Global\bladeRF-6627b218197f4ad68d89623a87608bcb
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 0
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:292] Read Firmware version: 2.3.2
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:299] Capability mask before FPGA load: 0x0000003f00000000
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:399] Found SPI flash manufacturer: WINBOND.
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:407] Found SPI flash device: W25Q128JV (128 Mbit).
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:445] SPI flash total size = 128 Mbit
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:446] SPI flash page size = 256 bytes
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:447] SPI flash erase block size = 65536 bytes
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:448] SPI flash number of pages = 65536
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:449] SPI flash number of erase blocks = 256 pages
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 2
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/usb.c:131] Reporting no FPGA present - BLADERF_FORCE_NO_FPGA_PRESENT is set.
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 3
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:407] Skipping FPGA configuration and initialization - BLADERF_FORCE_NO_FPGA_PRESENT is set.
Loading fpga...
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 3
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 1
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:125] Read FPGA version: 0.11.0
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:132] Capability mask after FPGA load: 0x0000003f00000abc
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:587] nios_ina219_write: Wrote 0x8000 to addr 0x00
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:574] nios_ina219_read: Read 0x399f from addr 0x00
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:587] nios_ina219_write: Wrote 0x019f to addr 0x00
[DEBUG @ host/libraries/libbladeRF/src/driver/ina219.c:70] Configuration register: 0x019f
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:587] nios_ina219_write: Wrote 0xa000 to addr 0x05
[DEBUG @ host/libraries/libbladeRF/src/driver/ina219.c:82] Calibration register: 0xa000
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/common.c:263] Default tuning mode: Host
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:2276] bladerf2_set_tuning_mode: New tuning mode: Host
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:2327] bladerf2_set_tuning_mode: Initializing Host RFIC control
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/rfic_host.c:84] _rfic_host_initialize: initializating
**ad9361_init : Unsupported PRODUCT_ID 0x0
ad9361_init : AD936x initialization error**
[ERROR @ host/libraries/libbladeRF/src/board/bladerf2/rfic_host.c:91] _rfic_host_initialize: ad9361_init(&phy, &bladerf2_rfic_init_params, dev) failed: An unexpected error occurred
[ERROR @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:187] _bladerf2_initialize: dev->board->set_tuning_mode(dev, default_tuning_mode(dev)) failed: An unexpected error occurred
[ERROR @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:2186] bladerf2_load_fpga: _bladerf2_initialize(dev) failed: An unexpected error occurred
Error: failed to load FPGA: An unexpected error occurred
[ERROR @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1936] bladerf2_trigger_init: Board state insufficient for operation (current "FPGA Loaded", requires "Initialized").
[ERROR @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1936] bladerf2_trigger_init: Board state insufficient for operation (current "FPGA Loaded", requires "Initialized").
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1143] Clearing RX1 retune queue.
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1166] RX1 operation duration: 0
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1143] Clearing TX1 retune queue.
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1166] TX1 operation duration: 0
C:\Program Files\bladeRF\x64>

carlonjohn, 2023-04-28:
I am receiving the exact same error.  Did you ever figure out what the issue is?

Abraham403, 2026-04-25:
It was because of faulty hardware, reach out to nuand support to have it replaced