Nuand/bladeRF #800: Nios2DebugException: Target is broken and needs to be reset,

giomate, 2020-11-29:
Hi there,
i need to implement an synthesizer, which should be controlled by the fpga. We bought the bladerf-micro thinking, that this interface would be doable at the expansion port.
Unfortunately, it has been everything ..but easy.
With a lot of trouble, I did until the compilation of the compilation of the build_bladerf.sh  and the compilation of the software in eclipse.
The next step consisted on configuring a SPI module in the Qsys/Platform Designer, which is went smoothly.
However, the debugging/running of this new module on eclipse seems to be impossible, with the current hardware configuration. 
Basically, the two possible way to debug  NIOS lead me to nowhere:
-Trying NIOS II Hardware ends up killing Quartus, because eclipse freezes
-Trying NIOS II Hardware beta  pops up 	Nios2DebugException: Target is broken and needs to be reset,

After Days:Hours:Minutes of googling , it seem that something on bladeRF-micro  hardware interference with the JTAG reset…. I guess..

I appreciate any comment or ideas

rghilduta, 2020-11-29:
It should be possible to debug the bladeRF's NIOS by following documentation in: https://github.com/Nuand/bladeRF/blob/master/hdl/quartus/launch_eclipse.sh

Please note that prior to connecting to the debugger, the FPGA should be loaded by libbladeRF, preferably by bladeRF-cli.

giomate, 2020-11-30:
Hi Robert,
thanks a lot for your quick answer,
In deed , i tried that at the beginning, but eclipse-nios2 did not like this line
include $(BLADERF_COMMON_DIR)/Makefile (https://github.com/Nuand/bladeRF/blob/master/hdl/fpga/platforms/bladerf-micro/software/bladeRF_nios/Makefile), at the end... i put everything together...ja Iknow...it is kind of doggy but it worked.
- A this point i wondering, if i can deactivate the cypress interface for a while, because i had the filling right there start the problem....., It is possible  thru the USB cable, or i need to weld an JTAG connector on J8?