Nuand/bladeRF #802: [hdl] Issue with modelsim scripts

Max-Manning, 2021-01-03:
I am trying to use modelsim to simulate the bladerf-micro platform. In ModelSim, I navigate to `bladeRF/hdl/fpga/platforms/bladerf-micro/modelsim/` and execute `do compile.do`. I get the following error:

```
# -- Compiling architecture arch of sample_stream_tb
# ** Error: (vcom-11) Could not find work.tx.
# ** Error (suppressible): ../../../ip/nuand/./simulation/sample_stream_tb.vhd(207): (vcom-1195) Cannot find expanded name "work.tx".
# ** Error: ../../../ip/nuand/./simulation/sample_stream_tb.vhd(207): Unknown expanded name.
# ** Error: (vcom-11) Could not find work.rx.
# ** Error (suppressible): ../../../ip/nuand/./simulation/sample_stream_tb.vhd(257): (vcom-1195) Cannot find expanded name "work.rx".
# ** Error: ../../../ip/nuand/./simulation/sample_stream_tb.vhd(257): Unknown expanded name.
# ** Error: ../../../ip/nuand/./simulation/sample_stream_tb.vhd(455): VHDL Compiler exiting
```

This appears to be because `sample_stream_tb.vhd` is being compiled before `rx.vhd` and `tx.vhd`. I've tried editing `compile.do` so that `rx.vhd` and `tx.vhd` are compiled first, but this doesn't work. 

I'm happy to explore the source of this issue in more detail, but if anyone can replicate the bug and/or offer suggestions it would be helpful. Thanks!

rghilduta, 2021-01-15:
compile.do has been updated. Please update 2020.12 tag and see if the problem persists!

Max-Manning, 2021-01-15:
This solves the issue. Thanks!

closed 2021-01-15 by Max-Manning

MatteoNapoleoni, 2021-01-15:
hi, after giving the command `do compile.do`, if i run: `vsim nuand.bladerf_tb` i get the following error:

```
# Loading nuand.nios_system(sim)
# ** Error: (vsim-3732) No default binding for component instance 'U_nios_system'.
#    The following component ports are not on the entity:
#           arbiter_ack
#           arbiter_granted
#           arbiter_request
#    Time: 0 ps  Iteration: 0  Instance: /bladerf_tb/U_bladerf/U_nios_system File: ../vhdl/bladerf-hosted.vhd Line: 425
```
I solved by copying these 3 lines in the nios_system.vhd file (/bladeRF/hdl/fpga/platforms/bladerf-micro/vhdl/tb/nios_system.vhd).
```
	arbiter_request                 :   in  std_logic_vector(1 downto 0)  := (others => 'X');
        arbiter_granted                 :   out std_logic_vector(1 downto 0);
        arbiter_ack                     :   in  std_logic_vector(1 downto 0)  := (others => 'X')
```