Nuand/bladeRF #807: rf_blade_imp.c.533

moe-aal, 2021-02-10:
I'm using **bladeRF micro A4** and cannot get _srsenb_ to run. I keep getting this error:
**/usr/src/srsLTE-release_20_10_1/lib/src/phy/rf/rf_blade_imp.c.533: TX failed: Operation timed out**
and sometimes the following would pop up:
"**Error - pool buffer is empt**y" 

What seems to be the problem here? 
I'm using DragosOS

rghilduta, 2021-02-16:
This appears to the be an srsenb error message, have you tried compiling libbladeRF and SRS-LTE from source?

moe-aal, 2021-02-16:
I have. 
Same error.

KarlL2, 2021-04-16:
This looks like [srsenb TX failed: Operation timed out #790](https://github.com/Nuand/bladeRF/issues/790).

hexcone, 2022-12-14:
Try modifying enb.conf like https://band.radio/srsLTE

rthomp10, 2023-11-30:
This issue should be further diagnosed within SRS-LTE before bringing it over to the bladeRF issue tracker. Narrowing the issue down to a reproducible set of steps outside of SRS-LTE is when the issue should be reopened here.

closed 2023-11-30 by rthomp10