Nuand/bladeRF #811: Floating point exception (core dumped)  using grgsm_livemon

jlkl, 2021-03-26:
Hello，when i use grgsm_livemon catch the gsm information, i meet this error
```
/home/opensourcesdr# grgsm_livemon --arg=bladerf
Warning: failed to XInitThreads()
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

Qt: Session management error: None of the authentication protocols specified are supported
gr-osmosdr v0.1.x-xxx-xunknown (0.1.5git) gnuradio 3.7.11
built-in source types: file osmosdr fcd rtl rtl_tcp uhd plutosdr miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
[bladeRF common] init: DEBUG: entering initialization
[bladeRF source] init: Opening Nuand bladeRF with device identifier string ''
[bladeRF source] Device: Nuand bladeRF Serial # 3a6a...35c1 FW v2.3.2 FPGA v0.10.2
[bladeRF source] init: Buffers: 512, samples per buffer: 4096, active transfers: 32
Floating point exception (core dumped)
```

and the bladrf info is
```
/home/opensourcesdr# bladeRF-cli -i
bladeRF> info

  Board:                    Nuand bladeRF (bladerf1)
  Serial #:                 xxxxxxxxxxxxxxxxxxxxxxxxx
  VCTCXO DAC calibration:   0xae00
  FPGA size:                40 KLE
  FPGA loaded:              yes
  Flash size:               32 Mbit
  USB bus:                  4
  USB address:              3
  USB speed:                SuperSpeed
  Backend:                  libusb
  Instance:                 0

bladeRF> 
```

Can you help me fix it？Thanks

jlkl, 2021-03-29:
oh, i use ubuntu 18.04 on vmware station 16.x, hope this can be useful

rghilduta, 2021-03-29:
Are you able to run this in gdb? And see exactly which library the error occurs in?
libbladeRF doesn't deal with floats, it's likely this is being caused by another library.