Nuand/bladeRF #813: nios ii: operation timed out

agrecascino, 2021-03-31:
hello! i have a bladerf x40 that seems to boot fpga(3 leds at bottom lit green), but all communication to the fpga fails between the fx3 and the fpga with a time out. bladerf -v debug -i doesn't give much more information and i've flashed fx3 and fpga so i'm not sure where to go from here.

brianmwaters, 2021-04-08:
I have an x40 that has what sounds like the same problem w/ FPGA image v0.12.0 (things work fine on v0.11.1). Running just about any application (both tx and rx) results in timeouts. This appears to render the v0.12.0 release completely unusable. Here's some console output for `gqrx`, for example:
`gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0                                                                                                                                                                                                   
built-in source types: file rtl rtl_tcp uhd bladerf                                                                                                                                                                                           
gr::log :WARN: file_source0 - file size is not a multiple of item size                                                                                                                                                                        
Resampling audio 96000 -> 48000                                                                                                                                                                                                               
BandPlanFile is /home/int10h/.config/gqrx/bandplan.csv                                                                                                                                                                                        
BookmarksFile is /home/int10h/.config/gqrx/bookmarks.csv                                                                                                                                                                                      
[INFO] [UHD] linux; GNU C++ version 10.2.0; Boost_107500; UHD_4.0.0.0-0-unknown                                                                                                                                                               
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0                                                                                                                                                                                                   
built-in source types: file rtl rtl_tcp uhd bladerf                                                                                                                                                                                           
[bladeRF common] init: DEBUG: entering initialization                                                                                                                                                                                         
[bladeRF source] init: Opening Nuand bladeRF with device identifier string '*:instance=0'                                                                                                                                                     
[bladeRF source] Device: Nuand bladeRF Serial # 7875...a2b2 FW v2.4.0 FPGA v0.12.0                                                                                                                                                            
[bladeRF source] init: Buffers: 512, samples per buffer: 4096, active transfers: 32                                                                                                                                                           
[bladeRF source] bladerf_source_c: DEBUG: initialization complete                                                                                                                                                                             
FOUND lna                                                                                                                                                   
FOUND rxvga1                                                                                                                                                            
FOUND rxvga2                                                                                                                                                            
[bladeRF source] start: DEBUG: starting source                                                                                                                                                                                                
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 30                                                                                                                        00 ms                                                                                                                                       
[bladeRF source] work: bladerf_sync_rx error: Operation timed out                                                                                                                                                                             
[ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1089] Transfer timed out for buffer 0x7f957c003b30                                                                                                                                
[ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1089] Transfer timed out for buffer 0x7f957c007b40                                                                                                                                
[... cut ...]
[ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1089] Transfer timed out for buffer 0x7f957c06fce0
[ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1089] Transfer timed out for buffer 0x7f957c073cf0
[ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1089] Transfer timed out for buffer 0x7f957c077d00
[ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1089] Transfer timed out for buffer 0x7f957c07bd10
[ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1089] Transfer timed out for buffer 0x7f957c07fd20`

brianmwaters, 2021-04-08:
@agrecascino, a work-around for now is to just downgrade to an earlier version of the FPGA image.

brianmwaters, 2021-04-08:
I have an x40 that has what sounds like the same problem w/ FPGA image v0.12.0 (things work fine on v0.11.1). Running just about any application (both tx and rx) results in timeouts. This appears to render the v0.12.0 release completely unusable. Here's some console output for `gqrx`, for example:
```gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0                                                                                                                                                                                                   
built-in source types: file rtl rtl_tcp uhd bladerf                                                                                                                                                                                           
gr::log :WARN: file_source0 - file size is not a multiple of item size                                                                                                                                                                        
Resampling audio 96000 -> 48000                                                                                                                                                                                                               
BandPlanFile is /home/int10h/.config/gqrx/bandplan.csv                                                                                                                                                                                        
BookmarksFile is /home/int10h/.config/gqrx/bookmarks.csv                                                                                                                                                                                      
[INFO] [UHD] linux; GNU C++ version 10.2.0; Boost_107500; UHD_4.0.0.0-0-unknown                                                                                                                                                               
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0                                                                                                                                                                                                   
built-in source types: file rtl rtl_tcp uhd bladerf                                                                                                                                                                                           
[bladeRF common] init: DEBUG: entering initialization                                                                                                                                                                                         
[bladeRF source] init: Opening Nuand bladeRF with device identifier string '*:instance=0'                                                                                                                                                     
[bladeRF source] Device: Nuand bladeRF Serial # [REDACTED] FW v2.4.0 FPGA v0.12.0                                                                                                                                                            
[bladeRF source] init: Buffers: 512, samples per buffer: 4096, active transfers: 32                                                                                                                                                           
[bladeRF source] bladerf_source_c: DEBUG: initialization complete                                                                                                                                                                             
FOUND lna                                                                                                                                                   
FOUND rxvga1                                                                                                                                                            
FOUND rxvga2                                                                                                                                                            
[bladeRF source] start: DEBUG: starting source                                                                                                                                                                                                
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 30                                                                                                                        00 ms                                                                                                                                       
[bladeRF source] work: bladerf_sync_rx error: Operation timed out                                                                                                                                                                             
[ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1089] Transfer timed out for buffer 0x7f957c003b30                                                                                                                                
[ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1089] Transfer timed out for buffer 0x7f957c007b40                                                                                                                                
[... cut ...]
[ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1089] Transfer timed out for buffer 0x7f957c073cf0
[ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1089] Transfer timed out for buffer 0x7f957c077d00
[ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1089] Transfer timed out for buffer 0x7f957c07bd10
[ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1089] Transfer timed out for buffer 0x7f957c07fd20
```

@agrecascino a functional workaround for now is to just downgrade to an older version of the FPGA image.

smunaut, 2021-05-24:
Exact same here :/

gisforgirard, 2021-09-01:
> I have an x40 that has what sounds like the same problem w/ FPGA image v0.12.0 (things work fine on v0.11.1). Running just about any application (both tx and rx) results in timeouts. This appears to render the v0.12.0 release completely unusable. Here's some console output for `gqrx`, for example:
> 
> ```
> built-in source types: file rtl rtl_tcp uhd bladerf                                                                                                                                                                                           
> gr::log :WARN: file_source0 - file size is not a multiple of item size                                                                                                                                                                        
> Resampling audio 96000 -> 48000                                                                                                                                                                                                               
> BandPlanFile is /home/int10h/.config/gqrx/bandplan.csv                                                                                                                                                                                        
> BookmarksFile is /home/int10h/.config/gqrx/bookmarks.csv                                                                                                                                                                                      
> [INFO] [UHD] linux; GNU C++ version 10.2.0; Boost_107500; UHD_4.0.0.0-0-unknown                                                                                                                                                               
> gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0                                                                                                                                                                                                   
> built-in source types: file rtl rtl_tcp uhd bladerf                                                                                                                                                                                           
> [bladeRF common] init: DEBUG: entering initialization                                                                                                                                                                                         
> [bladeRF source] init: Opening Nuand bladeRF with device identifier string '*:instance=0'                                                                                                                                                     
> [bladeRF source] Device: Nuand bladeRF Serial # [REDACTED] FW v2.4.0 FPGA v0.12.0                                                                                                                                                            
> [bladeRF source] init: Buffers: 512, samples per buffer: 4096, active transfers: 32                                                                                                                                                           
> [bladeRF source] bladerf_source_c: DEBUG: initialization complete                                                                                                                                                                             
> FOUND lna                                                                                                                                                   
> FOUND rxvga1                                                                                                                                                            
> FOUND rxvga2                                                                                                                                                            
> [bladeRF source] start: DEBUG: starting source                                                                                                                                                                                                
> [ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 30                                                                                                                        00 ms                                                                                                                                       
> [bladeRF source] work: bladerf_sync_rx error: Operation timed out                                                                                                                                                                             
> [ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1089] Transfer timed out for buffer 0x7f957c003b30                                                                                                                                
> [ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1089] Transfer timed out for buffer 0x7f957c007b40                                                                                                                                
> [ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1089] Transfer timed out for buffer 0x7f957c00bb50                                                                                                     [... cut ...]
> [ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1089] Transfer timed out for buffer 0x7f957c07fd20
> ```
> 
> @agrecascino a functional workaround for now is to just downgrade to an older version of the FPGA image.

thank you so much for this, I was seriously beginning to think this x40 I just acquired was broken or something, but downgrading to v11.1 instantly fixed it! I had recompiled everything else from the newest versions possible but nothing was changing, until I stumbled upon this github issue somehow... we will never know how many hours/days you have saved me, as well as my sanity.

brianmwaters, 2021-12-09:
Unfortunately, I can confirm this is still a problem on the v0.14.0 bitstream (from the 2021-10-04 release).

Unfortunately, FPGA development is well outside my skill set, so my hands are kind of tied here, otherwise I would have looked into this already.

Looking at the issue backlog and who is commenting, this doesn't seem like the place to get in place with the bladeRF developers. I will try some other channels.

hpux735, 2021-12-14:
I've seen this, too.  I thought it was an issue with SigDigger, but was able to isolate it down to the firmware version.

ShawInnes, 2022-12-31:
This still appears to be a problem, thankfully I stumbled across this issue as I was about to give up totally.

I can confirm that v0.14.0 does NOT work, but v0.12.0 does work okay.

ve7mjc, 2023-12-22:
Kudos gents. Was perplexed. Assumed the problem was due to overhead of VMWare and USB 3.0 controller and was about to dual boot a bare-metal Linux distribution but rolling back to v0.11.1 resolved my issues also.