Nuand/bladeRF #827: BladeRFx40 RX DC calibration table not found.

around-the-worldd, 2021-08-09:
Hi! I'm trying to build my own gsm base station on Ubuntu 18.04 with bladeRFx40. After installations of YateBTS and BladeRF; I configured Radio.Band, Radio.C0, Identity.MCC, Identity.MNC, Identity.LAC on YateBTS. After all;

```
> sudo bladeRF-cli -l /usr/share/Nuand/bladeRF/hostedx40.rbf

Loading fpga...
bladeRF[28250]: [WARNING @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:1705] RX DC calibration table not found. Manual gain control will be used instead.
bladeRF[28250]: [INFO @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:1706] To enable AGC, see "Generating a DC offset table" at https://github.com/Nuand/bladeRF/wiki/DC-offset-and-IQ-Imbalance-Correction
Successfully loaded FPGA bitstream!
```

I apply instructions on https://github.com/Nuand/bladeRF/wiki/DC-offset-and-IQ-Imbalance-Correction for getting rx dc calibration table.

`bladeRF> cal table dc rx`

The rx and tx calibration table files saved under /usr/share/Nuand/bladeRF. After that, I'm retrying;

```
> sudo bladeRF-cli -l /usr/share/Nuand/bladeRF/hostedx40.rbf

Loading fpga...
bladeRF[28250]: [WARNING @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:1705] RX DC calibration table not found. Manual gain control will be used instead.
bladeRF[28250]: [INFO @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:1706] To enable AGC, see "Generating a DC offset table" at https://github.com/Nuand/bladeRF/wiki/DC-offset-and-IQ-Imbalance-Correction
Successfully loaded FPGA bitstream!
```

I'm getting same warning and my mobile phone cannot find my gsm network to join. I'm using Ubuntu 18.04.

rghilduta, 2021-08-09:
What filename was the DC calibration file copied to? Please ensure the filename outputted from bladeRF-cli is not changed. Also depending on OS configurations, please also the locations mentioned in: https://nuand.com/bladeRF-doc/libbladeRF/v2.2.1/configfile.html

around-the-worldd, 2021-08-09:
The filenames are;

4a8a54b4dd653717e25379477a05909c_dc_rx.tbl
4a8a54b4dd653717e25379477a05909c_dc_tx.tbl

~/.config/Nuand/bladeRF (Nope)
~/.Nuand/bladeRF (Nope)
/etc/Nuand/bladeRF (Nope)
/usr/share/Nuand/bladeRF (There is)

I copied files under /usr/share/Nuand/bladeRF;

/usr/share/Nuand/bladeRF/4a8a54b4dd653717e25379477a05909c_dc_rx.tbl
/usr/share/Nuand/bladeRF/4a8a54b4dd653717e25379477a05909c_dc_tx.tbl

atwelm, 2022-08-10:
I'm facing the same issue. Cranking up debugging and adding additional logging it appears in main.c the call for searching for the file is done between BLADERF_FORCE_NO_FPGA_PRESENT being set and unset, which prevents the search from being done.  

I don't see it if I start the cli in interactive mode and it seems to find it. Are they simply being loaded silently by third party applications that use the libraries? If so I'd recommend cleaning up the error a bit since it is confusing.