Nuand/bladeRF #830: bladerf_sync_rx error: Operation timed out

arBalasquide, 2021-09-16:
Hello,

I'm following this GNURadio [tutorial](https://wiki.gnuradio.org/index.php/Guided_Tutorial_Hardware_Considerations) for making a spectrum analyzer. I'm using osmosdr block to use my bladerf as a source, I'm also using gnuradio 3.9.2. My bladeRF sdr is [this one](https://www.nuand.com/product/bladerf-x115/).

However, when I run the flowgraph I get the following error:
```
Generating: '/home/adrian/gr-bladeRF/spec.py'

Executing: /usr/bin/python3 -u /home/adrian/gr-bladeRF/spec.py

QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.9.2.0
built-in source types: file rtl rtl_tcp uhd hackrf bladerf rfspace airspy soapy redpitaya 
[INFO] [UHD] linux; GNU C++ version 11.1.0; Boost_107600; UHD_4.1.0.1-0-unknown
[bladeRF common] init: DEBUG: entering initialization
[bladeRF source] init: Opening Nuand bladeRF with device identifier string '*:instance=0'
[WARNING @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:592] FPGA v0.12.0 was detected, which requires firmware v2.2.0 or later. The device firmware is currently v2.0.0. Please upgrade the device firmware before continuing.

[bladeRF source] Device: Nuand bladeRF Serial # 78dc...17db FW v2.0.0 FPGA v0.12.0
[bladeRF source] init: Buffers: 512, samples per buffer: 4096, active transfers: 32
[bladeRF source] bladerf_source_c: DEBUG: initialization complete
gr::log :ERROR: vmcircbuf_prefs::get - /root/.gnuradio/prefs/vmcircbuf_default_factory: No such file or directory
gr::log :ERROR: vmcircbuf_createfilemapping - vmcircbuf_createfilemapping: createfilemapping is not available
[bladeRF source] start: DEBUG: starting source
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 3000 ms
[bladeRF source] work: [ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1089] Transfer timed out for buffer 0x7fed3c003b70
bladerf_sync_rx error: Operation timed out
[bladeRF source] work: bladerf_sync_rx error: Operation timed out
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 3000 ms
[bladeRF source] work: bladerf_sync_rx error: Operation timed out
[bladeRF source] work: Consecutive error limit hit. Shutting down.
[ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1089] Transfer timed out for buffer 0x7fed3c003b70
[bladeRF source] stop: DEBUG: stopping source
```

Here's the info for my device:
```
[adrian@ar ~]$ bladeRF-cli -i
[WARNING @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:592] FPGA v0.12.0 was detected, which requires firmware v2.2.0 or later. The device firmware is currently v2.0.0. Please upgrade the device firmware before continuing.

bladeRF> info

  Board:                    Nuand bladeRF (bladerf1)
  Serial #:                 78dc6f5d785c1eff4106363c67ab17db
  VCTCXO DAC calibration:   0x9370
  FPGA size:                115 KLE
  FPGA loaded:              yes
  Flash size:               32 Mbit (assumed)
  USB bus:                  4
  USB address:              3
  USB speed:                SuperSpeed
  Backend:                  libusb
  Instance:                 0

bladeRF>
```

I downloaded the [latest FPGA](https://www.nuand.com/fpga_images/) and flashed it to the device with no errors.

I've uploaded the grc flowgraph I made [here](https://pastebin.com/rMwKYb0E) if someone can spot any obvious errors.

I'm new to this so I'm not sure what might be causing this issue. Perhaps a hardware issue? I have it connected to a 3.0 USB port, although not entirely sure if the cable I'm using supports 3.0.

My computer specs:
```
System:
  Host: ar Kernel: 5.14.2-zen1-2-zen x86_64 bits: 64 compiler: gcc v: 11.1.0 
  Desktop: i3 4.19.1 dm: LightDM Distro: Arch Linux 
Machine:
  Type: Desktop Mobo: ASUSTeK model: ROG STRIX B450-F GAMING II v: Rev 1.xx 
  serial: <superuser required> UEFI: American Megatrends v: 4204 date: 01/28/2021 
CPU:
  Info: 8-Core model: AMD Ryzen 7 5800X bits: 64 type: MT MCP arch: Zen 3 rev: 0 
  cache: L2: 4 MiB 
  flags: avx avx2 lm nx pae sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3 svm 
  bogomips: 121594 
  Speed: 3146 MHz min/max: 2200/3800 MHz boost: enabled Core speeds (MHz): 1: 3146 
  2: 3247 3: 2994 4: 2926 5: 2914 6: 3270 7: 3150 8: 2856 9: 3634 10: 3058 
  11: 3218 12: 4544 13: 3438 14: 3351 15: 3111 16: 3329 
```

rghilduta, 2021-09-22:
It appears there might be a version/compatibility mismatch. To upgrade the bladeRF's firmware, please take a look at: https://www.nuand.com/fx3_images/ and update the bladeRF using bladeRF-cli -f .

If you're using GR 3.9, please give gr-bladeRF a look. https://github.com/Nuand/gr-bladeRF/

arBalasquide, 2021-09-22:
The time out seems to be fixed now by updating the firmware. Thank you!

I tried using gr-bladeRF blocks but I get a module not found error every time.

I've already opened an issue a few days ago [here](https://github.com/Nuand/gr-bladeRF/issues/2#issue-992370423)

arBalasquide, 2021-09-23:
Everything works now, the module error was an issue with my python path. Thanks for the help!

closed 2021-09-23 by arBalasquide