Nuand/bladeRF #836: BladeRF 2.0 xA9 AD9361 rx_path_rates

u-erdemir, 2021-11-16:
I am dealing with the wideband signals on Bladerf. The U-shaped noise floor (which is caused by the Sigma Delta ADC) distorts my signal. On the other hand, I tried the same signal with PlutoSDR. It gave me better results by far. I suspected that the variable rx_path_rates. Pluto uses ADCs with 8 or 12 times decimation rates. For example Fs=57.6e6 ->ADC=460.8e6. The problem with the bladerf could be caused by insufficient decimation, I think. How can I manipulate these decimation rates?
Thanks a lot!

small-mallet, 2021-11-18:
You can enable the decimate-by-2 option (bladerf_set_rfic_rx_fir), this mostly flattens the noise floor. But I think you might have to limit the taps in ad936x_params.c to 64 for the dec2 option, something about the [adc_rate : iq_rate] ratio needing to be greater than or equal to 8 (16 * 8 taps).

u-erdemir, 2021-11-22:
Thanks for your reply. It seems that, it only changes the property of the FIR filter. Because when I set it to DEC2 setting, the filter response canged but not adequately. The HB1/2/3 filters might be the solution. How can I adjust those filter's setup?

small-mallet, 2021-11-22:
You can't directly set the HB1/2/3 filters, that decision is taken by the AD9361 no-OS drivers. You can indirectly affect them through decimation settings. I think you need more gain to compensate the overall shape. I don't have the screenshots available but these were the settings I used (and order I set them up):

1. Tuning mode: FPGA
2. RX_FIR: DEC2
3. Samplerate : 52e6
3. Bandwidth: 50e6
4. Frequency: 5.7e9
5. Gain mode: Manual
6. Gain: 25

FPGA tuning mode has some other issues, like setting the RX_FIR after setting samplerate "corrupts" the samples/shape. For my daily driver, I use host mode and replace this function with custom taps (you wouldn't need to set RX_FIR anymore). It has a sharp cutoff but substantial ripple:

```
AD9361_RXFIRConfig bladerf2_rfic_rx_fir_config = {
    3,   // rx (RX1 = 1, RX2 = 2, both = 3)
    -6,  // rx_gain (-12, -6, 0, or 6 dB)
    2,   // rx_dec (decimate by 1, 2, or 4)
    {
          -279,       -1135,       -2156,       -2153,        -627,         995,
           887,        -496,        -883,         274,         892,        -156,
          -948,          61,        1050,          41,       -1189,        -167,
          1374,         334,       -1618,        -572,        1945,         922,
         -2417,       -1481,        3193,        2524,       -4780,       -5184,
         10455,       28832,       28832,       10455,       -5184,       -4780,
          2524,        3193,       -1481,       -2417,         922,        1945,
          -572,       -1618,         334,        1374,        -167,       -1189,
            41,        1050,          61,        -948,        -156,         892,
           274,        -883,        -496,         887,         995,        -627,
         -2153,       -2156,       -1135,        -279
    }, // rx_coef[64]
    // clang-format on
    64,                   // rx_coef_size
    { 0, 0, 0, 0, 0, 0 },  // rx_path_clks[6]
    0                      // rx_bandwidth
};
```