Nuand/bladeRF #839: BladeRF 2.0 xA5 FPGA flash issue

amalik18, 2021-11-19:
I have the xA5 BladeRF and I can't seem to flash the bitstream onto the board. 

On an Ubuntu system, I grab the bitstream from the Nuand website (https://www.nuand.com/fpga/hostedxA5-latest.rbf) and save it to the following location `/usr/share/Nuand/bladeRF/hostedxA5.rbf` and then attempt to flash it to the board using `bladeRF-cli -L /usr/share/Nuand/bladeRF/hostedxA5.rbf` but it gives my the following error:

'''
Writing FPGA to flash for autoloading...
Error: Invalid operation or parameter
'''

Any idea how I could successfully flash my board?

amalik18, 2021-11-23:
Could we bump this a bit? It seems like the `bladerf-fpga-hostedxa5` package doesnt exist in the `ppa:nuandllc/bladerf` repository for Ubuntu Focal.

meriac, 2021-11-28:
Created a pull-request for this - noticed that the built-in flash is not supported by the command line tools: https://github.com/Nuand/bladeRF/pull/842

amalik18, 2021-11-30:
Thank you. Greatly appreciated. Approximately how long would this take to propagate to the `pp:nuandllc/bladerf` repo for Ubuntu Focal?

rghilduta, 2023-02-28:
I believe this has been addressed and fixed in 2022.10 and 2023.02.

closed 2023-02-28 by rghilduta